kizniche/Mycodo #346: New Functions and Requesting Community Input

kizniche, 2017-12-15:
The latest releases of Mycodo has a new category, Functions, with PID Controllers placed as a type of Function.

I'd like to get some input from the community as to what type of functions you think are useful or interesting that should be added.

Functions are, well, functions that carry out specific tasks. These are distinct from Math Controllers, which perform math on other controllers and store a new value (or values) in the measurement database. Functions typically do more than merely store a new value, such as with PID Controllers, which take an input and manipulate outputs in response to the input and PID settings.

Some new Functions I think would be useful:

1. [CoolBot](https://www.storeitcold.com/) clone
2. Air Conditioner/Heater (temperature sensor, heater output, cooler output) - This can currently be achieved with a PID controller, but more AC/Heater-specific options can be included to gear it more toward the specific use.
3. ???

zsole2, 2017-12-16:
I always wanted to suggest it, now that you have asked for... irrigation control. With all the sensors, weather forecast checking, whatever. Like an opensprinkler clone,

RynFlutsch, 2017-12-16:
Convert funktion:
e.g.:  Lux [lx]to [PAR](https://en.m.wikipedia.org/wiki/Photosynthetically_active_radiation) [µmol*m^−2s−1]
If there is a second sensor without raspberry functionality?

Get weather data online (maybe as Input)

kizniche, 2017-12-16:
> I always wanted to suggest it, now that you have asked for... irrigation control. With all the sensors, weather forecast checking, whatever. Like an opensprinkler clone,

I'm not too familiar with the features of this, and I haven't found a comprehensive list of features on the website. Are there any specific features you have in mind? I'm sure Mycodo can already do a lot of what this can do, using just timers and conditionals alone, but I can see where there can be certain functions that utilize several sensors (such as float switch, temperature sensor, and rain sensor combinations) to determine if and how much a valve should be turned on.

> Convert function: e.g.: Lux [lx]to PAR [µmol*m^−2s−1]

What light sensor are you using? Would it be able to convert to PAR? Would a sensor [like this](https://www.licor.com/env/products/light/quantum.html) be more appropriate?

> If there is a second sensor without raspberry functionality?

What do you mean by this?

> Get weather data online (maybe as Input)

This is a good idea. I'll look into reliable weather sources with a good API.

RynFlutsch, 2017-12-16:
> Would a sensor [like this](https://www.licor.com/env/products/light/quantum.html) be more appropriate?
Nice sensor.
Use Lux Sensor and [Quantummeter](https://www.specmeters.com/lightmeters/quantum-meters/#description) Kalibrate Sensor 1 with Sensor 2 
Yes supported Quantumsensor is the better way. Can be used as irrigation method.

zsole2, 2017-12-16:
I'm not familiar either, just remembered it from ancient times, it was one of the first RasPi project. Just like beer-brewing, but I think Mycodo is already perfect for that without any further tweaking. After all, beer needs the  same as cheese, but at a higher temperature... (BTW, thanks for mentioning  CoolBot, I'll check it out, never heard of it).
With the garden irrigation, yes, I'd think mainly of duration timers. Then the timers should be able to be temporarily disabled, by e.g. soil moisture or an alternative sensor after rain. I did not think what conditional could be written, but I have not much experience with them. But this is a feature that should be included like a simple switch, If it rained, stop watering. The conditional for this should be built-in, if you look for user friendly additions.
Weather data  can also be integrated, if it is rain in the forecast, do not start watering (unless it does not come, then start after a preset period).
I did not think of the float switch, but it is indeed good idea when someone uses a cistern.
Maybe I'll find some time during the holidays to look into this.

zsole2, 2017-12-21:
After getting familiarized myself with all the new features and changes in the interface, I'd have a few suggestions. 

Math controllers seem to be great and I remember old times when the seeds of those were there, but this is on a whole new level. On the other hand, these are just ways to manipulate inputs, aren't they? That is what you mentioned in the original post, setting them apart from Functions.
In that sense, they could be placed on the Input page, as a second addition line, and put into a second table, or even into in the same table as other inputs. Whoever uses this, will name them appropriately and know what they are doing, i.e. generating a new input value to be used later. 
In that sense, I think it would help if the input conditionals were in a separate table, just like for output. The user could quickly see what conditionals are set up, rather then remembering for what input they were set up by clicking the '+' signs.

In the same line of thought, timers are just a specific case for output manipulation, could also be merged.

That way the input page would have Add Input Device, Add Math Controller, and Add Conditional, while the output page would have Add Output Device, Add Timer, and Add Conditional.

With respect to Functions, it might worth some consideration whether the conditionals should belong here, based on your definition of a function: they perform specific tasks based on various inputs. I know this contradicts my previous paragraph, but just putting up ideas...

kizniche, 2017-12-21:
I like compact interfaces, and I think Input and Math can be placed on the same page (along with Input and Math conditionals), keeping the menu name 'Input'.

However, I'm not sure I want to separate the conditionals from the actual controllers. Some people use a lot of conditionals (I see one install using 15 of them, with 8 activated, from the stats viewer). I think it would make it difficult to organize or understand which goes where if they're not listed under the controller they're being used with.

With Output and Timers, I think keeping them separate might be better, because some users have many outputs (I have 10, I see another install using 11), and timers (I have several, and see many installs using up to 8). If there were a low number being used, then it may be beneficial to have both on the same page, but for those who have a lot, I think it makes it easier to navigate and manager each when they're on separate pages.

As for conditionals being on the Function page, I think they would benefit from being associated with their respective controller. Although the person who sets up the system may know what's going on with them if they were separate, this may change if you come back after a long time and don't remember how you set it up, or bring a new user on to manage the system. If they are coupled to their controller, it is immediately obvious which controller the conditionals are acting upon, where as if they were in a separate list all together it would mean a lot of jumping around the interface to find which controller they belong to. This may become more confusing if the order of them isn't consistent, with several conditionals for one controller being out of order with other conditionals between them.

Thanks for the suggestion. I'll see about combining the Input and Math controllers to the same page, and think about your other suggestions. Mycodo is already pretty intimidating to new users and I want things to be as easy to understand as possible, even if it means keeping sharp distinctions between things like Outputs and Timers, or keeping groupings, like between a controller and its associated conditionals.

zsole2, 2017-12-21:
I see your point on complexity and fully agree that it can be daunting for beginners ...but it is good to have choices, an I commend your great work, as I regularly  do...
As for jumping around when managing the system, I'd disagree. As for sensors and relays and timers (I also have a few of those, 4 sensors, 8 relays and 3 timers), those are 'set-up-and-forget-it' features for me: once added, they are there, and usually not visited again, unless some manual manipulation is necessary, These are relied upon in PIDs and conditionals, which provide all the output that is needed to me later on (graphs, notifications, etc.), and should be changed if something needs to be adjusted. Which is not really needed, one of my systems is still on v4.0.24 and up since about a year...
I'm not fully using conditionals (I have no relay conditional, and a few for sensors), but one of my issues with the current setup is when I do a fresh install, first come  the general config options, and then start to populate the parameters, go by the interface's logical order, so start with sensors. I cannot add the sensor conditional when I go along, because there are no relays  setup yet, so need to jump forward to the Outputs, then come back for the sensor conditional. If I could set up my sensors, relays and timers (the basics), then I would go to the 'effector' page, i.e.  Functions, and stay  there "forever".
Of course every user has different objectives to achieve, and ways to reach them. I also know that my case may not be the most generic. But  from a new user's standpoint, I think the basics should come easily... I'm not sure though what could be more important for a new user, everyone's might be different, for me, it was the PIDs, after setting up the sensors and the relays.

Just a crazy idea popped into my mind: would it be hard to get a 'Favorites' page? And letting the user put whatever he wishes there, whether it's a sensor, a relay, a conditional, a timer, whatever?

kizniche, 2017-12-22:
The more I think about it, the more I like the idea of conditionals being separate from the controllers and placed on the Functions page with PIDs.

kizniche, 2017-12-22:
> Just a crazy idea popped into my mind: would it be hard to get a 'Favorites' page? And letting the user put whatever he wishes there, whether it's a sensor, a relay, a conditional, a timer, whatever?

A favorites page could work but a lot of work would have to be done in preparation for such a thing, making all other parts more modular. I'm going to work first on moving Maths with Inputs and conditionals with PIDs, then think about the feasibility of a favorites page.

zsole2, 2017-12-22:
Yes, I thought that it would be too much work to do favorites, and so was the idea deemed crazy. Probably there are more important things to do, anyway. Even if you got time, maybe you should do a poll to see how many people think it useful. Myself, I'm not even sure that it is... maybe only a 'nice-to-have' category... 
But if you achieve that kind of modularity, other thinks could become feasible, like a floating/popup window with a specific control, activated from e.g. the Live page or from the graphs...

kizniche, 2017-12-22:
I had some inspiration and redesigned the conditionals to be more modular, and moved them all to the Function page. I also made a separate, dedicated Conditional controller that runs in the daemon, instead of the Math controller having its own conditional code and the Input controller having its own conditional code. Overall, things are much cleaner and work better. Thanks for the suggestion, it really paid off. Here's what the current iteration of the Function page looks like:

[image]

kizniche, 2017-12-22:
In doing this, PID outputs can now be used as Conditional inputs, so you can have an email notification if your PID isn't running (detected as no PID output), or other interesting things.

kizniche, 2017-12-25:
I just finished testing all conditional typed (measurements, output, and edge detection), and am very close to making a release. I just have a few more things to test and tweak.

Theoi-Meteoroi, 2017-12-26:
Awesome work.  I like the changes in 5.5 so far. I haven't been using the more advanced features quite yet as I'm focused on making sure them sensors be sensing.  

I would like to make a RFE - the ability to turn off the LCD backlight.  I want to use the assembly in a grow area with light and dark cycles. The dark cycles must not have light emitted from the LCD display during those periods.  Display updates can continue, however the backlight must be off.

Theoi-Meteoroi, 2017-12-26:
BTW:  I like the PAR sensors y'all are talking about.  But way over my SRP for experimentation, I obtained this device - [https://www.hydrofarm.com/p/LGBQM](url)  It does have a serial port and a really crappy app to log the data.  I'll write a driver at some point soon that can at least put the data in Mycodo.  There are formula conversions between lumens and such, however light measurement is a complex topic and I doubt you could repurpose a general lux light sensor to umols/m-2/sec-1.  What you usually want is integrated PPDF - [https://www.apogeeinstruments.com/conversion-instantaneous-ppfd-to-integrated-ppfd/](url)

     Once you get to the actual mechanics of repeatable crop dynamics, you find out that Quantum PAR measurement and mols/day requirements are what drives modern productivity at scale.   The other thing I learned is that PAR is not what it used to be in respect to spectrum.  Most PAR charts have artificial roll-off in the blue and red in regions that have tremendous effect on phenotype expression.  The charts seem to reflect more of what we can produce artificially rather than what organisms can use photosynthetically.  This is an issue for sensors that are insensitive in those PAR roll-off areas, with the blue light being most energetic and typically unaccounted for in PPFD measurements.

Theoi-Meteoroi, 2017-12-26:
I like the way the web modules keep running - I can use it to bounce the daemon while I iterate with my terrible coding skill.  But the road to Python3 seems like breakage ahead.  I've tried two upgrades from 5.4.15 and well, not good. We will probably have to exercise all the sensor code paths  to check for types issues. So I play this for the developer - [https://youtu.be/qjGSXGCi4Fc](url)  - Ima sure there are gonna be regrets for some.  I think this is a worthwhile turn - Python2 was too permissive with types IMHO.   And Apache2 was apparently not your friend, my second upgrade hangs at:

pi@Dev1:/var/log/mycodo $ tail -f mycodoupgrade.log 
[2017-12-25 23:53:27]   apache2-data apache2-utils libpython-all-dev python-all python-all-dev
[2017-12-25 23:53:27]   python-keyring python-keyrings.alt python-secretstorage python-wheel
[2017-12-25 23:53:27]   ssl-cert
[2017-12-25 23:53:27] Use 'apt autoremove' to remove them.
[2017-12-25 23:53:27] The following packages will be REMOVED:
[2017-12-25 23:53:27]   apache2
[2017-12-25 23:53:35] 0 upgraded, 0 newly installed, 1 to remove and 4 not upgraded.
[2017-12-25 23:53:35] After this operation, 591 kB disk space will be freed.
(Reading database ... 138637 files and directories currently installed.)
[2017-12-25 23:53:37] Removing apache2 (2.4.25-3+deb9u3) ...

Yup. That timestamp was ... about 17 minutes ago. Its dead, Jim. The web upgrade status hung even earlier at:

[2017-12-25 23:50:08] 
[2017-12-25 23:50:08] 
[2017-12-25 23:50:08] Installed version: 5.4.15
[2017-12-25 23:50:08] Latest version: 5.5.1
[2017-12-25 23:50:08] 
[2017-12-25 23:50:08] #### Upgrade to v5.5.1 initiated 2017-12-25_23-50-04 ####
[2017-12-25 23:50:08] 
[2017-12-25 23:50:08] #### Beginning Upgrade: Stage 1 of 2 ####
[2017-12-25 23:50:12] Stopping the Mycodo daemon...Done.
[2017-12-25 23:50:15] Downloading latest Mycodo version to /home/pi/mycodo-5.5.1.tar.gz...Done.
[2017-12-25 23:50:15] Creating /tmp/Mycodo-5.5.1...Done.
[2017-12-25 23:50:15] Extracting /home/pi/mycodo-5.5.1.tar.gz to /tmp/Mycodo-5.5.1...Done.
[2017-12-25 23:50:15] Removing /home/pi/mycodo-5.5.1.tar.gz...Done.
[2017-12-25 23:50:15] Copying /tmp/Mycodo-5.5.1/.upgrade status file to /tmp/Mycodo-5.5.1...Done.

I don't have a bunch of data I can't live without so I typically re-install when upgrade fails.  I did have a successful upgrade from 5.3.0 once, it was a beauty.   I'll try to be more helpful with post 5.5 upgrades however I didn't want to get stuck at a down-rev state.

leoneo, 2017-12-28:
Hi,

I have a proposal for a new function.

I would like to install several RPi systems with mycodo (hive, henhouse, weather station, greenhouse, fish tank,...) and have them all on the same manager, with the possibility of having interactions between them.

Thank you

RynFlutsch, 2018-01-16:
Timespan Timer: Start and Endtime with Seconds

Conditional: If Output x turns on/off for x seconds or any duration
then turn on/off output y after time x

zsole2, 2018-01-19:
New function: ability to get a quick overview of Mycodo through terminal. Imagine the Live page, all important data, but on the command line. Could also include system info, too.

kizniche, 2018-01-19:
Good suggestions. I'll think about how to implement a few of these. Regarding the command line monitoring, I have been wanting to do this for a while (#17, so you can see how old that issue is).

zsole2, 2018-01-19:
Ah, those good old days! Speaking of those open issues, what about #45? I remember seeing a request for something similar recently...

kizniche, 2018-01-19:
I have a rudimentary system working called Remote Admin (accessed through the Config menu). Right now you can only add other Mycodo systems (authenticate) and it will compile and show all measurements from all added systems. I only recently got that all working, so right now it's very basic and mostly a proof of concept. I'd like to turn it into a full-fledged remote admin console, but it's pretty overwhelming to think about what needs to be done to get to that point. I have a constant pull in different directions to develop but I try to work on stuff that interests me at a particular time in order to keep my work on this fun instead of feeling like true work.

zsole2, 2018-01-20:
Agree completely! Keep it fun if you do it for fun and do what interests you. I think all Mycodo users should be  eternally thankful for your exceptionally fast reaction on any issue that is raised. 

So, does putting a relay control on the Dashboard sound fun? Like a series of red/green dots in a box, clicking any of those changes the relay's status. It could be quite useful. Similarly, temporarily disabling sensors and PIDs could also be useful. 

In this respect, I always wanted to request this: if a sensor is turned off, there is a notification on the  related PID being turned off. Can it be done the other way, too? If I turn on a sensor, Mycodo could ask  me if I want to turn on the PID. This could save a few page turns.

kizniche, 2018-01-20:
Now that the Graph page has turned into a true Dashboard, those sound like really good elements to add to it. I'm not sure how the multiple-activation element would fit in, other than it can be a function (activate/deactivate multiple function) and then that can be added to the Dash.

zsole2, 2018-01-20:
Speaking of new features: export/import for settings.
I mean not the saving of the full database, but each parameters separately. Those could be imported into a new install, even if the database version is already ahead a few cycles. I know this could be a lot of coding, and checks everywhere, but having e.g. a text/tagged file with all the parameters and a possibility to import would allow changing quite a few things at the same time, and reload once. This could result a lot of screwed up systems, so you'd need a disclaimer that it is  for advanced users, and no support, go for reinstall.

Theoi-Meteoroi, 2018-01-23:
I agree with zsole2, keep it fun fer yerself.  I can see that parameter export/import would hinge on database consistency/stability and strict versioning.   And the pain from that lasts ( in code overhead ) until you break/renew the version bond at major releases.   At this point myself, I'm not hung up on database version skew.  When I deploy, the versions are frozen (unless BUG!) until the task is complete for whatever task it was assigned. I treat it as an appliance at that point - compatible with similar/same versions, or until Kyle decides otherwise. :)

RynFlutsch, 2018-02-01:
> Conditional: 
If Output 1 turns on/off 
for x seconds or any duration
then turn on/off output 2
after time x [s]

I would like to controll a bigger amount of electrical loads with high Inrush Current.
So the goal is to activate the relays one by one with some seconds delay and turn them all together off.

I think i am the guy who u saw with so much timers ^^ (around 20)
It realy suxx now if i want to change the times.

Example:
Timer Output 1 on 07:00 - 16:00
off 16:01 - 23:00
on 23:01 - 01:00
off 01:01 - 6:59

(4 timers needed for 1 Output)

Conditional for Output 2 -5
behave exactly like the timers for Output 1 but with 5 seconds delay for each output @ turning-on.

Or maybe a dublicate function? So i have only to change the gpio pin.

kizniche, 2018-02-01:
I could add a delay option to the Output Conditional Actions.

> Timer Output 1 on 07:00 - 16:00
> off 16:01 - 23:00
> on 23:01 - 01:00
> off 01:01 - 6:59

Also, an alternative way to setup this (theoretically), is to create PWM Outputs, then create a PWM Method and set when you want your relay to be on to 100 (100% duty cycle) and when you want it to be off to 0 (0% duty cycle). This will act essentially as an on/off instead of an actual PWM, and you can have it all work form one timer. You can also set up different methods so you can easily switch between timer regimens with just a few clicks.

zsole2, 2018-02-02:
I have no experience with high power devices on the relays, but on my PiZero the timers starting in sequence has quite a good delay:
```
2018-01-31 06:46:01,439 - mycodo.timer_1 - INFO - Activated in 550.9 ms
2018-01-31 06:46:03,494 - mycodo.timer_2 - INFO - Activated in 1555.1 ms
2018-01-31 06:46:05,701 - mycodo.timer_3 - INFO - Activated in 1479.9 ms
```
Finally a situation where it is good to have a slow system.
Checking back historically, starting the timers takes around 500-1500 msec, while stopping is in the range of 5000-7000ms.
Isn't this amount of delay enough for you?
And  if there is an inherent  delay anyway, @kizniche can include a little more, for general usage a few secs shouldn't matter in these daily timers.

pekkakorre, 2018-02-09:
Hi,
Thanks for this awesome system!

Possibility to make notes would be nice, kinda diary. Maybe just simple where to write things down, or tap point of graph example strange peak in graphs can be explained.. Maybe timed reminders "fill humidifier!".

I dont know is it possible to make Output Function to start taking timelapse? It would be nice too.

kizniche, 2018-02-09:
There used to be a notes feature that also showed the notes on a graph, in a previous version. I should add that feature again.

And I also like the idea of a conditional action triggering a timelapse start.

RynFlutsch, 2018-02-11:
Hello thanks for the input, was quite busy ;)

> I could add a delay option to the Output Conditional Actions.
The Outputs all turn off, at the same time ?

> Alternatively, I could create a timer similar to the PWM Method timer, that would be a Relay Method timer, which would accept a Relay Output and an On/Off Method.

Sounds great.
Possible to extend that Method-range over Days / Weeks / Months ;) ?

So users have the possibility to build their own day. i.E. with 22h.

frodus17, 2018-02-12:
It would be cool to be able to move around the dashboard items. Also add things like indicator lights to show status. The bar graphs work well to show on/off currently.

RPi-Growbox, 2018-02-12:
I used this script to make Output conditional timelapse, maybe it can be used:
`#!/bin/bash
# Timelapse controller for USB webcam

DIR=/home/pi//Mycodo/timelapse

x=1
while [ $x -le 1440 ]; do

filename=$(date -u +"%d%m%Y_%H%M-%S").jpg

fswebcam -d /dev/video0 -r 1280x720 $DIR/$filename

x=$(( $x + 1 ))

sleep 10;

done;
`

From: [https://www.makeuseof.com/tag/how-to-capture-time-lapse-photography-with-your-raspberry-pi-and-dslr-or-usb-webcam/](url)

> It would be cool to be able to move around the dashboard items. Also add things like indicator lights to show status. The bar graphs work well to show on/off currently.

You can move dashboard items with those Up/Down-buttons. Also like idea of indicator light.

frodus17, 2018-02-12:
RPi-Growbox - Yeah, I'm imagining something like being able to click and drag around. Up and down is OK. I just started playing with MyCodo this last Saturday.

kizniche, 2018-02-13:
> Indicator lights to show status of outputs
> Digital display to show at a glance the temperature, humidity, etc.
> Buttons or toggles to turn outputs on/off either momentarily or toggle
> Input for a setpoint for the PID

All great ideas. The outputs on the dash has been suggested before and I had it on my mind to add. The large digital display would be a nice addition. I'll also think about how to add the Input as a PID Setpoint... Can you give me an example of a use for this (I'm having difficulty seeing utility with this one).

frodus17, 2018-02-13:
Concerning adding an input for PID setpoint: Homebrewing, where I need to set a temperature easily. I'd have the Hot Liquor tank and Boil kettle. I can set PID's up for the HLT and Boil kettle. I can still go into PID's and change there, just thought it'd be nice to set temperature from your dash. 

I can do everything I need now. I just saw the thread where you were looking at ideas.

Another would be Sous Vide, just have it all on one screen. The set up is all done on other pages, so just doing small things like Setpoint on the dash would be cool.

zsole2, 2018-02-13:
In the spirit of the top of the issue, where I believe generalized Functions were sought (but I like that all kind of improvement ideas are welcomed), I also join to propose a specialized  **Sous-vide** controller. 
In fact, I just built one over the weekend from leftover components, and seems to work well. I will summarize my thoughts on this soon, whenever I have time to collect them and write it down, as well as more experience is gathered.

zsole2, 2018-02-13:
Onto  the wishlist of minor  of improvements:  for PID tuning purposes it would be useful to see the Kp, Ki and Kd values on the graph.

zsole2, 2018-02-13:
For the Sous-vide controller, or other major appliance style changes (like Colbot, AC control, irrigation control), the interface should somewhat redesigned, to allow for a specialized start screen for the specific appliance. 
It should include the ability to set the startpage Mycodo comes up with, and probably locking out users from the other pages (thinking of my wife with the sous-vide, she would be frightened from the myriads of options). It can be debated whether this specialized page should be designed specifically with its own controls, or the Dashboard could be adapted. 
For the sous-vide, ability to include simple touch-button interface would be great plus. Adafruit has a combo LCD-keypad modul with I2c interface, that could be easy to implement. Although I'd prefer something more granular, with ability to leave the buttons detached  from the control PCB, but this would obviously include more coding and planning for what the buttons should be able to operate within the whole Mycodo interface. 
On-off button(s) are now  implemented outside of Mycodo, but what if they could be easily included as an Input?
OK, now my commute soon ends, so leave it at this...

kizniche, 2018-02-17:
I just finished adding a new Dashboard element for monitoring and controlling Outputs. I think it came out quite nice. Suggestions for improvement are welcome.

[image]

zsole2, 2018-02-17:
This is great progress!!
If I'm thinking about adding my 8 relays, this would be a little oversized, though. I'd prefer a smaller footprint, with less info at first (simple on/off buttons, last value), and after extending it you can get all other data and the settings, of course.
It never occurred to me until now, is there a specific reason for having 1/12th step for the with of the elements?

kizniche, 2018-02-17:
> It never occurred to me until now, is there a specific reason for having 1/12th step for the with of the elements?

On very large screens, 1/12 can be a lot of space. It's in 12th increments because that's the column system used by Bootstrap, the framework I use for the UI.

Here's after a bit more tweaking, showing the variability of how the elements can be set up. If the Output Controls aren't featured, they will now appear in the dropdown.

[image]

[image]

kizniche, 2018-02-17:
Since the Output  Dashboard element is so similar to a Measurement element, I was able to easily add that with a little modification to the Output element. Configuration is similar to the Output:

[image]

kizniche, 2018-02-17:
I think allowing the number of decimal places to be defined would be a good improvement.

kizniche, 2018-02-18:
I just added the ability to view the P, I, and D values of a PID controller on graphs. This should make it easier to tune. This was a great suggestion.

kizniche, 2018-02-18:
> the interface should somewhat redesigned, to allow for a specialized start screen for the specific appliance.

Like a Kiosk mode?

> For the sous-vide, ability to include simple touch-button interface would be great plus.

I have a 7-inch touch screen (1024x600) on one of mine. It auto-logs in when booted up and automatically starts the web browser and full-screens it.

https://www.waveshare.com/product/7inch-hdmi-lcd-c.htm
https://www.amazon.com/Waveshare-Capacitive-Interface-Raspberry-Beaglebone/dp/B015E8EDYQ

kizniche, 2018-02-19:
I've been working on adding a PID Control element to the dashboard. This would allow viewing all the parameters of the PID, including P, I, and D values as well as control the PID. This would be handy for making an interface for a particular appliance, such that your wife you mentioned could feel comfortable using.

zsole2, 2018-02-19:
> On very large screens, 1/12 can be a lot of space. It's in 12th increments because that's the column system used by Bootstrap, the framework I use for the UI.

I was more like going for 1/16th... But if that is the limitation of the programming, then 1/12 is OK.

> Here's after a bit more tweaking, showing the variability of how the elements can be set up. If the Output Controls aren't featured, they will now appear in the dropdown.

What I think would be useful to have at least an  On/Off button visible to be able to quickly control the output.BTW, can it be a single button? On-button if the output is turned off, and vice versa. 

> Since the Output Dashboard element is so similar to a Measurement element, I was able to easily add that with a little modification to the Output element. 

Very nice! Now that I spent a few hours to perfect my gauges, I'll have to decide to use this instead.

> I've been working on adding a PID Control element to the dashboard.

Can't wait to try it!

> Like a Kiosk mode?

Yes, that is  the word I was looking for.

> I have a 7-inch touch screen (1024x600) on one of mine. It auto-logs in when booted up and automatically starts the web browser and full-screens it.

I remember that you have mentioned it somewhere here, but it is too much for  simple control. And for Sous vide, it can be quite splashy...

kizniche, 2018-02-20:
I just finished the first iteration of the PID control element. It came out pretty nice. I integrated the Output green/red to indicate if it's on or off and the refresh-less button press that activates/deactivates it, like the Outputs have.

[image]

kizniche, 2018-02-20:
I'm probably going to be tweaking the display of these. I want to add units to the PID setpoint and measurement values. I also want to add a checkbox to enable the display of more detailed information like P, I, and D values, output value, etc.

kizniche, 2018-02-20:
I think I'm going to make a release so people can start using it and give feedback about how it can be improved.

kizniche, 2018-02-20:
It works pretty nice alongside the graph, for easy viewing of current conditions.

[image]

zsole2, 2018-02-20:
> ... and the refresh-less button press that activates/deactivates it, like the Outputs have.

That will be very useful! It would also be good to implement throughout, not just here and for the Outputs. And I just realized a few days ago that it is  possible to change the PID setpoint without deactivating the controller. It would a useful feature at  more places, LCD messages come to my mind... or changing Input periods.

kizniche, 2018-02-20:
The issue is WTForms does not play nicely with the ajax form submission (refresh-less form submission). I got it to work for buttons that have a dedicated task, but refactoring the entire form system of Mycodo would be an immense feat.

kizniche, 2018-02-20:
The P, I, and D values added to the dash element really helps with understanding what's going on with tuning. Coupled with showing them on the graph should help a lot with tuning.

[image]

zsole2, 2018-02-21:
Someone above mentioned reordering elements by drag-and-drop. I guess is not feasible without a lot of work. 
However, having a dozen items makes using up/down quite time consuming to get the desired placement. Like now, for testing the new dashboard items, and moving them from 14th place to the top to see them easily...
I can imagine that setting a position number for a given element and then move it into that position could be much easier to code than implementing drag-and-drop, but get us much batter control.

frodus17, 2018-02-21:
I think that'd be the easiest, rather than having to click up a dozen times.

Also, is there a way to have a master page save? rather than in each page, work on a few things on a few elements and have to hit save on each one at a time? I forget sometimes that I changed something on another element, and have to go back through it again.

zsole2, 2018-02-22:
The dashboard widgets for Output and PID reset the Refresh period to None when a setting is changed. If I remember correctly, this was the situation with the gauges, so should be an easy fix.

[58 more replies]

closed 2018-12-09 by kizniche