kizniche/Mycodo #350: Cannot delete input

zsole2, 2017-12-18:
## Mycodo Issue Report:
  - Specific Mycodo Version:

 _5.4.19_

#### Problem Description
  - what were you trying to do

_On the input page, deleting HTU21D sensor_
 
### Errors
  - List any errors you encountered.

_Error: Delete Input: type object 'Conditional' has no attribute 'input_id'
There was no conditional set up for this sensor_

  - Copy and pasting crash logs...

_Nothing apparent in the logs_

kizniche, 2017-12-18:
I can reproduce this, but I'm not going to push a fix for this just yet.

The reason is that I'm currently working on moving Mycodo from using Python 2.7 to Python 3.4. I just managed to get both the backend and the frontend working and am in the process of fixing all the minute details related to moving form python 2.x to 3.x. I have a fix committed in the [python3_merge branch](https://github.com/kizniche/Mycodo/tree/python3_merge), and will make a big release that will include the fix for this issue.

kizniche, 2017-12-20:
I've incorporated the fix for this somewhere in the commits from the python3_merge branch that was recently merged with master. I should have a release ready in the next day or two. I've tested everything (including a fresh install) except the upgrade process.

closed 2017-12-20 by kizniche