kizniche/Mycodo #352: Nginx web server

kizniche, 2017-12-21:
This is a continuation of #305

With v5.5.0 about to be released, there has been one issue holding it back, and that is Apache2 returning a 400: Bad Request error when a backup file/database is uploaded from the Export/Import page.

This wasn't an issue all throughout testing. It was working flawlessly. Then, last night when I was about to make a release, I went to the Export/Import page and tried to restore a database and it gave that error. I spent many hours trying to diagnose the issue, without success.

With that, I think I'm through with Apache2 and would like to attempt to get the Flask frontend running on nginx+wsgi. Previously I was unable to get this working. However, I just recently was able to get nginx running the app. Below is how I did that.

Install Mycodo normally according to the instructions in the README, then create ~/Mycodo/wsgi.py with these contents:

```python
from mycodo.start_flask_ui import app

if __name__ == "__main__":
    app.run()
```

Activate the virtualenv and start the server:

```bash
source ~/Mycodo/env/bin/activate
uwsgi --socket 0.0.0.0:5000 --protocol=http -w wsgi:app
```

Go to http://IP-ADDRESS:5000 and the frontend loads

I'll be working on this and determining if it's stable and if this should be the default web server of 5.5.0 and onward.

kizniche, 2017-12-21:
So far nginx appears much faster than apache2

kizniche, 2017-12-22:
I haven't been having success with the uwsgi module, but when I just tried gunicorn from [this tutorial](https://gist.github.com/xaratustrah/0e648a0dca74c661c1a1c78acbd5e224) I was able to get the mycodo frontend working! Details to come.

pukkita, 2017-12-24:
Congrats! nginx runs circles around apache2 performance.

kizniche, 2017-12-25:
So far I've enjoyed working with nginx. I haven't had any issues so far and it appears faster. I'd like to do a side-by-side speed comparison with pre-4.4.19 and 5.5.0 with the [profiler](https://github.com/kizniche/Mycodo/blob/1bc708e916e1c055978b2cb595d8f72a699447b5/mycodo/scripts/profile_analyzer.py) I made a while back to analyze page loading times (back when Mycodo pages took sometimes up to minutes to load, I needed a way to test my code improvements).

closed 2017-12-25 by kizniche