kizniche/Mycodo #353: Gauge & graph problem

pewi81, 2017-12-22:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 5.4.19
#### Problem Description
Please list:
When modifying existing gauges they will not who at all when saved, for example if I wish to cha
ngt size and/or scale it won't show at all after saved. Just the square fram (window) of the gauge is shown.

Also, the graphs don't show any information at all, just the windows/frames of the graph can be seen. I tried to delete gauges and graphs with no success. 

No error messages shown and I cant find anything unusual in the logs.

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Creta gauger and/graph
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-12-22:
There were some gauge issues I fixed recently that will come out with the next 5.5.0 release. I'm working on a few more things and testing a bit before releasing. Since it's a major python version change and a move from apache2 to nginx, I want to make sure I found all the issues (hah!) before letting it into the wild.

pewi81, 2017-12-24:
Ah, I see! Thanks!

kizniche, 2017-12-25:
Let me know if the issue persists after you upgrade to 5.5.0

closed 2017-12-25 by kizniche