kizniche/Mycodo #354: i2cdetect status display incorrect - duplicate and missing bus v5.4.15

Theoi-Meteoroi, 2017-12-22:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.4.15
 
#### Problem Description
Please list: 

     i2cdetect listing on status page skips bus 12, lists bus 1 twice.

  - what were you trying to do
  - specific setup details that are involved

 I have systems with both TCA9545 and TCA9548 multiplexers that have 15 enumerated bus entries.

pi@Dev1:~ $ ls -al /dev/i2c*
crw-rw---- 1 root i2c 89,  1 Nov  3  2016 /dev/i2c-1
crw-rw---- 1 root i2c 89, 10 Nov  3  2016 /dev/i2c-10
crw-rw---- 1 root i2c 89, 11 Nov  3  2016 /dev/i2c-11
crw-rw---- 1 root i2c 89, 12 Nov  3  2016 /dev/i2c-12
crw-rw---- 1 root i2c 89, 13 Nov  3  2016 /dev/i2c-13
crw-rw---- 1 root i2c 89, 14 Nov  3  2016 /dev/i2c-14
crw-rw---- 1 root i2c 89, 15 Nov  3  2016 /dev/i2c-15
crw-rw---- 1 root i2c 89,  3 Nov  3  2016 /dev/i2c-3
crw-rw---- 1 root i2c 89,  4 Nov  3  2016 /dev/i2c-4
crw-rw---- 1 root i2c 89,  5 Nov  3  2016 /dev/i2c-5
crw-rw---- 1 root i2c 89,  6 Nov  3  2016 /dev/i2c-6
crw-rw---- 1 root i2c 89,  7 Nov  3  2016 /dev/i2c-7
crw-rw---- 1 root i2c 89,  8 Nov  3  2016 /dev/i2c-8
crw-rw---- 1 root i2c 89,  9 Nov  3  2016 /dev/i2c-9

### Errors

The bus works fine configured for devices in the system. 
Below is the output from that page.

[Dev1 - Info - Mycodo 5.4.pdf](https://github.com/kizniche/Mycodo/files/1583674/Dev1.-.Info.-.Mycodo.5.4.pdf)

### Steps to Reproduce the issue: 

Add multiplexers using device tree overlays to obtain a bus count higher than 12

Check the Web status page.

### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-12-23:
I fixed the order a few days ago (unreleased), but the duplication is new to me. I'll look into it. Thanks.

kizniche, 2017-12-23:
Can you confirm it is actually skipping 12 and not just dropping the "2" of the 12 in the display on the info page?

Theoi-Meteoroi, 2017-12-23:
pi@Dev1:~ $ sudo i2cdetect -y 12
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- 4d -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- UU -- -- -- --                      

0x4d is a MH-Z16  sensor - the only one in the system.

kizniche, 2017-12-26:
Let me know, when you have a chance, if v5.5.0+ still has this issue.

Theoi-Meteoroi, 2017-12-26:
Yup, still missing in v5.5

kizniche, 2017-12-26:
I found the issue. I was misunderstanding how replace() works. It strips all characters in the input string, not the literal input string itself. Because there was a "2" in my input string, it removed the 2 from "12":

```device_int = int(each_dev.strip("/dev/i2c-"))```

replace() is more appropriate for this, to acquire the device number:

```device_int = int(each_dev.replace("/dev/i2c-", ""))```

closed 2017-12-26 by kizniche

Theoi-Meteoroi, 2017-12-26:
Yum Yum. (burp).  Tested the change and it is goodness.  Bus 12 shows face on 5.5.x

I2C Bus 12: i2cdetect -y 12
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- 4d -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- UU -- -- -- --