kizniche/Mycodo #356: Suggestion: Motion sensor

pewi81, 2017-12-24:
Just some suggestions here.
I was thinking about the ability to add some sort of motion sensoring feature using a PIR like [Adadfruits PIR](https://www.adafruit.com/product/189) or a camera. This feature could be used for multiple purposes, such as activating a heater when a reptile enters a certain area of a terrarium, or as a sensor to automatically close a window/door of a greenhouse if it gets to windy outside, or just to sense when someone enters the area of interest to avoid such things like spraying water or to shut down fans to avoid injury. Could also be a great burglary alarm as a bonus.
If using a camera one could possibly chose a part of the frame in which motion will activate a conditional/relay etc. but I guess a PIR is much easier to implement.
Regards
Pete

kizniche, 2017-12-24:
I've been using a sensor like this to detect motion for about 6 months now :)

The Edge Detection Conditional is what's used to trigger events, like take a photo, email, etc.

I plan to add more Conditional Actions to make Conditionals more versatile, to do all the things you mention.

pewi81, 2017-12-24:
Great, I will order a PIR and give it a try :)

closed 2017-12-25 by kizniche