kizniche/Mycodo #362: Error 502: Bad Gateway Error with v5.5 install

Theoi-Meteoroi, 2017-12-26:
## Mycodo Issue Report:
  - Specific Mycodo Version: Wipe/Replace failed 5.4.15-> 5.5.1

rm -rf ~/Mycodo

Copy 5.5.1 into place @ ~pi and run install.
 
#### Problem Description
Please list:

Web page never goes live, bad gateway error persists.
Reboot cleared error on another setup.

  - what were you trying to do
  - specific setup details that are involved
 
### Errors

Error 502: Bad Gateway Error

Something bad happened but it's probably not your fault. Try refreshing the page to see if it will load. If that doesn't work, try rebooting your system, then try refreshing it again. If that still doesn't work, submit a new issue on GitHub with any usefull information that can help determine the cause of this issue.

  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?

2 fer 2.  Will try on another node after reboot of this node.
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

I expect rebooting this node will correct the startup problem. Including setup.log.
[setup.log](https://github.com/kizniche/Mycodo/files/1586979/setup.log)

Theoi-Meteoroi, 2017-12-26:
Yup. Reboot and refresh of the page, get to useradd and the cycle begins.  Checking the I2C variant of MH-Z16 with v5.5.1 now.

Theoi-Meteoroi, 2017-12-26:
The i2c variant of MH-Z16 seems to work fine without mods.  No errors and BME280 works fine!

kizniche, 2017-12-26:
Some part of the web server restart seems to hang, causing a broken socket. I added more logging lines at the end of the upgrade script to find out what part is the issue, during my next upgrade. Since it's only a minor inconvenience (a reboot fixes the issue), I'm not considering it a critical issue to fix. It may come down to what I did for the daemon, which was create a script to attempt to shut it down nicely, and if that fails to occur after a reasonable amount of time, kill it forcefully.

kizniche, 2017-12-26:
Thanks for the testing and fixing the serial code for the MH-Z sensors!

Theoi-Meteoroi, 2017-12-26:
My pleasure, to pay it forward.  That is how I see my contribs. You deserve awesome support from users for providing a excellent playground. I like to see engineering that way, Engineering is fun.

RynFlutsch, 2017-12-30:
got this after 2 reboots:
`pi@raspberrypi:/var/log/nginx $ less error.log 

2017/12/30 17:44:28 [notice] 5599#5599: signal process started
2017/12/30 17:44:38 [crit] 5601#5601: *2 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 127.0.0.1, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/", host: "localhost"
2017/12/30 17:45:38 [crit] 5601#5601: *4 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 127.0.0.1, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/", host: "localhost"
2017/12/30 17:46:38 [crit] 5601#5601: *6 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 127.0.0.1, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/", host: "localhost"
2017/12/30 17:47:38 [crit] 5601#5601: *8 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 127.0.0.1, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/", host: "localhost"
2017/12/30 17:48:38 [crit] 5601#5601: *10 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 127.0.0.1, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/", host: "localhost"
2017/12/30 18:00:50 [crit] 5604#5604: *13 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.178.23, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/", host: "192.168.178.28"
2017/12/30 18:05:06 [crit] 573#573: *2 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.178.23, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/", host: "192.168.178.28"
2017/12/30 18:05:26 [crit] 573#573: *2 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.178.23, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/", host: "192.168.178.28"
2017/12/30 18:17:07 [crit] 573#573: *5 connect() to unix:/var/run/mycodoflask.sock failed (2: No such file or directory) while connecting to upstream, client: 192.168.178.23, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock:/", host: "192.168.178.28"`

kizniche, 2017-12-30:
This appears to be after the latest upgrade. Make sure the new Config files have been installed. Try this:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-web-server
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh restart-web-server
```

Or just run the post upgrade script that incorporates these commands:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh.sh
```

jj2473, 2017-12-31:
pi@Raspibator:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-web-server

#### Installing and configuring nginx web server
Removed symlink /etc/systemd/system/multi-user.target.wants/mycodoflask.service.
Removed symlink /etc/systemd/system/mycodoflask.service.
Synchronizing state for nginx.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d nginx defaults
Executing /usr/sbin/update-rc.d nginx enable
Created symlink from /etc/systemd/system/multi-user.target.wants/mycodoflask.service to /home/pi/Mycodo/install/mycodoflask.service.
Created symlink from /etc/systemd/system/mycodoflask.service to /home/pi/Mycodo/install/mycodoflask.service.
pi@Raspibator:~ $ sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh restart-web-server

#### Reloading nginx
#### Reloading mycodoflaskJob for mycodoflask.service failed. See 'systemctl status mycodoflask.service' and 'journalctl -xn' for details.

#### Creating Mycodo database if it doesn't exist
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...

jj2473, 2017-12-31:
systemctl status mycodoflask.service
● mycodoflask.service - gunicorn daemon for Mycodo flask web user interface
   Loaded: loaded (/home/pi/Mycodo/install/mycodoflask.service; enabled)
   Active: failed (Result: exit-code) since Sun 2017-12-31 12:51:37 EST; 8min ago
 Main PID: 584 (code=exited, status=203/EXEC)
pi@Raspibator:~ $

kizniche, 2017-12-31:
I can post the command later to try to run the web server manually to see if it's throwing an error. I'm currently out and only have my phone. If you want to try, open the nginx Config file in the install directory and the gunicorn start command is in there. Make sure you're in the Mycodo/mycodo directory when you execute it. Also make sure the flask service is stopped before executing it, with ```sudo service mycodoflask stop```

jj2473, 2017-12-31:
Sounds good.

Thanks,
Jim

kizniche, 2017-12-31:
I didn't see the initialize command executed, above, so make sure that's executed at least once. Here are the commands to try to determine if the web server can start:

```bash
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize
sudo service myodoflask stop
cd ~/Mycodo/mycodo
sudo /var/www/mycodo/env/bin/gunicorn --workers 2 --worker-class gthread --threads 2 --pid /var/lock/mycodoflask.pid --bind unix:/var/run/mycodoflask.sock start_flask_ui:app
```

jj2473, 2018-01-01:
pi@Raspibator:~/Mycodo/mycodo $ sudo /var/www/mycodo/env/bin/gunicorn
--workers 2 --worker-class gthread --threads 2 --pid
/var/lock/mycodoflask.pid --bind unix:/var/run/mycodoflask.sock
start_flask_ui:app
sudo: /var/www/mycodo/env/bin/gunicorn: command not found

kizniche, 2018-01-01:
For some reason you don't have gunicorn. Run the post upgrade script I suggested and that should fix it. Watch the output for errors.

```sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh```

jj2473, 2018-01-01:
Not installing gunicorn, where is it supposed to be grabbing it?

jj2473, 2018-01-01:
Ign http://archive.raspberrypi.org jessie/ui Translation-en_US
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_US
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'apache2' is not installed, so not removed
The following packages were automatically installed and are no longer
required:
  apache2-data apache2-utils
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
fswebcam is already the newest version.
gawk is already the newest version.
gcc is already the newest version.
git is already the newest version.
libav-tools is already the newest version.
libffi-dev is already the newest version.
logrotate is already the newest version.
moreutils is already the newest version.
nginx is already the newest version.
python-setuptools is already the newest version.
python3 is already the newest version.
python3-dev is already the newest version.
python3-numpy is already the newest version.
sqlite3 is already the newest version.
wget is already the newest version.
libi2c-dev is already the newest version.
python3-pigpio is already the newest version.
python3-smbus is already the newest version.
The following packages were automatically installed and are no longer
required:
  apache2-data apache2-utils
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.
Searching for pip
Best match: pip 9.0.1
Processing pip-9.0.1-py2.7.egg
pip 9.0.1 is already the active version in easy-install.pth
Installing pip script to /usr/local/bin
Installing pip2.7 script to /usr/local/bin
Installing pip2 script to /usr/local/bin

Using /usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg
Processing dependencies for pip
Finished processing dependencies for pip
Requirement already up-to-date: pip in
/usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg

#### Installing and configuring nginx web server
Removed symlink
/etc/systemd/system/multi-user.target.wants/mycodoflask.service.
Removed symlink /etc/systemd/system/mycodoflask.service.
Synchronizing state for nginx.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d nginx defaults
Executing /usr/sbin/update-rc.d nginx enable
Created symlink from
/etc/systemd/system/multi-user.target.wants/mycodoflask.service to
/home/pi/Mycodo/install/mycodoflask.service.
[... 147 lines cut ...]
#### Reloading mycodoflaskJob for mycodoflask.service failed. See
'systemctl status mycodoflask.service' and 'journalctl -xn' for details.

#### Creating Mycodo database if it doesn't exist
#### Could not connect to http://localhost. Waiting 60 seconds then trying
again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying
again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying
again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying
again (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying
again (up to 5 times)...
#### Trying again...
pi@Raspibator:/var/www/mycodo/env/bin $ sudo
/var/www/mycodo/env/bin/gunicorn --workers 2 --worker-class gthread
--threads 2 --pid /var/lock/mycodoflask.pid --bind
unix:/var/run/mycodoflask.sock start_flask_ui:app
sudo: /var/www/mycodo/env/bin/gunicorn: command not found
pi@Raspibator:/var/www/mycodo/env/bin $

kizniche, 2018-01-01:
It appears there are several issues with your install (it appears to be an upgrade). You have a database issue and a virtualenv issue. I would recommend a fresh installation.

kizniche, 2018-01-01:
I pushed a fix for the virtualenv check, but that won't fix your database issue. The database issue came about sometime in the past several releases (who knows exactly where). But, it would be difficult to fix. I've done it in the past, but had to go through each database version and try each one until we found the correct command. It would probably be easier to just install from scratch and reconfigure your system. The latest versions don't appear to have the database issue anymore.

jj2473, 2018-01-01:
Hello Kyle,
Ok, was thinking about that anyway.

Thanks,
Jim

kizniche, 2018-01-04:
I'm pretty sure the issues are resolved from this thread, so I'm closing it.

closed 2018-01-04 by kizniche

pukkita, 2018-01-04:
Same issue happened to me, databases doesn't seem to have been upgraded. virtualenv is fine.

I assume the mycodo config is stored inside the influxd database? Would like to recover it, as a reference at least to re-install and reconfigure

kizniche, 2018-01-04:
What version did you upgrade from? I believe the database issue has been resolved, so if you were upgrading you may have had a database with an issue from a previous upgrade attempt.

In any case, the settings database is located at Mycodo/databases/mycodo.db and if you know the version it worked with, you can download that specific Mycodo version release, install it, and copy your database in that directory, restart the daemon, and it will start using it.

If you want to attempt to recover your database, it can be done, but you'll have to go through a few commands to find the right one, after investigating the particular database upgrade error to determine exactly which version it's stuck on. I can try to put together a small guide how to do this with the reference to the other thread where it was successful twice, for you to follow.

kizniche, 2018-01-04:
Also, there's a backup created during an upgrade, at /var/Mycodo-backups that you can find your old database in.

kizniche, 2018-01-04:
@pukkita, I created a wiki page for you to follow, at https://github.com/kizniche/Mycodo/wiki/Fix-Database-Version-Mismatch-Issue

Let me know how it goes and if you need assistance or if your database hasn't been fixed by following those instructions.

pukkita, 2018-01-04:
Wow, thanks Kyle!

Do you have a crystall ball or something?  :) scenario was exactly as described there. 

In fact, I "fixed" the database just by following the guide to the letter, including the second alembic stamp/upgrade.

I'm afraid though I didn't formulate properly my question,  sorry about that:

I read you advised to fresh install (which I found absolutely logical) that's what I did; what I was asking for:  where to recover configuration details (GPIO pins, graphs etc) from the previous install database; I assume configuration details are inside a table inside a database?

Tried to start the previous version with fixed, upgraded database to have a look, but seems the service files reference new paths, and mycodo doesn't start.

I went back to mycodo 5.5.4 I installed from scratch and it started fine, prompting me to upgrade to 5.5.5, which went smoothly.

kizniche, 2018-01-04:
You can probably do this to use your "fixed" database on a current version of Mycodo (no matter what the version is):

1. Stop the daemon and frontend:

```
sudo service mycodo stop
sudo service mycodoflask stop
sudo service nginx stop
```

2. make a backup of your current database and copy in your new database:

```
cd ~/Mycodo/databases
cp mycodo.db mycodo.db.backup
cp ~/where/my/fixed/db/is/located/mycodo.db ./mycodo.db
```

3. Upgrade the database:

```
../env/bin/alembic upgrade head
```

4. Initialize for good measure, then start the daemon and frontend:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize
sudo service mycodo start
sudo service nginx start
sudo service mycodoflask start
```

That should get your fixed database upgraded and working with the 5.5.4 version. Let me know if that's not what you were asking about.

kizniche, 2018-01-04:
Otherwise, to view/edit my databases outside Mycodo, I use [SQLiteStudio](https://sqlitestudio.pl/index.rvt?act=download).

kizniche, 2018-01-04:
You'll need to look at the graph, sensor, etc. tables for each set of configurations.

pukkita, 2018-01-05:
Worked like a charm (transferring old database on new install).

Thousand thanks again! Impressive support, software companies should take note!