kizniche/Mycodo #363: Add ability to turn LDC backlight off

kizniche, 2017-12-26:
Ref. #346

> I would like to make a RFE - the ability to turn off the LCD backlight. I want to use the assembly in a grow area with light and dark cycles. The dark cycles must not have light emitted from the LCD display during those periods. Display updates can continue, however the backlight must be off.

Places to add:

1. Conditional Action
2. ?

Theoi-Meteoroi, 2017-12-26:
Conditional action seems the right place in the framework.

kizniche, 2017-12-26:
I haven't tested this latest commit because I'm not with my development Pi and LCD screens, but I should be able to test later tonight.

closed 2017-12-30 by kizniche

Theoi-Meteoroi, 2018-01-10:
Tested this function on 5.5.7 - Works very well. Thank you!  I'm using the TSL2561 sensor to trigger the conditional.

kizniche, 2018-01-10:
Thanks for reporting back. I wasn't able to turn the back light off and keep the text, as you suggested. All I was able to do was turn the light and text off. If you have played around with the LCD code and was able to do this, that would be nice to amend my code to do such. But as it is, I think I worked out all the bugs that came with changing the code, like rare gibberish output or corrupted text, or text offset on the lines.

kizniche, 2018-01-26:
I just realized, after looking at the back of my LCD, that the I2C backpack has a jumper on two pins that complete the circuit to the backlight. If you wanted to keep the display on, and have the backlight off, you can remove this jumper or connect it in series with a relay connected to Mycodo to control.

[image]

Theoi-Meteoroi, 2018-01-27:
I had considered that option - but I found the LCD API allowed it to be turned off and your solution makes it work with only 4 wires.   Most of my i2c devices terminate with a 4-pin JST connector so plug right into the 9545 mux board. That board lets me run the LCD and AM2315 ports at +5v 

Are you considering deprecating the function? I’ve been busy working a contract so I haven’t been keeping up with your source stream.
I looked at some of the diffs when you put the feature in.

I’ve been thinking about the hardware side of building these and I’ve compiled a notebook of the hardware and software hacks I’ve used to connect the various sensors to Mycodo.  I think some of it might round out the Mycodo manual.  When I have a chance I’ll pull the .txt, add my stuff and submit a pull req. You can pick what you want to keep out of it.

There are several Pi-B (40-pin) HATs that have a arduino-compatible micro controller that allows some extended sensor interfacing via several interfaces. And you can host the arduino toolchain on the pi - this is how I was building my projects before I found Mycodo.   I did most of the sensor conditioning and used the USB serial connection for data. I was using the 4-pin “grove” since the cables are fairly cheap.  I’ve been building more with protoboards since the mux chips come on breakouts.

Let me know if you want to drop the conditional I2C function.
Great work you are doing on the code. I like to follow along when I have time.

Thanks!
Dave

kizniche, 2018-01-27:
I don't plan on depreciating it. I just noticed those pins today and wanted to mention it in case you overlooked it as an option and you specifically wanted to have the LCD still display text without the backlight on. I was installing one of my LCDs in the front panel of a large incubator and noticed the pins.