kizniche/Mycodo #364: Rf Not Working?

berserktron3k, 2017-12-27:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.1
 
#### Problem Description
Please list:

  - Raspberry Pi 3, 433MHz RF Transmitter
 
### Errors
  - Output device shows as unconfigured if 0 is selected.

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Clean install of Mycodo 5.5.1
 2. Install 433MHz transmitter as output device
 
### Additional Notes
I was going back and looking at the original request to add support for the 433MHz transmitter, and along the way, there's a note you made about leaving the pin unselected. Unfortunately, not specifying a pin when adding or editing the output device doesn't work, and specifying the pin doesn't seem to work either.

Hoping I'm just doing something wrong, but in any case, would appreciate your assistance. Thanks again for all your work with this.

kizniche, 2017-12-27:
I'll try to test this before the weekend, but I'll be out of town until next week if I can't. I see one error in determining the pin state (showing if it's inactive, on, or off on the output page), but other than that, the code looks good at a glance.

Just to be sure, you're specifying the WiringPi pin you have connected to your 433MHz transmitter, correct? Also, you're using the On/Off Commands that were previously-determined by the Receiver test script?

berserktron3k, 2017-12-27:
Thanks.

If I specify the pin (or any pin) it shows as Active/Inactive, but doesn
actually seem to fire the transmitter. Using the known good codes (I still
have the test code I wrote when you were still working to include this in
Mycodo).

If I specify 0, it shows as unconfigured. And if I try to leave it blank,
it doesn’t allow me to save it.

Leaving blank on the assumption that your comment back when you were first
testing (something like “leave the pin unspecified”) meant that.

Unfortunately, I originally configured it way back. Many revisions ago. Due
to the most recent upgrades you’ve been working on, I’ve had to reinstall a
few times rather than upgrade, so I can’t say with certainty how it was
originally configred.

closed 2018-06-24 by kizniche