kizniche/Mycodo #365: 504 Gateway Time-out

zsole2, 2017-12-27:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.1
 
#### Problem Description

Over previous install of v5.4.19, i did 
`sudo rm -rf Mycodo
`
 then reinstalled v5.5.1.

It stopped at 
```
 #### Creating Mycodo database if it doesn't exist
 #### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
 #### Trying again...
```
Reboot did not help.

With a fresh SD image, the setup stopped the same place, I had only waited for the second cycle of the connection (it took about 20 minutes on my PiZero).

After interrupting setup.sh, I completed it manually by finishing the last step of setting up permissions, and rebooted.
The error came back the same.
Here is my [setup.log](https://github.com/kizniche/Mycodo/files/1589043/setup.log).

kizniche, 2017-12-27:
Raspbian Jessie or Stretch?

zsole2, 2017-12-27:
The latest available,  2017-11-29-raspbian-stretch-lite

kizniche, 2017-12-27:
What's in your /var/log/nginx/error.log?

zsole2, 2017-12-27:
[nginx_error.log](https://github.com/kizniche/Mycodo/files/1589459/nginx_error.log)

kizniche, 2017-12-27:
Can you give me ssh access?

kizniche, 2017-12-27:
I haven't a clue, and would need to poke around.

kizniche, 2017-12-28:
There have been a lot of successful installs that I can see from my stats console, and I've successfully installed it on top of a fresh Raspbian install as well. I'd guess there's something wrong with your Raspbian install or similar, like corrupted files or something. I'd suggest trying on another fresh copy of Raspbian on a different SD card.

zsole2, 2017-12-28:
Thanks for the suggestions. I tried on another SD card, with a fresh install, the result is the same. Here is the [setup.log](https://github.com/kizniche/Mycodo/files/1591397/setup.log) and the [nginx_err.log](https://github.com/kizniche/Mycodo/files/1591399/nginx_err.log).
I'll be quite busy in the next few days, but et you know when I set up the SSH access (still need to figure out ho to do it the easy way).

einsfck, 2018-01-02:
I'm in the same boat as zsole2.  Raspberry Pi Zero w, fresh install of stretch lite, fresh install of Mycodo, same results of 504 Gateway Time-out.  I did have Jessie on this micro sd with an older version of Mycodo (don't remember which one), and this issue did not come up.  I can grant SSH access if you need it still.

Thanks.
-einsfck

kizniche, 2018-01-02:
SSH will be the quickest way for me to find the issue, as I don't have a Pi Zero to test with. Contact me at https://kylegabriel.com/contact

einsfck, 2018-01-02:
I seem to be having issues sending you a message on your site.  The submit button is stuck on Processing. 
 Can you verify that you've received my message?  Thanks.

kizniche, 2018-01-02:
No, I haven't received an email. I did just update to the latest version of the contact form plugin I use and tested the form, which worked for me. Try again and see if ti works now. Also make sure you disable any script-blocking browser plugins if you have any enabled. If it doesn't work, we can figure out another way to exchange the SSH credentials.

zsole2, 2018-01-02:
I had also the same issue, but now seems to be OK. Sent you the connection details.

kizniche, 2018-01-02:
Found some strange behavior. I've tested this with both the gthread and sync workers.

```
kyle@humidorpi:/home/pi/Mycodo/mycodo $ sudo /var/www/mycodo/env/bin/gunicorn \
> --workers 2 \
> --worker-class gthread \
> --threads 2 \
> --pid /var/lock/mycodoflask.pid \
> --bind unix:/var/run/mycodoflask.sock start_flask_ui:app
[2018-01-02 13:38:00 -0600] [1617] [INFO] Starting gunicorn 19.7.1
[2018-01-02 13:38:00 -0600] [1617] [INFO] Listening at: unix:/var/run/mycodoflask.sock (1617)
[2018-01-02 13:38:00 -0600] [1617] [INFO] Using worker: gthread
[2018-01-02 13:38:00 -0600] [1620] [INFO] Booting worker with pid: 1620
[2018-01-02 13:38:00 -0600] [1621] [INFO] Booting worker with pid: 1621
[2018-01-02 13:38:30 -0600] [1617] [CRITICAL] WORKER TIMEOUT (pid:1620)
[2018-01-02 13:38:30 -0600] [1617] [CRITICAL] WORKER TIMEOUT (pid:1621)
[2018-01-02 13:38:30 -0600] [1620] [INFO] Worker exiting (pid: 1620)
[2018-01-02 13:38:30 -0600] [1621] [INFO] Worker exiting (pid: 1621)
[2018-01-02 13:38:31 -0600] [1624] [INFO] Booting worker with pid: 1624
[2018-01-02 13:38:32 -0600] [1625] [INFO] Booting worker with pid: 1625
[2018-01-02 13:39:02 -0600] [1617] [CRITICAL] WORKER TIMEOUT (pid:1624)
[2018-01-02 13:39:02 -0600] [1617] [CRITICAL] WORKER TIMEOUT (pid:1625)
[2018-01-02 13:39:02 -0600] [1624] [INFO] Worker exiting (pid: 1624)
[2018-01-02 13:39:02 -0600] [1625] [INFO] Worker exiting (pid: 1625)
[2018-01-02 13:39:03 -0600] [1629] [INFO] Booting worker with pid: 1629
[2018-01-02 13:39:03 -0600] [1630] [INFO] Booting worker with pid: 1630
```

kizniche, 2018-01-02:
I have a few meetings for the next few hours but can resume working on a fix later today

kizniche, 2018-01-02:
Just about to leave but I got the server to work by setting the timeout very high (120 seconds... 15 sec is default). I'll leave a more detailed comment when I get back to testing.

kizniche, 2018-01-02:
Go ahead and check your severs (both of you). Let me know if they're working.

einsfck, 2018-01-03:
It looks like everything is working as expected.  Thanks for looking into this and fixing it!

-einsfck

zsole2, 2018-01-03:
I also confirm that the Pi is accessible.
Thanks for the fix, and wish you a happy new year!

closed 2018-01-03 by zsole2

kizniche, 2018-01-03:
You're welcome. Thanks for providing SSH access to diagnose the issue. I'll try to get a release out tonight.