kizniche/Mycodo #366: 403 Forbidden

RynFlutsch, 2017-12-27:
## Mycodo Issue Report:
  - Specific Mycodo Version:
5.5.1
Raspberry Pi 3
 
#### Problem Description
Please list:
Fresh instal @ Stretch lite, could not connect to local host @ mycodo installation. But finished without errors.

First visit of the webinterface resulted in 403 forbidden.
nginx error.log
`2017/12/27 21:17:37 [crit] 761#761: *1 connect() to unix:/var/www/mycodo/mycodo/mycodoflask_socket.sock failed (13: Permission denied) while connecting to upstream, client: 192.168.178.25, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/www/mycodo/mycodo/mycodoflask_socket.sock:/", host: "192.168.178.20"
2017/12/27 21:17:37 [error] 761#761: *1 open() "/var/www/mycodo/mycodo/mycodo_flask/templates/502.html" failed (13: Permission denied), client: 192.168.178.25, server: , request: "GET / HTTP/1.1", upstream: "http://unix:/var/www/mycodo/mycodo/mycodoflask_socket.sock/", host: "192.168.178.20"`

kizniche, 2017-12-28:
I don't really know why it would be reporting that.

There have been a lot of successful installs that I can see from my stats console, and I've successfully installed it on top of a fresh Raspbian install as well. I'd guess there's something wrong with your Raspbian install or similar, like corrupted files or something. I'd suggest trying on another fresh copy of Raspbian on a different SD card.

RynFlutsch, 2017-12-31:
Ok now working with new instal.

closed 2017-12-31 by RynFlutsch