kizniche/Mycodo #368: Capture still -> Error 500

pekkakorre, 2018-01-03:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.3
 
#### Problem Description
Please list:

  - what were you trying to do
  - specific setup details that are involved

I was trying to take still image with MS Lifecam Cinema (work with mycodo 5.0.40). Same error every time.
Setup: RPi3, 16GB class 10 adata microSD, 1x DHT22 sensor, 1x DHT11 sensor, 4x relay, MS Lifecam Cinema
 
### Errors
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/page_routes.py", line 136, in page_camera
    camera_record('photo', mod_camera, flash=flash)
TypeError: camera_record() got an unexpected keyword argument 'flash'

kizniche, 2018-01-03:
Thanks. Seems like it will be a simple fix. I'll be out all day but will try to get a fix pushed tonight.

kizniche, 2018-01-03:
I just tested and verified the last commit fixed camera issue.

closed 2018-01-04 by kizniche