kizniche/Mycodo #369: pH reading from i2c is not working

alonrab, 2018-01-03:
Hi, Thank you so much for this incredible tool! will use it to study livestock wastewater remidiation. looking forward to share this work!
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.3.3
 
#### Problem Description
after fresh installation added atlas scientific pH meter over i2c. the adress is showing as 0x63 but no reading is take. when i tested the connection with python code i got a reading.
I'm using Pi3 B, atlas EZO i2c and GPIO pins 3 ( SDA.1),5 (SCL.1)

see: 
I2C Bus 1: i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- 63 -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

the data shows:

ATLAS_PH_I2C Input (2)
Input (ATLAS_PH_I2C), 15.0 second interval
Measurement | Timestamp
0 pH | No Data Last 1800m

and setup is  
[image]

see reading:
[image]

finally the older code i used (from hydropi)

import io  # used to create file streams
import fcntl  # used to access I2C parameters like addresses

import time  # used for sleep delay and timestamps

class atlas_i2c:
    long_timeout = 1.5  # the timeout needed to query readings and
                        #calibrations
    short_timeout = .5  # timeout for regular commands
    default_bus = 1  # the default bus for I2C on the newer Raspberry Pis,
                     # certain older boards use bus 0
    default_address = 99  # the default address for the pH sensor

    def __init__(self, address=default_address, bus=default_bus):
        # open two file streams, one for reading and one for writing
        # the specific I2C channel is selected with bus
        # it is usually 1, except for older revisions where its 0
        # wb and rb indicate binary read and write
        self.file_read = io.open("/dev/i2c-" + str(bus), "rb", buffering=0)
        self.file_write = io.open("/dev/i2c-" + str(bus), "wb", buffering=0)

        # initializes I2C to either a user specified or default address
        self.set_i2c_address(address)

    def set_i2c_address(self, addr):
        # set the I2C communications to the slave specified by the address
        # The commands for I2C dev using the ioctl functions are specified in
        # the i2c-dev.h file from i2c-tools
        I2C_SLAVE = 0x703
        fcntl.ioctl(self.file_read, I2C_SLAVE, addr)
        fcntl.ioctl(self.file_write, I2C_SLAVE, addr)

    def write(self, string):
        # appends the null character and sends the string over I2C
        string += "\00"
        self.file_write.write(bytes(string, 'UTF-8'))

    def read(self, num_of_bytes=31):
        # reads a specified number of bytes from I2C,
        # then parses and displays the result
        res = self.file_read.read(num_of_bytes)  # read from the board
        # remove the null characters to get the response
        response = [x for x in res if x != '\x00']
        if response[0] == 1:  # if the response isnt an error
            # change MSB to 0 for all received characters except the first
            # and get a list of characters
            char_list = [chr(x & ~0x80) for x in list(response[1:])]
            # NOTE: having to change the MSB to 0 is a glitch in the
            # raspberry pi, and you shouldn't have to do this!
            # convert the char list to a string and returns it
            return "Command succeeded " + ''.join(char_list)
        else:
            return "Error " + str(response[0])

    def query(self, string):
        # write a command to the board, wait the correct timeout,
        # and read the response
        self.write(string)

        # the read and calibration commands require a longer timeout
        if((string.upper().startswith("R")) or
           (string.upper().startswith("CAL"))):
            time.sleep(self.long_timeout)
        elif((string.upper().startswith("SLEEP"))):
            return "sleep mode"
        else:
            time.sleep(self.short_timeout)

        return self.read()

    def close(self):
        self.file_read.close()
        self.file_write.close()

def main():
    device = atlas_i2c()  # creates the I2C port object,
                          #specify the address or bus if necessary
    print(">> Atlas Scientific sample code")
    print(">> Any commands entered are passed to the board via I2C except:")
    print(">> Address,xx changes the I2C address the Raspberry Pi "
         "communicates with.")
    print(">> Poll,xx.x command continuously polls the board every "
          "xx.x seconds")
    print(" where xx.x is longer than the {} second timeout.".
              format(atlas_i2c.long_timeout))
    print(" Pressing ctrl-c will stop the polling")

    # main loop
    while True:
        myinput = input("Enter command: ")

        # address command lets you change which address
        # the Raspberry Pi will poll
        if(myinput.upper().startswith("ADDRESS")):
            addr = int(myinput.split(',')[1])
            device.set_i2c_address(addr)
            print("I2C address set to " + str(addr))

        # contiuous polling command automatically polls the board
        elif(myinput.upper().startswith("POLL")):
            delaytime = float(myinput.split(',')[1])

            # check for polling time being too short,
            # change it to the minimum timeout if too short
            if(delaytime < atlas_i2c.long_timeout):
                print("Polling time is shorter than timeout, "
                      "setting polling time to {}".
                          format(atlas_i2c.long_timeout))
                delaytime = atlas_i2c.long_timeout

            # get the information of the board you're polling
            info = device.query("I").split(",")[1]
            print("Polling {} sensor every {} seconds, press ctrl-c "
                  "to stop polling".
                      format(info, delaytime))

            try:
                while True:
                    print(device.query("R"))
                    time.sleep(delaytime - atlas_i2c.long_timeout)
            except KeyboardInterrupt:
                # catches the ctrl-c command, which breaks the loop above
                print("Continuous polling stopped")

        # if not a special keyword, pass commands straight to board
        else:
            try:
                print(device.query(myinput))
            except IOError:
                print("Query failed")

if __name__ == '__main__':
    main()

Thanks for any intake you have on this!

Alon

alonrab, 2018-01-03:
wrong pic for the setting, sorry.
[image]

kizniche, 2018-01-03:
Hopefully there's an error in the daemon log. Please check and copy any errors here. You can access it from the web interface by selecting the Gear Icon in the top right, then Mycodo Logs, then Daemon Log.

kizniche, 2018-01-03:
I don't have this sensor, so I won't be able to perform the final test of the new code to see if it works, but what I can do is try to activate the input device without the sensor installed and see what errors occur. I just did that and came across this error:

```
2018-01-03 17:11:27,244 - mycodo.input_15 - INFO - Activated in 90.9 ms
2018-01-03 17:11:27,245 - mycodo.inputs.atlas_ph - ERROR - 1
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/inputs/atlas_ph.py", line 75, in get_measurement
    if ',' in self.sensor_sel.calibrate_sensor_measure:
TypeError: argument of type 'NoneType' is not iterable
2018-01-03 17:11:31,564 - mycodo.input_15 - INFO - Deactivated in 68.7 ms
```

This is easily fixable. I'll see if anything else weird happens after I fix that.

Just to note, the pH I2C code I use is an adaptation of the code you mention, above. Check out [Mycodo/mycodo/devices/atlas_scientific_i2c.py](https://github.com/kizniche/Mycodo/blob/d2641331e4801d53d526602f5dcffca5712413ea/mycodo/devices/atlas_scientific_i2c.py) to see how I use it. However, I don't think there's going to be an issue with this code. Rather, The sensor module is probably where we're going to find the issues, [Mycodo/mycodo/inputs/atlas_ph.py](https://github.com/kizniche/Mycodo/blob/d2641331e4801d53d526602f5dcffca5712413ea/mycodo/inputs/atlas_ph.py).

kizniche, 2018-01-03:
Could you test a fix and see if it allows the sensor to be read?

Change line 75 of atlas_ph.py from:

```python
            if ',' in self.sensor_sel.calibrate_sensor_measure:
```

to

```python
            if (self.sensor_sel.calibrate_sensor_measure and
                    ',' in self.sensor_sel.calibrate_sensor_measure):
```

Save the file, restart the mycodo daemon with the command:

```bash
sudo service mycodo restart
```

Then see if the sensor is working or any errors appeared in the daemon log.

alonrab, 2018-01-04:
Hi, Thanks for the these update and the in-depth insight. I will try these tomorrow morning when i'm back in the lab and update what i found, hopefully this is right. 
Thanks again!

kizniche, 2018-01-04:
I released v5.5.4 tonight that incorporates this fix. You can upgrade from the web interface and see if it allows your pH measurements to be acquired.

alonrab, 2018-01-05:
Hi, did some testing with new version. still no read. firs thing I did was look at Mycodo/mycodo/devices/atlas_scientific_i2c.py ,  i thing that the i2c address in line 20 should be default_address = 99  instead of 98 (for 0x63), however -still not taking a reading trying to see why

kizniche, 2018-01-05:
That default address is a fallback for the class. The Input Controller will specify the address set in the web interface, which it forces to be set before allowing input activation. So, you don't have to worry about changing the hard-coded default address, just make sure the correct address is set in the input config.

Are there any errors in the daemon log?

alonrab, 2018-01-05:
think i messed up the mycodo installation on my side. will do a fresh install and post the error report

alonrab, 2018-01-05:
2018-01-05 14:46:40,592 - mycodo.input_1 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2018-01-05 14:46:42,599 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:46:44,606 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:46:45,489 - mycodo.input_1 - INFO - Deactivated in 32.9 ms
2018-01-05 14:47:14,906 - mycodo.input_2 - INFO - Activated in 96.7 ms
2018-01-05 14:47:30,403 - mycodo.input_1 - INFO - Activated in 109.8 ms
2018-01-05 14:47:30,406 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:32,413 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:34,419 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:34,420 - mycodo.input_1 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2018-01-05 14:47:36,425 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:38,432 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:40,438 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:40,439 - mycodo.input_1 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2018-01-05 14:47:42,445 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:44,460 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:46,466 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:46,467 - mycodo.input_1 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2018-01-05 14:47:48,472 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:50,478 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:52,484 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:52,485 - mycodo.input_1 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2018-01-05 14:47:54,498 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:56,505 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:58,512 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:47:58,512 - mycodo.input_1 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2018-01-05 14:48:00,519 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:48:02,525 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:48:04,531 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 14:48:04,532 - mycodo.input_1 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.

kizniche, 2018-01-05:
Let's find out exactly where that byte error is occurring. Change [line 108 of ](https://github.com/kizniche/Mycodo/blob/5cb87e0e6bb759d51c256376d4cf9c7afc913008/mycodo/devices/atlas_scientific_i2c.py#L108) from:

```python
            self.logger.debug(
```

to

```python
            self.logger.exception(
```

Save, then restart the daemon with ```sudo service mycodo restart```

alonrab, 2018-01-05:
sorry for the delayed response. I had some lab issues to look into. I'll follow up on testing soon. 
Thanks again,
Alon

alonrab, 2018-01-05:
added below the error report after applying change, will continue looking into this tomorrow
Thanks,
TypeError: a bytes-like object is required, not 'str'
2018-01-05 23:14:13,590 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: a bytes-like object is required, not 'str'
2018-01-05 23:14:13,591 - mycodo.input_2 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2018-01-05 23:14:28,625 - mycodo.device.atlas_scientific_i2c_99 - ERROR - AtlasScientificI2C raised an exception when taking a reading: a bytes-like object is required, not 'str'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/devices/atlas_scientific_i2c.py", line 90, in query
    self.write(query_str)
  File "/var/mycodo-root/mycodo/devices/atlas_scientific_i2c.py", line 55, in write
    self.file_write.write(cmd)

kizniche, 2018-01-06:
Try changing [line 55 of  Mycodo/mycodo/devices/atlas_scientific_i2c.py](https://github.com/kizniche/Mycodo/blob/0d98af00d59078ba28e3c484b6ad7a4f1f0de564/mycodo/devices/atlas_scientific_i2c.py#L55) from:

```python
        self.file_write.write(cmd)
```

to

```python
        self.file_write.write(cmd.encode())
```

alonrab, 2018-01-08:
After changing input with encode - both atlas pH and temp (T1000) take reading. this is indicated by device LED light blinking "green". the measurement is not showing on deamon

error log : 
2018-01-08 11:44:51,381 - mycodo.inputs.atlas_ph - ERROR - Sensor read unsuccessful: 'filter' object is not subscriptable
2018-01-08 11:44:51,381 - mycodo.input_2 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2018-01-08 11:44:51,385 - mycodo.inputs.atlas_pt1000_1_102 - ERROR - Sensor read unsuccessful: 'filter' object is not subscriptable
2018-01-08 11:44:51,386 - mycodo.input_3 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.

kizniche, 2018-01-08:
We're having to port the code that was designed for Python 2 to work with Python 3, so we're going to make incremental successes until everything works. Good thing is we're making progress.

Another good thing is my employer decreed today a snow day, so I have the day off. I actually have the Atlas PT1000 temperature sensor/probe. I will bring that out to hook up to my dev Pi so I may try to get all the bugs fixed, instead of going back and forth with you for each test. I'll met you know how it goes.

alonrab, 2018-01-08:
awsome!

kizniche, 2018-01-08:
I was able to fix the communication issues with the Atlas I2C module to be able to read my PT1000. I just made a release with the fixes. Let me know if this also works for your pH sensor, after you perform an upgrade.

alonrab, 2018-01-08:
Ok, doing that

alonrab, 2018-01-08:
Tested and working for atlas pH and temp sensors,
Many thanks!

kizniche, 2018-01-08:
Great! Glad to help. Thanks for the bug report.

closed 2018-01-08 by kizniche