kizniche/Mycodo #370: DHT11 doesn't show result on web

r1rider01, 2018-01-04:
## Mycodo Issue Report:
  - Specific Mycodo Version:  5.5.4
 
#### Problem Description
Please list:  DHT11 doesn't display anything, deamon log shows error

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

ERROR: 2018-01-04 09:02:06,949 - mycodo.inputs.dht11_4 - ERROR - Could not initialize sensor. Check if gpiod is running. Error: 'NoneType' object has no attribute 'send'

 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

Running the manual test 
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ sudo python test_gpio_DHTx.py -g 3 DHT11

works

pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ sudo python test_gpio_DHTx.py -g 3 DHT11
Temperature: 17.0
Humidity: 48.0
Temperature: 17.0
Humidity: 46.0
Temperature: 17.0
Humidity: 53.0
Temperature: 17.0
Humidity: 53.0
Temperature: 17.0
Humidity: 53.0
Temperature: 17.0
Humidity: 53.0
Temperature: 17.0
Humidity: 53.0
Temperature: 17.0
Humidity: 53.0
Temperature: 17.0
Humidity: 53.0
Temperature: 17.0
Humidity: 53.0

kizniche, 2018-01-04:
Can you verify if gpiod is running? execute ```ps aux | grep gpiod```

r1rider01, 2018-01-04:
I know its running, because the test works just fine.  Here is output:

pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ ps aux | grep gpiod
pi        2462  0.0  0.0   4372   556 pts/0    S+   09:16   0:00 grep --color=auto gpiod

kizniche, 2018-01-04:
The test script is much older than the current DHT sensor module in Mycodo, and does not use gpiod. Your ```ps``` command output indicates gpiod is not running.

What is returned when you execute ```sudo pigpiod```?

This is what you should see if gpiod is running:

```
pi@delta:~ $ ps aux|grep gpiod
root       477 21.6  0.1  20440  1524 ?        SLsl Jan03 170:17 /usr/local/bin/pigpiod -s 1
pi       26022  0.0  0.0   4380   560 pts/1    S+   10:16   0:00 grep --color=auto gpiod
```

r1rider01, 2018-01-04:
sudo pigpiod, returned nothing, but then when running
grep gpiod i now get

pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ ps aux | grep gpiod
root      2561  6.5  0.1   9896  1508 ?        SLsl 09:28   0:03 pigpiod
pi        2588  0.0  0.0   4372   528 pts/0    S+   09:28   0:00 grep --color=auto gpiod

r1rider01, 2018-01-04:
Ok now getting this error:

2018-01-04 09:31:46,234 - mycodo.input_4 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.

kizniche, 2018-01-04:
can you open your cron file and paste the contents? Open it with ```sudo crontab -e```

r1rider01, 2018-01-04:
# cmID:mycodo #
@reboot /usr/local/bin/pigpiod -s 1 &
# cm~mycodo #
# cmID:restart_daemon #
@reboot /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/scripts/restart_daemon.py --continuouscheck &
# cm~restart_daemon #

r1rider01, 2018-01-04:
`# cmID:mycodo #
@reboot /usr/local/bin/pigpiod -s 1 &
# cm~mycodo #
# cmID:restart_daemon #
@reboot /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/scripts/restart_daemon.py --continuouscheck &
# cm~restart_daemon #
`

kizniche, 2018-01-04:
Can you download this file: https://github.com/joan2937/pigpio/blob/master/EXAMPLES/Python/DHT11_SENSOR/dht11.py

Then edit line 157 so it is your DHT11 data pin, instead of "4", then run it and let me know if it works?

r1rider01, 2018-01-04:
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ python dht11.py
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/pigpio.py", line 1134, in run
    cb.func(cb.gpio, newLevel, tick)
  File "dht11.py", line 69, in either_edge_callback
    handler(tick, diff)
  File "dht11.py", line 91, in _edge_RISE
    raise
TypeError: exceptions must be old-style classes or derived from BaseException, not NoneType

temperature: 17
humidity: 57
temperature: 17
humidity: 57
temperature: 17
humidity: 57
temperature: 17
humidity: 57

kizniche, 2018-01-04:
Is that what happened? It threw an exception, but then output temp/humidity data? That's weird.

r1rider01, 2018-01-04:
correct, that is copy/paste

kizniche, 2018-01-04:
Can you run it with python 3 from the Mycodo virtualenv to test if it works with python 3?

```
sudo ~/Mycodo/env/bin/python dht11.py
```

r1rider01, 2018-01-04:
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ sudo ~/Mycodo/env/bin/python dht11.py
Traceback (most recent call last):
  File "dht11.py", line 158, in <module>
    for d in sensor:
TypeError: iter() returned non-iterator of type 'DHT11'
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $

kizniche, 2018-01-04:
Let me see if I can get it working with python 3 (what Mycodo now runs with as of v5.5.0).

r1rider01, 2018-01-04:
Can i downgrade mycodo?  or just wait?

kizniche, 2018-01-04:
To what version?

r1rider01, 2018-01-04:
i can wait, not a rush really.   eta on update?

kizniche, 2018-01-04:
I'll try to get something for you to test in the next hour.

r1rider01, 2018-01-04:
Kyle i appreciate your dedication and quick responses!

kizniche, 2018-01-04:
Are you able and willing to give me SSH access to your Pi to perform tests? It will be much quicker for me to just upload a test script and perform tweaks until I can get it working, than to go back-and-forth here with test scripts. Though, I understand if it's a difficulty or a security risk to let a stranger have access to your Pi.

kizniche, 2018-01-04:
I think I got rid of that exception. Change "next(self)" on line 143 to "__next__(self)"

This is the same code I used in Mycodo, so I'm looking for what can be causing the issue.

kizniche, 2018-01-04:
Before we get deep into changing code, try deactivating your sensor, then activating it again, in Mycodo.

r1rider01, 2018-01-04:
I would allow SSH but my firewall doesn't allow me to open ports.  sorry.

Ive deactivated and re activated non stop even rebooted, changed GPIO ports, etc..

r1rider01, 2018-01-04:
your line 143 change is the same thing....  next(self)   what am i changing?

kizniche, 2018-01-04:
The line 143 change is for the test script I mentioned, at https://github.com/joan2937/pigpio/blob/master/EXAMPLES/Python/DHT11_SENSOR/dht11.py

kizniche, 2018-01-04:
Just to confirm, since we started pigpiod and confirmed it was running, you have deactivated the DHT11 sensor and activated it again, in Mycodo?

r1rider01, 2018-01-04:
correct

r1rider01, 2018-01-04:
i see line 143   but it already says  next(self)    what does it need to change to?

kizniche, 2018-01-04:
Change the next() in the test script to include the underscores, as it is [here](https://github.com/kizniche/Mycodo/issues/370#issuecomment-355329919)

r1rider01, 2018-01-04:
Kyle i must be missing something this is what i see in your comment:

I think I got rid of that exception. Change "next(self)" on line 143 to "next(self)"

I don't see any _ underscores

kizniche, 2018-01-04:
Refresh your browser. You are seeing an old comment.

r1rider01, 2018-01-04:
DOH

r1rider01, 2018-01-04:
ok even after that change:

pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ sudo ~/Mycodo/env/bin/python dht11.py
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.5/dist-packages/pigpio.py", line 1134, in run
    cb.func(cb.gpio, newLevel, tick)
  File "dht11.py", line 69, in either_edge_callback
    handler(tick, diff)
  File "dht11.py", line 91, in _edge_RISE
    raise
RuntimeError: No active exception to reraise

temperature: 17
humidity: 68
temperature: 17
humidity: 68
temperature: 17
humidity: 68

kizniche, 2018-01-04:
Okay. Change line 91 from:

```python
                    raise
```

to

```python
                    raise Exception
```

This should get rid of the exceptions. I'm still struggling to find any discernible differences from the test code and the Mycodo code.

kizniche, 2018-01-04:
What pin is your DHT11 on? I'm making a test script to try.

r1rider01, 2018-01-04:
```
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ sudo ~/Mycodo/env/bin/python dht11.py
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.5/dist-packages/pigpio.py", line 1134, in run
    cb.func(cb.gpio, newLevel, tick)
  File "dht11.py", line 69, in either_edge_callback
    handler(tick, diff)
  File "dht11.py", line 91, in _edge_RISE
    raise Exception
Exception

temperature: 17
humidity: 53
temperature: 17
humidity: 53
temperature: 17
humidity: 53
```

r1rider01, 2018-01-04:
PIN 3

kizniche, 2018-01-04:
[dht11_test01.txt](https://github.com/kizniche/Mycodo/files/1604087/dht11_test01.txt)

Rename this txt file to .py, then run with:

~/Mycodo/env/bin/python3 dht11_test01.py

r1rider01, 2018-01-04:
```
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ sudo ~/Mycodo/env/bin/python dht11_test01.py
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.5/dist-packages/pigpio.py", line 1134, in run
    cb.func(cb.gpio, newLevel, tick)
  File "dht11_test01.py", line 225, in either_edge_callback
    handler(tick, diff)
  File "dht11_test01.py", line 250, in _edge_rise
    self.temp_humidity = (self.temp_humidity << 1) + val
TypeError: unsupported operand type(s) for <<: 'NoneType' and 'int'

Traceback (most recent call last):
  File "dht11_test01.py", line 287, in <module>
    print(test.next())
  File "dht11_test01.py", line 80, in next
    raise StopIteration  # required
StopIteration
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $
```

r1rider01, 2018-01-04:
sorry didn't run with python3   here is python3

```

pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ sudo ~/Mycodo/env/bin/python3 dht11_test01.py
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.5/dist-packages/pigpio.py", line 1134, in run
    cb.func(cb.gpio, newLevel, tick)
  File "dht11_test01.py", line 225, in either_edge_callback
    handler(tick, diff)
  File "dht11_test01.py", line 250, in _edge_rise
    self.temp_humidity = (self.temp_humidity << 1) + val
TypeError: unsupported operand type(s) for <<: 'NoneType' and 'int'

Traceback (most recent call last):
  File "dht11_test01.py", line 287, in <module>
    print(test.next())
  File "dht11_test01.py", line 80, in next
    raise StopIteration  # required
StopIteration
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $
```

kizniche, 2018-01-04:
They're the same executable, so it shouldn't matter, as long as it's one of the python executables in the env/bin directory.

Try this: [dht11_test02.txt](https://github.com/kizniche/Mycodo/files/1604135/dht11_test02.txt)

kizniche, 2018-01-04:
I think we're close. I can feel it's going to work very soon.

r1rider01, 2018-01-04:
```
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ sudo ~/Mycodo/env/bin/python3 dht11_test02.py             {'temperature': 0.0, 'humidity': 0.0, 'dewpoint': 0.0}
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $
```

kizniche, 2018-01-04:
Try this: [dht11_test03.txt](https://github.com/kizniche/Mycodo/files/1604161/dht11_test03.txt)

r1rider01, 2018-01-04:
```
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ sudo ~/Mycodo/env/bin/python3 dht11_test03.py             
Exception raised when taking a reading: Bad Checksum.
{'dewpoint': 0.0, 'humidity': 60.0, 'temperature': 17.0}
pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $
```

kizniche, 2018-01-04:
Great! Found the issue. Let me put together the code you can replace Mycodo/mycodo/inputs/dht11.py with and see if it will work within Mycodo.

kizniche, 2018-01-04:
Replace your Mycodo/mycodo/inputs/dht11.py with the contents of [dht11.txt](https://github.com/kizniche/Mycodo/files/1604177/dht11.txt), then restart the daemon with ```sudo service mycodo restart``` and see if measurements start coming in.

kizniche, 2018-01-04:
You may also need to change line 263 from:

```python
                    self.logger.exception(
```

to

```python
                    self.logger.debug(
```

r1rider01, 2018-01-04:
:-( 

```
2018-01-04 11:45:18,356 - mycodo.input_4 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
```

r1rider01, 2018-01-04:
oh wait the GPIO thing

r1rider01, 2018-01-04:
nope thats still running:

```
pi@raspberrypi:~/Mycodo/mycodo/inputs $ ps aux | grep gpiod
root      2561  6.2  0.1  12456  1528 ?        SLsl 09:28   8:39 pigpiod
pi        3863  0.0  0.0   4372   568 pts/0    S+   11:46   0:00 grep --color=auto gpiod
```

kizniche, 2018-01-04:
Did you change the logger line as well?

r1rider01, 2018-01-04:
yes

r1rider01, 2018-01-04:
pretty sure i can't run the file like this, but just incase?

```
pi@raspberrypi:~/Mycodo/mycodo/inputs $ sudo ~/Mycodo/env/bin/python3 dht11.py
Traceback (most recent call last):
  File "dht11.py", line 5, in <module>
    from mycodo.databases.models import Output
ImportError: No module named 'mycodo'
pi@raspberrypi:~/Mycodo/mycodo/inputs $
```

kizniche, 2018-01-04:
no, there are dependencies in the module that requires it be run from the mycodo daemon. The previous test scripts I modified to run independently. Give me a min to find any potential issues with my edits that didn't translate well from the test script to the mycodo module.

kizniche, 2018-01-04:
Can you modify the next() function in dht11.py, line 87 and beyond, so it includes a logging line, like this:

```python
    def next(self):
        """ Get next measurement reading """
        if self.read():  # raised an error
            self.logger.error("TEST01: {}, {}, {}".format(self._dew_point, self._humidity, self._temperature))
            raise StopIteration  # required
        return dict(dewpoint=float('{0:.2f}'.format(self._dew_point)),
                    humidity=float("{0:.2f}".format(float(self._humidity))),
                    temperature=float("{0:.2f}".format(float(self._temperature))))
```

kizniche, 2018-01-04:
Make sure you restart the mycodo daemon for it to take effect, after saving, using ```sudo service mydodo restart```

kizniche, 2018-01-04:
Also, deactivate and reactivate the sensor after the daemon has fully restarted.

r1rider01, 2018-01-04:
weird now getting:

```
pi@raspberrypi:~/Mycodo/mycodo/inputs $ sudo service mycodo restart
Job for mycodo.service failed because the control process exited with error code.
See "systemctl status mycodo.service" and "journalctl -xe" for details.
```

[34 more replies]

closed 2018-01-04 by kizniche