kizniche/Mycodo #371: Feature Request

r1rider01, 2018-01-04:
Feature 1:
Allow for PULSE on the relay's (on then quick off)  I use relays tied to motherboards to reboot machines.  so just need a quick contact.

Feature 2: 
Allow for a timed press on relays (on for X secs then off)  Same as above, but this would hold the power button down to do a force Off on the PC not just a reboot

Feature 3:
User accounts
  - Ability to allow user(s) to see or use individual inputs/outputs  (I'd like to use this to allow a manager or end user able to reboot a certain machine they control, not all machines in my list)

kizniche, 2018-01-04:
Can you elaborate how 1 and 2 are different. Also, there is already the ability to turn a relay on for x seconds.

As for user accounts, there is the ability to create users and assign restrictive roles. See Configure -> Users

r1rider01, 2018-01-04:
so on 1, it would just be a buttons says PULSE or QUICK   and its set to run for just .5sec.
2 i see the option, just hoping for a perm button, like the on off so don't have to dig into the output and type in the time each time want to use it.

for the user accounts, i see how you can allow them access to pages, but i would like more granular.
EX:
 - If i had 6 outputs setup
  -- output 1 controls  machine 1
 -- output 2 controls machine 2
 -- output 3 controls machine 3
etc...etc...

I want to give User 1 the ability to control output 1,2,3   but not 4,5,6
I want to give User 2 the ability to control output 2,3,6   but not 1,4,5
etc.. etc..

same for the data/info screen...

kizniche, 2018-01-04:
Thinking of how this could be implemented, it appears it would be lot of work. The user role system in place only affects whole systems, so you would give all or nothing access to outputs. I'm trying to figure out an elegant way to incorporate something like this, but coming up blank. I'll think about it some more and see if I can think of a way.

kizniche, 2018-01-05:
I'm still not sure how this would be integrated into Mycodo, but I did have an idea for how you could implement this locally. Note that because upgrades overwrite edits, this would have to be reinserted after any upgrade you perform. However, this can be easily done if you save the code you insert. Edit Mycodo/mycodo/mycodo_flask/general_routes.py, starting at line 379:

```python
@blueprint.route('/output_mod/<output_id>/<state>/<out_type>/<amount>')
@flask_login.login_required
def output_mod(output_id, state, out_type, amount):
    """Manipulate output"""
    if not utils_general.user_has_permission('edit_controllers'):
        return 'Insufficient user permissions to manipulate outputs'

    if (flask_login.current_user.name == 'sally' and
            output_id in ['1', '2', '3']):
        pass  # Allow sally to manipulate only outputs 1, 2, and 3
    elif (flask_login.current_user.name == 'billy' and
            output_id in ['1', '3', '6']):
        pass  # Allow billy to manipulate only outputs 1, 3, and 6
    elif (flask_login.current_user.name == 'jimmy' and
            output_id in ['4', '4', '5']):
        pass  # Allow billy to manipulate only outputs 2, 4, and 5
    else:
        return 'Insufficient user permissions to manipulate output'

    daemon = DaemonControl()
    if (state in ['on', 'off'] and out_type == 'sec' and
            (str_is_float(amount) and float(amount) >= 0)):
        return daemon.output_on_off(int(output_id), state, float(amount))
    elif (state == 'on' and out_type == 'pwm' and
              (str_is_float(amount) and float(amount) >= 0)):
        return daemon.relay_on(int(output_id), state, duty_cycle=float(amount))
```

kizniche, 2018-01-05:
Though, a user could still set up things like conditionals that manipulate outputs, or set up PIDs that manipulate outputs, and timers, etc. so it doesn't completely stop their use, only when manipulated from the Output page. Something that covers all use cases, like I mentioned, would be difficult to implement.

kizniche, 2018-01-05:
You could put the user in a role that restricts editing controllers, then use the restricting code above, but remove the initial role check. This way, the users cannot edit anything and can only manipulate the specified outputs. Here's an example (notice it's the same code, but with the first 3 lines removed:

```python
@blueprint.route('/output_mod/<output_id>/<state>/<out_type>/<amount>')
@flask_login.login_required
def output_mod(output_id, state, out_type, amount):
    """Manipulate output"""
    # Below is restrictive code that will only allow certain outputs to be
    # manipulated by certain users
    if (flask_login.current_user.name == 'sally' and
            output_id in ['1', '2', '3']):
        pass  # Allow sally to manipulate only outputs 1, 2, and 3
    elif (flask_login.current_user.name == 'billy' and
            output_id in ['1', '3', '6']):
        pass  # Allow billy to manipulate only outputs 1, 3, and 6
    elif (flask_login.current_user.name == 'jimmy' and
            output_id in ['4', '5', '7']):
        pass  # Allow jimmy to manipulate only outputs 4, 5, and 7
    else:
        return 'Insufficient user permissions to manipulate output'

    daemon = DaemonControl()
    if (state in ['on', 'off'] and out_type == 'sec' and
            (str_is_float(amount) and float(amount) >= 0)):
        return daemon.output_on_off(int(output_id), state, float(amount))
    elif (state == 'on' and out_type == 'pwm' and
              (str_is_float(amount) and float(amount) >= 0)):
        return daemon.relay_on(int(output_id), state, duty_cycle=float(amount))
```

r1rider01, 2018-01-05:
Very good.   I’ll take a look and test

Keep this in mind thou if you see something down the road or even another script you might be able to implement into this (might be a little rework of the overall code)

kizniche, 2018-01-05:
Furthermore, if you want to create custom buttons on the output page, just add code to [line 167 of  Mycodo/mycodo/mycodo_flask/templates/pages/output.html](https://github.com/kizniche/Mycodo/blob/5cb87e0e6bb759d51c256376d4cf9c7afc913008/mycodo/mycodo_flask/templates/pages/output.html#L167):

```html
  {% if current_user.name == 'sally' %}
  <form method="post" action="/output">
    <input class="form-control" id="sec_on_amt_1" name="sec_on_amt_1" type="hidden" value="0.5">
    <input class="form-control btn btn-default sec_on_amt" id="turn_on" name="1/on/sec/" type="button" value="Turn 1 On 0.5 Seconds">
  </form>
  {% endif %}
<div style="clear: both; padding: 1em 0;"></div>
```

This will add a button that will only appear for user 'sally'. If you couple this with the above code addition to output_mod(), you can ensure sally can only manipulate output 1 and she is the only user who sees the button to turn output 1 on for 0.5 seconds.

r1rider01, 2018-01-05:
whats best way to restart flask to rebuild the templates?  i've been doing PI reboots, but thats a pain for each small change

kizniche, 2018-01-05:
I actually added [the ability to do that](https://github.com/kizniche/Mycodo/blob/master/CHANGELOG.md) form the web UI (Config -> Restart Frontend) last night. I just haven't made a release. Let me make that release now so you can start testing. Make sure you save any code edits you made before upgrading. There will be a backup made in /var/Mycodo-backups but it's better to have all your edits organized beforehand. Just a suggestion.

kizniche, 2018-01-05:
But you can also issue the command ```sudo service mycodoflask restart```

kizniche, 2018-01-05:
Just made v5.5.6 release. It should be available to upgrade from the web UI.

r1rider01, 2018-01-08:
So i'm trying to do something a bit different then what you have laid out for the time being....this granular control will come into play later:

<input class="form-control" id="sec_on_amt_1" name="sec_on_amt_1" type="hidden" value="0.5">
    <input class="form-control btn btn-default sec_on_amt" id="turn_on" name="1/on/sec/" type="button" value="Turn 1 On 0.5 Seconds">

in this code, i changed the 1/on/sec/    to 2/on/sec   etc etc, but it only controlled relay 1  no others.    do i need to change the   sec_on_amt_1   ....2   ....3   etc?

kizniche, 2018-01-08:
Yes, for output 2, this would be the code:

```html
  <form method="post" action="/output">
    <input class="form-control" id="sec_on_amt_2" name="sec_on_amt_2" type="hidden" value="0.5">
    <input class="form-control btn btn-default sec_on_amt" id="turn_on" name="2/on/sec/" type="button" value="Turn 2 On 0.5 Seconds">
  </form>
```

closed 2018-02-26 by kizniche