kizniche/Mycodo #373: camera live feed?

r1rider01, 2018-01-05:
anyway to get the camera to display an image or even play like a 10FPS stream on the info/live measurement screen?

Kind of hoping to make a dashboard type display, that stays up all the time and just auto refreshes TEMP settings and live feed camera

r1rider01, 2018-01-08:
Kyle think of way to do this?

kizniche, 2018-01-08:
The live page is just meant to be a quick way to view data. It's currently unconfigurable.

A more appropriate place to allow the camera stream or a still image (captured at intervals) would probably be the graph page. Though, adding a camera viewing option would probably necessitate it being renamed to "Dashboard"

r1rider01, 2018-01-08:
or what about making another tab called "Dashboard" that we can configure with graphs, cameras, etc..?

kizniche, 2018-01-08:
I would just add the ability to put cameras on the graph page, and rename Live Graphs to Dashboard. I do like the idea of adding camera captures, though I'll have to test the long-term streaming capability, because users commonly have the graph page open for long durations.

kizniche, 2018-01-10:
Initial working iteration of the dashboard with camera:

[image]

r1rider01, 2018-01-10:
Looking good!

[screenshot-192 168 0 10-2018-01-09-19-01-08]<https://user-images.githubusercontent.com/838427/34749464-98dd8164-f56f-11e7-9d95-50e81f0ddb99.png>

zsole2, 2018-01-11:
Dashboard sounds good!
I agree that the Live page is for only quickly checking data. I only use it when adding new components, or want to check quickly remotely if Mycodo is running. Otherwise, I always in live graphs. And sometimes check the historical data on async graphs.
So...can't you integrate these together into a single interface? I could imagine having the info/dashboard page with expandable sections. First it comes up with the live page style, simple raw data. If you expand it, it could show  the graphs or other detailed view. And for graphs, I even could live with a popup window (usually I hate those) for async graphs, if there is a button for it on the graphs settings.

closed 2018-02-26 by kizniche