kizniche/Mycodo #374: Suggestion: Wonderful Cable Terminal Blocks (WAGO)

pukkita, 2018-01-05:
Just wanted to share a nice find for the junction/project boxes and Relay installs: WAGO connectors:

[image]

They're single polarity terminal blocks, i.e. for a typical electricity wiring (neutral, phase, earth) three terminal blocks would be needed, with proper number of wire "bays".

E.g. If we wanted to join two devices to electricity, we will need the three cable "bay" ones.

They're IP66, sporting a nice touch: a probe point, and has international certifications for electricity installs.

They're available via Amazon for cheap, a 50 terminal box (3 cable bays) goes for about 14€ in Europe (0.28€/ea).

r1rider01, 2018-01-05:
those looks nice!

pukkita, 2018-01-05:
They are :) and moreso, really safe and versatile!

zsole2, 2018-01-05:
Yes, they are nice, I have a few, I use them for quick a dirty prototyping with wires of almost any diameter. The only problem that they are a little bit bulky. The Wago 221 series seems the same, but smaller. Next time I go for those. 
On the other hand, if you work with solid core wires and household currents, and no need for remove the wires, try Wago 2273 series, I like those very much, though the wire cannot be removed (but never slips out, too).

closed 2020-07-23 by kizniche