kizniche/Mycodo #375: picamera error and no image

r1rider01, 2018-01-05:
So I hit "Start Stream" and got the "no image" icon....

Went to hit "Stop Stream" or even "Capture Still" and get the below message:
*I have rebooted
*I did restart deamon 

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/page_routes.py", line 132, in page_camera
    lib=mod_camera.library)).Camera
  File "/home/pi/Mycodo/env/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 673, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/pi/Mycodo/mycodo/mycodo_flask/camera/camera_picamera.py", line 8, in <module>
    from base_camera import BaseCamera
ImportError: No module named 'base_camera'

r1rider01, 2018-01-05:
Let me add to this:

I deleted the camera from config....added the camera back
Hit the capture still button, got an image, hit the button again get this message:

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/page_routes.py", line 136, in page_camera
    camera_record('photo', mod_camera)
  File "/home/pi/Mycodo/mycodo/devices/camera.py", line 64, in camera_record
    with picamera.PiCamera() as camera:
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/picamera/camera.py", line 433, in __init__
    self._init_preview()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/picamera/camera.py", line 513, in _init_preview
    self, self._camera.outputs[self.CAMERA_PREVIEW_PORT])
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/picamera/renderers.py", line 558, in __init__
    self.renderer.inputs[0].connect(source).enable()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/picamera/mmalobj.py", line 2212, in enable
    prefix="Failed to enable connection")
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to enable connection: Out of resources

kizniche, 2018-01-05:
I fixed this issue last night, but you won't see it until I make a release. I'll make a release you can upgrade to later today.

closed 2018-01-08 by kizniche