kizniche/Mycodo #376: Gauge refresh rate resets to 'None'

zsole2, 2018-01-07:
Specific Mycodo Version: 
 **5.5.6**

Add new gauge, set basic parameters. After the gauge comes back, edit, and set color ranges. At this times, the refresh rate is shown as expected. After save, the gauge shows 0, and checking the details, the refresh rate shows 'None'.

kizniche, 2018-01-08:
I can't reproduce this issue. Can you post a screenshot of your gauge settings?

kizniche, 2018-01-08:
I was just able to reproduce this issue. Looking into fixing it.

kizniche, 2018-01-08:
Thanks for the bug report. I made a fix and will make a release soon. In the meantime, you can save the value as long as you use an integer (not a float). So, just remove the period (".") and the "0" to allow it to save (e.g. "40" instead of "40.0").

closed 2018-01-08 by kizniche