kizniche/Mycodo #377: Remote add error

r1rider01, 2018-01-08:
Trying to add a second device to see/control.

When i do i get the following error:
Error: Error: Remote Host Add: [Errno 2] No such file or directory: '/home/pi/Mycodo/mycodo/../mycodo/mycodo_flask/ssl_certs/remote_admin/192.168.112.101_cert.pem'

Both are on local network.   I'm guessing since I don't have a cert for the SSL that is why i see this?  got steps to setup self cert?

RPi-Growbox, 2018-02-13:
Okay, have same issue. 
What I did for it to work:

> cd /home/Mycodo/mycodo/mycodo_flask/ssl_certs/
> mkdir remote_admin
 -> Add remote host, it make that _cert.pem -file..

Now I can add remote, but status show red? I'm 100% sure it's online and working. Also added new admin user for remote use. Going to Input-page give error 500 and here's traceback:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_remote_admin.py", line 62, in remote_input
    host_inputs[each_host.host] = json.loads(host_inputs[each_host.host])
  File "/usr/lib/python3.5/json/__init__.py", line 312, in loads
    s.__class__.__name__))
TypeError: the JSON object must be str, not 'NoneType'

kizniche, 2018-02-14:
This was one of many issues that appeared after moving from Python 2,7 to 3.5. Keep in mind the Remote Admin interface is very new and very experimental. Therefore I won't usually be responding to issues with it. I did however fix the issues with adding and viewing new hosts. Adding to this area of Mycodo will be very slow.

closed 2018-02-14 by kizniche