kizniche/Mycodo #378: Inquiry

berserktron3k, 2018-01-08:
Thanks again for all your work on this. My RF has been flaky lately, so I'm thinking it's time to move to a relayed switch.

I found these: https://www.adafruit.com/product/2935?gclid=Cj0KCQiAyszSBRDJARIsAHAqQ4r-lxWW4_--e508ut7Us5aPENZComJsOFwMW8qr_x5NLTyqGiG7HTEaAojvEALw_wcB

Pro: Already built.
Con: Switch controls all 3 switched outlets at once.

What I'm curious about is this:
1. If you used a switch relay (generally) it should provide feedback to Mycodo to tell us if it is powered on or not, correct? I ask because with the wireless, I'm finding they are becoming less reliable. For example, fans get turned on, but sometimes they won't turn off, even though Mycodo has told them to turn off.
2. If Mycodo is able to determine the state, would the switch I linked to above be readable by Mycodo?

Not looking for you to spend money buying and testing one of these, just looking for your educated guess.

Thanks in advance.

kizniche, 2018-01-09:
There's no feedback used with switched relays. It's assumed that the external hardware will interpret the GPIO signal (high or low) and act appropriately. You can always build in precautions to detect when a relay doesn't work properly. Such as, if you're regulating temperature and the temperature suddenly falls below a certain amount, have a Conditional notify you by email, or deactivate a PID controller, execute a command, etc.

kizniche, 2018-01-09:
I could add an option to wireless relays titled "Repeat Command" that would repeat the command, either on or off, a number of times. For instance, if set to 5, whenever the wireless relay were turned off, it would send 5 off signals in order to increase the chances of the signal actually being received properly. Let me know what you think and if you're willing to test.

berserktron3k, 2018-01-09:
I like the idea of repeating the command, however I'm not in a position
right now to be able to test it, unfortunately. I might be able to in a
month or so.

Brady Hummel

fry42, 2018-03-08:
Hi!

I think this would be a great option!
I've done this before on projects using the cheap 433MHz outlets... Maybe also add a delay value - like x repeats y seconds apart.

btw. phantastic peace of software!!!

closed 2018-04-06 by kizniche