kizniche/Mycodo #379: Error 502: Bad Gateway Error

skeuneblav, 2018-01-10:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 
#### Problem Description
Please list:

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

skeuneblav, 2018-01-10:
[setup.log](https://github.com/kizniche/Mycodo/files/1620613/setup.log)
[setup.log](https://github.com/kizniche/Mycodo/files/1620622/setup.log)

kizniche, 2018-01-10:
Thanks. I'll try to look at this later tonight. I'm going to be busy though for the next few days and this weekend traveling doing field work, so I may not be able to get to this until next week.

Is there the possibility of allowing SSH access to your pi and sending me authentication credentials to kylegabriel.com/contact ? If so, please make a strong password to prevent brute force password guessing attempts if anyone comes across your Pi IP.

skeuneblav, 2018-01-10:
Thanks, thats a fast reaction,

Unfortunately it is not possible for me to create SSH access to my pi form outside this network.
If you need additional information please let me know.

I wish you save traveling in the next few days.

Best Regards,

Martin Jan Ringnalda

From: Kyle Gabriel
Sent: 10 January 2018 23:21
To: kizniche/Mycodo
Cc: skeuneblav; Manual
Subject: Re: [kizniche/Mycodo] Error 502: Bad Gateway Error (#379)

Thanks. I'll try to look at this later tonight. I'm going to be busy though for the next few days and this weekend traveling doing field work, so I may not be able to get to this until next week.
Is there the possibility of allowing to your pi and sending me authentication credentials to kylegabriel.com/contact ? If so, please make a strong password to prevent brute force password guessing attempts if anyone comes across your Pi IP.

zsole2, 2018-01-11:
The setup.log looks like mine on the PiZero. 
When investigating the same issue, it looks finally the limiting the thread count for mycodoflask can be a solution. See #349. Here is my /Mycodo/install/mycodoflask.service:
```
[Service]
User=root
Group=mycodo
WorkingDirectory=/var/mycodo-root/mycodo
ExecStart=/var/mycodo-root/env/bin/gunicorn \
--workers 1 \
--timeout 900 \
--pid /var/lock/mycodoflask.pid \
--bind unix:/var/run/mycodoflask.sock start_flask_ui:app
```
I can access the GUI with this after a while. However, the daemon still not starts on reboot. The timeout above does not affect the daemon startup, apparently. After the server is up for a time, then the daemon can be started, though. The processor usage is very high for several components, right now I'm investigating this. We already know the influxd could easily take 5 mins to load on the Zero, this can be  a problem too.

skeuneblav, 2018-01-11:
> Should probably set the daemon not to start until influxdb has started,
> and the front end not to start until both of them have started.

In a few hours I will take a look, I am working at the moment.
The 2 raspberry's u use are raspberry 3 versions I don't use the zero.

Regards jan

kizniche, 2018-01-11:
It's probably not your issue if they're Pi3s. I'll post a command for you to run in a bit that you can execute to potentially determine what the issue is.

zsole2, 2018-01-11:
Good news on my end with the Zero:
Putting `TimeoutSec=infinity` into mycodo.service solved the reboot issue. Infinity seems an overkill, but is very handy, since mycodo needed 3m35s to start...
I'll try to upgrade and see how it goes with the default settings on all other services which were modified several times...

zsole2, 2018-01-11:
OK, after upgrade, editing mycodo.service, reboot was OK, except for mycodoflask.service, it timed out. Started to edit the settings (timeoutsec, forking), as discussed in #349, but could not restart it, either timed out or stuck. Finally I reset everything, except set the timeout value to 900, and rebooted. And here it is, everything is up and running!
mycodo started up in 3m31s, mycodoflask in 2m17s.

My remaining concern is CPU utilization, everything in  v5.5 uses more CPU, the worst is pigpiod, well over 20%, whereas on v5.0.17, it is about 8%. It seems that influxd also uses more CPU,  it fluctuates widely, but on the old version, rarely goes above 20%, while on v5.5 never seems to go below 20%, but jumps up to 50-70%.

kizniche, 2018-01-11:
@skeuneblav, issue the following commands and let me know if there are any errors that result:

```bash
sudo service mycodoflask stop
cd ~/Mycodo
sudo /var/mycodo-root/env/bin/gunicorn --workers 1 --worker-class gthread --pid /var/lock/mycodoflask.pid --bind unix:/var/run/mycodoflask.sock start_flask_ui:app
```

@zsole2, so what exactly needs to be changed, and what do you think would be a safe timeout to set for each service?

skeuneblav, 2018-01-11:
Hi,

It still does not work, in the annex the errorlog.

Regards,

Jan

From: Kyle Gabriel
Sent: 11 January 2018 20:00
To: kizniche/Mycodo
Cc: skeuneblav; Mention
Subject: Re: [kizniche/Mycodo] Error 502: Bad Gateway Error (#379)

@skeuneblav, issue the following commands and let me know if there are any errors that result:

@zsole2, so what exactly needs to be changed, and what do you think would be a safe timeout to set for each service?
—

pi@raspberrypi:~/Mycodo $ sudo /var/mycodo-root/env/bin/gunicorn --workers 1 --w                                                   orker-class gthread --pid /var/lock/mycodoflask.pid --bind unix:/var/run/mycodof                                                   lask.sock start_flask_ui:app
[2018-01-11 20:22:35 +0100] [1986] [INFO] Starting gunicorn 19.7.1
[2018-01-11 20:22:35 +0100] [1986] [INFO] Listening at: unix:/var/run/mycodoflas                                                   k.sock (1986)
[2018-01-11 20:22:35 +0100] [1986] [INFO] Using worker: gthread
[2018-01-11 20:22:35 +0100] [1989] [INFO] Booting worker with pid: 1989
[2018-01-11 20:22:35 +0100] [1989] [ERROR] Exception in worker process
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/arbiter.py", li                                                   ne 578, in spawn_worker
    worker.init_process()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/gthread                                                   .py", line 109, in init_process
    super(ThreadWorker, self).init_process()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/base.py                                                   ", line 126, in init_process
    self.load_wsgi()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/workers/base.py                                                   ", line 135, in load_wsgi
    self.wsgi = self.app.wsgi()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/app/base.py", l                                                   ine 67, in wsgi
    self.callable = self.load()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/app/wsgiapp.py"                                                   , line 65, in load
    return self.load_wsgiapp()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/app/wsgiapp.py"                                                   , line 52, in load_wsgiapp
    return util.import_app(self.app_uri)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/gunicorn/util.py", line                                                    352, in import_app
    __import__(module)
ImportError: No module named 'start_flask_ui'
[2018-01-11 20:22:35 +0100] [1989] [INFO] Worker exiting (pid: 1989)
[2018-01-11 20:22:35 +0100] [1986] [INFO] Shutting down: Master
[2018-01-11 20:22:35 +0100] [1986] [INFO] Reason: Worker failed to boot.

kizniche, 2018-01-11:
You must be in ~/Mycodo/mycodo to execute the command. I amended my initial comment for correctness moments after I posted it.

zsole2, 2018-01-11:
I think for the 502 error on the Zero, the issue can be solved by `TimeoutSec=infinity` on mycodo.service.  I don't think that playing around with number of seconds necessary.
For the 504 error, mycodoflask.service starts up quickly, but the workers need the extra time, thus the timeout parameter needs to be set up high, but I cannot be sure how high. As a mentioned above, 2m16s was needed in this particular restart, so even the 120 secs tried in #349 would not be enough. I'd put 180 secs to be safe, or even more. I'll restart my test Zero a few more times...

kizniche, 2018-01-11:
Can you open ~/Mycodo/mycodo/config.py and on line 12, read what MYCODO_VERSION is set to?

skeuneblav, 2018-01-11:
MYCODO_VERSION = `5.5.7`

Regards,

jan

From: Kyle Gabriel
Sent: 11 January 2018 20:40
To: kizniche/Mycodo
Cc: skeuneblav; Mention
Subject: Re: [kizniche/Mycodo] Error 502: Bad Gateway Error (#379)

Can you open ~/Mycodo/mycodo/config.py and on line 12, read what MYCODO_VERSION is set to?

zsole2, 2018-01-11:
Reboot resulted the startup times as follows:
mycodo: 3m44s
mycodoflask: 2m25s

Here are my working mycodo.service and mycodoflask.service files:

```
[Unit]
Description=Mycodo server
After=syslog.target network.target influxd.service

[Service]
Type=forking
TimeoutSec=infinity
ExecStart=/var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py
ExecStop=/var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_client.py -t
ExecReload=/bin/kill -HUP $MAINPID

[Install]
WantedBy=multi-user.target
Alias=mycodo.service
```

```
[Unit]
Description=gunicorn daemon for Mycodo flask web user interface
After=network.target

[Service]
User=root
Group=mycodo
WorkingDirectory=/var/mycodo-root/mycodo
ExecStart=/var/mycodo-root/env/bin/gunicorn \
--workers 2 \
--worker-class gthread \
--threads 2 \
--timeout 900 \
--pid /var/lock/mycodoflask.pid \
--bind unix:/var/run/mycodoflask.sock start_flask_ui:app

ExecReload=/bin/kill -s HUP $MAINPID
ExecStop=/bin/kill -s TERM $MAINPID

[Install]
WantedBy=multi-user.target
```

kizniche, 2018-01-11:
900 is a long timeout time. 300 sounds like it would be sufficient and still be a safe amount. See any problem with that?

zsole2, 2018-01-11:
No, that's OK with me.

kizniche, 2018-01-11:
I just released v5.5.8 that incorporates fixes for both the installation problem in this issue, and the fixes for the Pi Zeros.

@skeuneblav, issue the following command to upgrade, then see if everything works:

```bash
sudo mycodo-commands upgrade
```

skeuneblav, 2018-01-11:
Thanks Kyle, it worked.

Regards,

Jan Ringnalda

rpcjacobs, 2018-02-02:
Was helping out a friend of mine, who is running v5.5.16 on raspbian jessie with a Pi Zero, and noticed a dmesg error concerning the applied fix

```
TimeoutSec=infinity
```

> systemd[1]: [/home/pi/Mycodo/install/mycodo.service:7] Failed to parse usec_t value, ignoring: infinity

It seems if you are using a version of systemd older than 229, [you will need to use 0 instead of infinity to disable the timeout](https://stackoverflow.com/a/48168113/1683164) (jessie is currently at 215).

> root@raspberrypi:/home/pi# systemd --version
systemd 215

Fixed it by changing https://github.com/kizniche/Mycodo/blob/d78af592ccdb24bd48b3afe35fb4f48c9c6a687a/install/mycodo.service#L7 to;

```
TimeoutSec=0
```

to complete the fix, reload the changed service and start it;
```
systemctl daemon-reload
systemctl start mycodo.service
```

Now the daemon starts properly, without issue's;
```
root@raspberrypi:/home/pi# systemctl status mycodo
● mycodo.service - Mycodo server
   Loaded: loaded (/home/pi/Mycodo/install/mycodo.service; enabled)
   Active: active (running) since Fdg 2018-02-02 02:41:11 UTC; 3s ago
  Process: 1037 ExecStart=/var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py (code=exited, status=0/SUCCESS)
 Main PID: 1051 (python)
   CGroup: /system.slice/mycodo.service
           └─1051 /var/mycodo-root/env/bin/python /var/mycodo-root/mycodo/mycodo_daemon.py

Feb 02 02:41:11 raspberrypi systemd[1]: Started Mycodo server.
Feb 02 02:41:11 raspberrypi mycodo_daemon[1051]: Starting daemon.
```
:wink:

kizniche, 2018-02-02:
Thanks! I'll apply this to the next release.

closed 2018-02-03 by kizniche