kizniche/Mycodo #380: Output usage -> Error 500

pekkakorre, 2018-01-13:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.6
 
#### Problem Description
Going Output usage -> error 500
 
I changed voltage 120 -> 230, amperage 15 -> 10 and $ -> € in configuration before error, but I dont think that matters..

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Go to Output usage
 2. Error 500
 
 
### Errors

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/page_routes.py", line 1233, in page_usage
    output_stats = return_relay_usage(misc, output)
  File "/home/pi/Mycodo/mycodo/utils/tools.py", line 107, in return_relay_usage
    past_1d_hours = relay_sec_on(each_output.id, 86400) / 3600
  File "/home/pi/Mycodo/mycodo/utils/influx.py", line 172, in relay_sec_on
    output_time_on = control.relay_sec_currently_on(output_id)
  File "/home/pi/Mycodo/mycodo/mycodo_client.py", line 121, in relay_sec_currently_on
    return self.rpyc_client.root.relay_sec_currently_on(relay_id)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/rpyc/core/netref.py", line 199, in __call__
    return syncreq(_self, consts.HANDLE_CALL, args, kwargs)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/rpyc/core/netref.py", line 72, in syncreq
    return conn.sync_request(handler, oid, *args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/rpyc/core/protocol.py", line 523, in sync_request
    raise obj
_get_exception_class.<locals>.Derived: 'OutputController' object has no attribute 'relay_sec_currently_on'

========= Remote Traceback (1) =========
Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.5/site-packages/rpyc/core/protocol.py", line 347, in _dispatch_request
    res = self._HANDLERS[handler](self, *args)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/rpyc/core/protocol.py", line 624, in _handle_call
    return self._local_objects[oid](*args, **dict(kwargs))
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 216, in exposed_relay_sec_currently_on
    return mycodo.controller['Output'].relay_sec_currently_on(relay_id)
AttributeError: 'OutputController' object has no attribute 'relay_sec_currently_on'

kizniche, 2018-01-13:
Thanks for the bug report. I'll fix this when I can get to my development system next.

kizniche, 2018-01-15:
I just released v5.5.9 that incorporates a fix for this issue.

closed 2018-01-15 by kizniche

pekkakorre, 2018-01-15:
Thank you! You doing good job with this :)