kizniche/Mycodo #381: EDGE conditional cannot be set

zsole2, 2018-01-13:
## Mycodo Issue Report:
  - Specific Mycodo Version:
5.5.8
 
#### Problem Description
I set up the conditional as this screenshot shows:

[image]

Hitting Save gives 

> Error: Mod Conditional: If the GPIO State radio button is selected, GPIO State must be set

This error did not depend on whether the edge detection was selected and configured beforehand, or not. Neither f the Rising state was deleted, contrary to what is shown on the screenshot. 

### Additional Notes
I was able to set it on a single try, when changed the period value, and the the help bubble covered the Save button. Then the first click made the bubble disappear, and the second click reached the Save button. But this phenomenon could not be repeated, though.

kizniche, 2018-01-13:
Thanks, I'll look into it.

kizniche, 2018-01-15:
I just released v5.5.9 that incorporates a fix for this issue.

closed 2018-01-15 by kizniche

zsole2, 2018-01-17:
Thanks for the fix, I am now able to save the LOW setting. On the other hand, activating still gives trouble:
`Error: Activate Conditional: If the GPIO State radio button is selected, a GPIO State must be set`

kizniche, 2018-01-17:
Sorry about that. I just pushed a fix. I'm going to work on a few more issues before making a release later today.