kizniche/Mycodo #382: Error 500 pH calibration

leoneo, 2018-01-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.9
 
#### Problem Description
Please list:
In Calibration page after Select Device give error 500
In Data page after activate ph atlas uart sensor give the Error: Could not activate Input controller with ID 4: 'loc'
 
### Errors
In live page : no data

In Calibration page :  
 Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 106, in calibration_atlas_ph
    next_stage > 1):
TypeError: unorderable types: NoneType() > int()

 
### Additional Notes
fresh install, all upgraded, pH sensor checked with Atlas sample code works

kizniche, 2018-01-17:
Can you look in the daemon log after enabling the pH sensor and see if there are any errors?

leoneo, 2018-01-17:
ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 256, in __init__
    sensor_sel=self.input_sel)
  File "/var/mycodo-root/mycodo/inputs/atlas_ph.py", line 31, in __init__
    self.initialize_sensor()
  File "/var/mycodo-root/mycodo/inputs/atlas_ph.py", line 66, in initialize_sensor
    uart=self.device_loc))
KeyError: 'loc'
2018-01-17 21:29:58,465 - mycodo.input_5 - INFO - Activated in 257.2 ms
2018-01-17 21:29:58,467 - mycodo.daemon - INFO - All activated Input controllers started
2018-01-17 21:29:58,468 - mycodo.daemon - INFO - All activated Math controllers started
2018-01-17 21:29:58,469 - mycodo.daemon - INFO - All activated PID controllers started
2018-01-17 21:30:00,627 - mycodo.lcd_1 - INFO - Activated in 2093.9 ms
2018-01-17 21:30:00,629 - mycodo.daemon - INFO - All activated LCD controllers started
2018-01-17 21:30:01,288 - mycodo.conditional - INFO - Conditional settings refreshed
2018-01-17 21:30:01,291 - mycodo.conditional - INFO - Conditional controller activated in 159.5 ms
2018-01-17 21:30:01,293 - mycodo.daemon - INFO - Mycodo daemon v5.5.9 started in 15.948 seconds
2018-01-17 21:30:01,302 - mycodo.daemon - INFO - 40.76 MB RAM in use
2018-01-17 21:31:59,069 - mycodo.daemon - WARNING - Input controller with ID 4 not found
2018-01-17 21:32:09,841 - mycodo.daemon - ERROR - Could not activate Input controller with ID 4: 'loc'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 476, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 256, in __init__
    sensor_sel=self.input_sel)
  File "/var/mycodo-root/mycodo/inputs/atlas_ph.py", line 31, in __init__
    self.initialize_sensor()
  File "/var/mycodo-root/mycodo/inputs/atlas_ph.py", line 66, in initialize_sensor
    uart=self.device_loc))
KeyError: 'loc'

kizniche, 2018-01-17:
I just pushed a fix to this issue. If you can edit [Line 65 of Mycodo/mycodo/inputs/atlas_ph.py](https://github.com/kizniche/Mycodo/blob/677fc63b4a46db4302bb99c4844da687bd05c8db/mycodo/inputs/atlas_ph.py#L65) from:

```python
                "mycodo.inputs.atlas_ph_{loc}".format(
```

to

```python
                "mycodo.inputs.atlas_ph_{uart}".format(
```

Also change [Line 106 of  Mycodo/mycodo/mycodo_flask/routes_calibration.py](https://github.com/kizniche/Mycodo/blob/677fc63b4a46db4302bb99c4844da687bd05c8db/mycodo/mycodo_flask/routes_calibration.py#L106) from:

```python
            next_stage > 1):
```

to

```python
            (next_stage is not None and next_stage > 1)):
```

Save, then click Config -> Restart Backend, and try again.

leoneo, 2018-01-17:
I had do change in files and reboot. Now I have 500 issues after clic on Clear calibration and after put the temperature and clic on Continue to next stage : 

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 122, in calibration_atlas_ph
    selected_input, 'temperature', temp, 'continuous', 1)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 208, in dual_commands_to_sensor
    atlas_command = AtlasScientificCommand(input_sel)
  File "/home/pi/Mycodo/mycodo/utils/calibration.py", line 30, in __init__
    self.board_version, self.board_info = self.board_version()
  File "/home/pi/Mycodo/mycodo/utils/calibration.py", line 44, in board_version
    info = self.ph_sensor_uart.query('i')[0]
TypeError: 'NoneType' object is not subscriptable

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_calibration.py", line 83, in calibration_atlas_ph
    atlas_command = AtlasScientificCommand(selected_input)
  File "/home/pi/Mycodo/mycodo/utils/calibration.py", line 30, in __init__
    self.board_version, self.board_info = self.board_version()
  File "/home/pi/Mycodo/mycodo/utils/calibration.py", line 44, in board_version
    info = self.ph_sensor_uart.query('i')[0]
TypeError: 'NoneType' object is not subscriptable

leoneo, 2018-01-17:
daemon 

Last 30 lines of /var/log/mycodo/mycodo.log:
  File "/var/mycodo-root/mycodo/devices/atlas_scientific_uart.py", line 109, in send_cmd
    self.ser.write(buf)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/serial/serialposix.py", line 532, in write
    d = to_bytes(data)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/serial/serialutil.py", line 63, in to_bytes
    raise TypeError('unicode strings are not supported, please encode to bytes: {!r}'.format(seq))
TypeError: unicode strings are not supported, please encode to bytes: 'R\r'
2018-01-18 00:54:03,854 - mycodo.input_4 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2018-01-18 00:54:18,892 - mycodo.device.atlas_scientific_uart_/dev/ttyAMA0 - ERROR - AtlasScientificUART raised an exception when taking a reading: unicode strings are not supported, please encode to bytes: 'R\r'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/devices/atlas_scientific_uart.py", line 65, in query
    self.send_cmd(query_str)
  File "/var/mycodo-root/mycodo/devices/atlas_scientific_uart.py", line 109, in send_cmd
    self.ser.write(buf)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/serial/serialposix.py", line 532, in write
    d = to_bytes(data)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/serial/serialutil.py", line 63, in to_bytes
    raise TypeError('unicode strings are not supported, please encode to bytes: {!r}'.format(seq))
TypeError: unicode strings are not supported, please encode to bytes: 'R\r'
2018-01-18 00:54:33,923 - mycodo.device.atlas_scientific_uart_/dev/ttyAMA0 - ERROR - AtlasScientificUART raised an exception when taking a reading: unicode strings are not supported, please encode to bytes: 'R\r'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/devices/atlas_scientific_uart.py", line 65, in query
    self.send_cmd(query_str)
  File "/var/mycodo-root/mycodo/devices/atlas_scientific_uart.py", line 109, in send_cmd
    self.ser.write(buf)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/serial/serialposix.py", line 532, in write
    d = to_bytes(data)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/serial/serialutil.py", line 63, in to_bytes
    raise TypeError('unicode strings are not supported, please encode to bytes: {!r}'.format(seq))
TypeError: unicode strings are not supported, please encode to bytes: 'R\r'

kizniche, 2018-01-17:
Add the following code after [Line 107 of ](https://github.com/kizniche/Mycodo/blob/c37e18b6f1f0f6ee6e018f0b49791939089f036f/mycodo/devices/atlas_scientific_uart.py#L107), so it looks like this, save, restart the backend, and test again:

```python
        buf = "{cmd}\r".format(cmd=cmd)  # add carriage return
        try:
```

to

```python
```python
        buf = "{cmd}\r".format(cmd=cmd)  # add carriage return
        if type(buf) is str:
            buf = buf.encode()
        try:
```
```

leoneo, 2018-01-18:
this give me error 502, all Mycodo doesn't work.

leoneo, 2018-01-18:
it is my fault, I have missing ( ) in (next_stage is not None and next_stage > 1))

So I have reinstall mycodo and do the files changes. Now i have error 504 Gateway Time-out after Clear calibration and after continue to next stage

kizniche, 2018-01-18:
Are there any errors in the Web Error or Daemon logs?

kizniche, 2018-01-18:
Is there the possibility of allowing me SSH access and a Mycodo UI login to perform testing? It's often much faster because I can edit and test much more rapidly.

leoneo, 2018-01-23:
Hi,

Sorry for the delay, we had a cyclone (Berguita)

Web error: 
2018/01/23 10:17:44 [error] 2290#2290: *161 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "POST /calibration_atlas_ph HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/calibration_atlas_ph", host: "192.168.1.26", referrer: "https://192.168.1.26/calibration_atlas_ph"
2018/01/23 10:28:18 [error] 2289#2289: *280 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "POST /data HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/data", host: "192.168.1.26", referrer: "https://192.168.1.26/data"
2018/01/23 10:28:58 [error] 2290#2290: *311 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "GET /daemonactive?_=1516275545989 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/daemonactive?_=1516275545989", host: "192.168.1.26", referrer: "https://192.168.1.26/settings/general"
2018/01/23 10:29:21 [error] 2289#2289: *280 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "POST /data HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/data", host: "192.168.1.26", referrer: "https://192.168.1.26/data"
2018/01/23 10:29:58 [error] 2290#2290: *311 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "GET /daemonactive?_=1516275545990 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/daemonactive?_=1516275545990", host: "192.168.1.26", referrer: "https://192.168.1.26/settings/general"
2018/01/23 10:30:58 [error] 2289#2289: *280 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "GET /daemonactive?_=1516275545991 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/daemonactive?_=1516275545991", host: "192.168.1.26", referrer: "https://192.168.1.26/settings/general"
2018/01/23 10:31:58 [error] 2290#2290: *311 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "GET /daemonactive?_=1516275545992 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/daemonactive?_=1516275545992", host: "192.168.1.26", referrer: "https://192.168.1.26/settings/general"
2018/01/23 10:32:18 [error] 2289#2289: *280 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "POST /data HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/data", host: "192.168.1.26", referrer: "https://192.168.1.26/data"
2018/01/23 10:32:58 [error] 2290#2290: *311 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "GET /daemonactive?_=1516275545993 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/daemonactive?_=1516275545993", host: "192.168.1.26", referrer: "https://192.168.1.26/settings/general"
2018/01/23 10:33:24 [error] 2288#2288: *429 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "GET /live HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/live", host: "192.168.1.26"
2018/01/23 10:33:59 [error] 2290#2290: *432 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "GET /daemonactive?_=1516275545994 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/daemonactive?_=1516275545994", host: "192.168.1.26", referrer: "https://192.168.1.26/settings/general"
2018/01/23 10:34:58 [error] 2288#2288: *429 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "GET /daemonactive?_=1516275545995 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/daemonactive?_=1516275545995", host: "192.168.1.26", referrer: "https://192.168.1.26/settings/general"
2018/01/23 10:35:58 [error] 2290#2290: *432 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "GET /daemonactive?_=1516275545996 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/daemonactive?_=1516275545996", host: "192.168.1.26", referrer: "https://192.168.1.26/settings/general"
2018/01/23 10:36:15 [error] 2288#2288: *429 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "GET /live HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/live", host: "192.168.1.26"
2018/01/23 10:36:58 [error] 2290#2290: *432 upstream timed out (110: Connection timed out) while reading response header from upstream, client: 192.168.1.30, server: , request: "GET /daemonactive?_=1516275545997 HTTP/1.1", upstream: "http://unix:/var/run/mycodoflask.sock/daemonactive?_=1516275545997", host: "192.168.1.26", referrer: "https://192.168.1.26/settings/general"
[... cut ...]
2018-01-23 12:34:05,402 - mycodo.lcd_1 - ERROR - IOError: Unable to output to LCD.
2018-01-23 12:35:44,642 - mycodo.lcd_1 - ERROR - IOError: Unable to output to LCD.
2018-01-23 12:36:00,023 - mycodo.lcd_1 - ERROR - Count not initialize LCD. Error: 121
2018-01-23 12:36:00,066 - mycodo.lcd_1 - ERROR - IOError: Unable to output to LCD.
2018-01-23 12:36:15,366 - mycodo.lcd_1 - ERROR - Count not initialize LCD. Error: 121
2018-01-23 12:36:15,387 - mycodo.lcd_1 - ERROR - IOError: Unable to output to LCD.

I will send you SSH and Mycodo access

kizniche, 2018-01-23:
You can contact me at https://kylegabriel.com/contact with the credentials.

leoneo, 2018-01-24:
did you receive the information?

kizniche, 2018-01-25:
I haven't. I just tested it and the message went through fine. I would suggest trying again. There should be a confirmation page that the message was sent and a confirmation email sent to the address you put in the Email field. Make sure any browser add-ons that block javascript is disabled.

leoneo, 2018-01-28:
Hi,

Have you all you need?

kizniche, 2018-01-28:
I received the credentials. Sorry for not verifying. I haven't had a chance to start testing. I'm about to make a cup of coffee and will check it out after that kicks in.

kizniche, 2018-01-28:
I will have to restart the frontend every time I make a test edit, so you may see pages loading slowly or have a gateway error. I may also have to restart the daemon. Just a heads-up.

kizniche, 2018-01-28:
I finally got measurements working. Now to start testing the calibration.

kizniche, 2018-01-28:
Success was shirt-lived. I'm not sure why, but after the sensor worked at least once, it then stopped working. I attempted to reboot the Pi but haven't been able to connect to it again.

kizniche, 2018-01-28:
I'm going to push my edits that enabled measurements and wait for you to investigate why the Pi is no longer accessible.

kizniche, 2018-01-29:
Your Pi is accessible again. And the pH measurements are working. I'm going to resume testing.

kizniche, 2018-01-29:
Progress! I'm in the middle of calibrating!

kizniche, 2018-01-29:
I finally got through the entire calibration process and fixed all the errors impeding its success. I went ahead and used the "Clear Calibration" command to remove any stored calibration. Thanks for letting me test on your system to get everything working. It would have taken weeks potentially if we merely went back-and-forth in this forum.

When you have time, go through the calibration procedure with your pH solutions (4, 7 and 10 pH), then see if the measurement responds appropriately to different pH solutions (i.e. see if it's properly calibrated). Let me know how it goes, so we can continue fixing problems or close this issue.

leoneo, 2018-01-29:
Super Kyle 
Thanks

I will check the calibration with pH solutions and let you know how it goes.

kizniche, 2018-02-14:
Have you had a chance to go through the calibration?

kizniche, 2018-11-30:
I've had several similar issues appear sine this one and I believe this has been fixed.

closed 2018-11-30 by kizniche