kizniche/Mycodo #383: Error 500 camera pi

leoneo, 2018-01-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.9
 
#### Problem Description
Please list:
pi camera doesn't work
 
### Errors
Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 456, in _init_camera
    self._camera = mo.MMALCamera()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2279, in __init__
    super(MMALCamera, self).__init__()
  File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 633, in __init__
    prefix="Failed to create MMAL component %s" % self.component_type)
  File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
    raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.ril.camera': Out of memory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 136, in page_camera
    camera_record('photo', mod_camera.unique_id)
  File "/home/pi/Mycodo/mycodo/devices/camera.py", line 72, in camera_record
    with picamera.PiCamera() as camera:
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 431, in __init__
    self._init_camera(camera_num, stereo_mode, stereo_decimate)
  File "/usr/lib/python3/dist-packages/picamera/camera.py", line 460, in _init_camera
    "Camera is not enabled. Try running 'sudo raspi-config' "
picamera.exc.PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.

### Additional Notes
camera has been enabled in raspi-config

kizniche, 2018-01-17:
What model Pi?

Have any settings been changed outside of the Mycodo install instructions after a fresh install of the latest Raspbian?

What did you do to cause the error? Please list steps.

leoneo, 2018-01-17:
I have the Pi 2 model B
I have install Raspbian, after LCD-show 3.5'' and Mycodo and Atlas scientific pH sample code

kizniche, 2018-01-17:
> What did you do to cause the error? Please list steps.

leoneo, 2018-01-17:
In cameras page when I clic on Capture still
[image]

kizniche, 2018-01-17:
Are you able to get an image with any other software on the Pi? If not, it's probably a hardware issue and not a software issue with Mycodo.

leoneo, 2018-01-17:
Do you know what software can I use please ?

kizniche, 2018-01-17:
There's [```raspistill``` and ```raspivid```](https://www.raspberrypi.org/documentation/raspbian/applications/camera.md)

```picamera``` is the other popular software, which is what Mycodo uses.

leoneo, 2018-01-18:
I try with this command : raspistill -v -o test.jpg
but Camera is not detected, she is maybe dead

leoneo, 2018-01-18:
ok i have solve the problem : the cable connection 

thank you

kizniche, 2018-01-18:
Aha! Good to hear it was easily fixable.

closed 2018-01-18 by kizniche