kizniche/Mycodo #384: No gauge after setting it up and adding

vankomicin, 2018-01-17:
## Mycodo Issue Report:
Version:5.5.9, rpi zero
 
#### Problem Description
After adding a gauge for my am2302 readouts (graphs show ok) it doesn't show up in dashboard. I don't get any error messages.
If I try to add it again or after reboot it's the same problem. It's the same in different web browsers.

### Additional Notes
PS. Is there any chance of manually setting the y axis range of graphs? That would be extremely useful.

Thanks.

kizniche, 2018-01-17:
Thanks for the bug report. I've been able to replicate the issue and am working on it. I think manual y-axis range selection would be a good feature (that's also easily implemented). I'll see about adding that with the next release.

kizniche, 2018-01-17:
This seemed to be an issue that affected adding new gauges ***and*** graphs. It was a simple fix.

closed 2018-01-17 by kizniche