kizniche/Mycodo #385: Error 500 reset LCD

leoneo, 2018-01-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.9
 
#### Problem Description
Please list:

In LCD page after Reset LCD
 
### Errors
 Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 664, in page_lcd
    utils_lcd.lcd_reset_flashing(form_lcd_mod.lcd_id.data)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_lcd.py", line 192, in lcd_reset_flashing
    return_value, return_msg = control.lcd_flash(lcd_id, False)
TypeError: 'NoneType' object is not iterable

kizniche, 2018-01-17:
Please list the steps to cause this error. It works as expected for me and I haven't been able to replicate it.

leoneo, 2018-01-17:
[image]

I have the error when I clic on Reset LCD

kizniche, 2018-01-17:
Are there errors in the daemon log?

leoneo, 2018-01-17:
Now it works, I have disconect the power and reboot because the text was not correct with default.

leoneo, 2018-01-17:
After reboot I have an other problem : no data actualised in live page. I need to Deactivate/Activate in Data page to correct the problem.
[image]

kizniche, 2018-01-17:
Did you shut down the Pi or just unplugged it? I would recommend shutting down with ```sudo shutdown now -h``` or use the option in theMycodo web UI Config Menu -> Shutdown System.

leoneo, 2018-01-17:
I have reboot the pi with the Mycodo web UI and unplugged the LCD

kizniche, 2018-01-17:
Deactivating and reactivating is sometimes an issue with some sensors. This is due to a number of factors that include how the physical sensor was designed, the communication method programmed into each sensor, and the software (Mycodo input module) used to query it. I've been wanting to implement a type of auto-reactivator function that will attempt a deactivating/reactivation of inputs to alleviate some of these initialization-type issues.

closed 2018-02-26 by kizniche