kizniche/Mycodo #387: Conditional error

zsole2, 2018-01-19:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.9
 
#### Problem Description
I found this in the daemon log:
```
2018-01-18 06:10:04,236 - mycodo.conditional - ERROR - Run Error: list index out of range
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_conditional.py", line 122, in run
    self.check_conditionals(each_cond_id)
  File "/var/mycodo-root/mycodo/controller_conditional.py", line 203, in check_conditionals
    device_measurement = cond.if_sensor_measurement.split
```

Cannot really add more info, it appears that one of my EDGE sensors does not  work.  The setup is the one you have suggested way back, #107. Will test more later.

kizniche, 2018-01-19:
Thanks. I'll look into it. What are the settings for this Conditional? Can you take a screenshot?

zsole2, 2018-01-19:
Oh wait... the one I suspected bad is the conditional I could not activate, #381. Here is the screenshot:
[image]
All of those send me e-mail, except the inactive one, which operates a relay for a few seconds.

kizniche, 2018-01-19:
I also need to add the Conditional ID to the logging line in order to be able to determine which conditional is throwing the error.

So which conditional is causing the error, if you can tell?

zsole2, 2018-01-19:
I suspected the 3rd one, since the water level is low in that container. But that one being inactive, I have no idea... The 4th one related to the 3rd, so I'd exclude it. The temp notifications should have occurred more than once, having been testing the rewired refrigerator in the last few days. So I really stumped, will try to check it tomorrow somehow.

kizniche, 2018-01-19:
I found the issue to this. It supposedly affects all Edge Conditionals. I'll see if I can make a fix and release a new version tonight.

zsole2, 2018-01-20:
So I found a clue. First time error as reported:
```
2018-01-17 20:09:54,187 - mycodo.conditional - INFO - Conditional settings refreshed
2018-01-17 20:09:54,195 - mycodo.conditional - INFO - Conditional controller activated in 1725.5 ms
2018-01-17 20:09:54,203 - mycodo.daemon - INFO - Mycodo daemon v5.5.9 started in 84.659 seconds
2018-01-17 20:09:54,232 - mycodo.daemon - INFO - 41.48 MB RAM in use
2018-01-17 20:22:45,108 - mycodo.pid_3 - INFO - Deactivated in 1124.6 ms
2018-01-17 20:23:43,343 - mycodo.pid_3 - INFO - Activated in 3430.4 ms
2018-01-18 06:10:04,236 - mycodo.conditional - ERROR - Run Error: list index out of range
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_conditional.py", line 122, in run
    self.check_conditionals(each_cond_id)
  File "/var/mycodo-root/mycodo/controller_conditional.py", line 203, in check_conditionals
    device_measurement = cond.if_sensor_measurement.split(',')[1]
IndexError: list index out of range
```

Now this morning it is repeated:
```
2018-01-19 20:05:10,472 - mycodo.conditional - INFO - Conditional settings refreshed
2018-01-19 20:05:10,484 - mycodo.conditional - INFO - Conditional controller activated in 1457.0 ms
2018-01-19 20:05:10,491 - mycodo.daemon - INFO - Mycodo daemon v5.5.9 started in 86.334 seconds
2018-01-19 20:05:10,544 - mycodo.daemon - INFO - 42.23 MB RAM in use
2018-01-19 20:05:42,243 - mycodo.pid_1 - ERROR - Failed to read measurement from the influxdb database: Could not connect.
2018-01-19 20:06:31,973 - mycodo.pid_3 - ERROR - Failed to read measurement from the influxdb database: Could not connect.
2018-01-19 20:06:42,207 - mycodo.pid_1 - ERROR - Failed to read measurement from the influxdb database: Could not connect.
2018-01-20 06:05:19,562 - mycodo.conditional - ERROR - Run Error: list index out of range
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_conditional.py", line 122, in run
    self.check_conditionals(each_cond_id)
  File "/var/mycodo-root/mycodo/controller_conditional.py", line 203, in check_conditionals
    device_measurement = cond.if_sensor_measurement.split(',')[1]
IndexError: list index out of range
```
It is about 10 hours and 10 secs after startup. I should have included all relevant information at the time of the first report, though I had no clue what could be relevant...
Based on this, the 4th conditional is the one throwing the error when accessed... since it has no business to send anything, since the container is well over the sensor's level.

kizniche, 2018-01-20:
Actually, all Edge Conditionals should have the issue and cause this error, at least that's what I see in the code. If it's only this particular edge conditional that is strange.

zsole2, 2018-01-20:
I cannot know, my other Edge conditional was the one that was not able to  activate in the other thread.

kizniche, 2018-01-20:
I'll try to make a release later today. I'm currently doing field work all this weekend and my travel laptop has been failing lately.

zsole2, 2018-01-20:
I'm in no rush, trying to set up the fridge again with new wiring, and testing the current release, I was quite behind with 5.0.x

kizniche, 2018-01-20:
I just pushed a commit that should fix this issue and made a release for v5.5.10. I haven't tested this fix. I'm fairly sure it will work, but I'm sure I'll hear soon if I've broken the conditional system.

zsole2, 2018-01-21:
I confirm that the conditional activated after the upgrade. Now I'll have to wait for the other error to appear.

zsole2, 2018-01-21:
OK, errors everywhere:
```
2018-01-21 13:53:42,701 - mycodo.conditional_2 - ERROR - Exception reading the GPIO pin
2018-01-21 13:53:42,709 - mycodo.conditional_2 - ERROR - GPIO not configured correctly or GPIO state not verified
2018-01-21 14:53:38,373 - mycodo.conditional_2 - ERROR - Exception reading the GPIO pin
2018-01-21 14:53:38,404 - mycodo.conditional_2 - ERROR - GPIO not configured correctly or GPIO state not verified
2018-01-21 15:53:42,417 - mycodo.conditional_2 - ERROR - Exception reading the GPIO pin
2018-01-21 15:53:42,427 - mycodo.conditional_2 - ERROR - GPIO not configured correctly or GPIO state not verified
2018-01-21 16:53:37,967 - mycodo.conditional_2 - ERROR - Exception reading the GPIO pin
2018-01-21 16:53:37,978 - mycodo.conditional_2 - ERROR - GPIO not configured correctly or GPIO state not verified
2018-01-21 17:53:41,848 - mycodo.conditional_2 - ERROR - Exception reading the GPIO pin
2018-01-21 17:53:41,859 - mycodo.conditional_2 - ERROR - GPIO not configured correctly or GPIO state not verified
2018-01-21 18:53:43,198 - mycodo.conditional_2 - ERROR - Exception reading the GPIO pin
2018-01-21 18:53:43,206 - mycodo.conditional_2 - ERROR - GPIO not configured correctly or GPIO state not verified
2018-01-21 18:58:41,547 - mycodo.conditional - INFO - Conditional settings refreshed
2018-01-21 18:59:49,972 - mycodo.conditional - INFO - Conditional settings refreshed
2018-01-21 19:00:20,174 - mycodo.conditional - INFO - Conditional settings refreshed
2018-01-21 19:00:43,735 - mycodo.conditional - INFO - Conditional settings refreshed
2018-01-21 19:01:05,938 - mycodo.conditional - INFO - Conditional settings refreshed
2018-01-21 20:01:22,833 - mycodo.conditional_2 - ERROR - Exception reading the GPIO pin
2018-01-21 20:01:22,844 - mycodo.conditional_2 - ERROR - GPIO not configured correctly or GPIO state not verified
```
The second conditional is for the DHT22, which apparently functions normally.

To make things more interesting, the 19:00 log entries were the result of the following actions:
- turn off conditional No. 3
- set it to edge detection, with RISING, save
- set it back to GPIO state detection, LOW, save
- activate

After this, I have received two e-mail notifications:

2018-01-21 19-21-18
[Conditional 2 (Hum_buffer_low)] Input 8 (Hum_buffer_low) Rising edge detected on pin 19 (BCM)
[Conditional Action 3]: Turn output 7 on for 3.0 seconds.
[Conditional Action 4]: Notify ...

The other two was sent at 2018-01-21 19-41-20 and 2018-01-21 20-04-41
My reset time for the edge sensors is 1200s, and it seems that after reading it, triggers a RISING edge, and carries out the conditional, which was overwritten earlier.

kizniche, 2018-01-21:
Strange, because the below error should only be able to happen with Edge Conditionals. The Measurement Conditional shouldn't even be attempting to execute this code. I haven't been able to reproduce this issue.

```
2018-01-21 18:53:43,198 - mycodo.conditional_2 - ERROR - Exception reading the GPIO pin
2018-01-21 18:53:43,206 - mycodo.conditional_2 - ERROR - GPIO not configured correctly or GPIO state not verified
```

I would recommend deleting all your conditionals and adding them again. You may have some weird database remnant that's causing the code to think it's an Edge Conditional. I've also added the ID numbers to each conditional (on the web UI page) so you can identify which conditional is actually number 2. Otherwise, you can determine this by viewing the page source. Since the number is not listed, and they can be rearranged, I suspect it's actually one of your Edge Conditionals that is 2.

kizniche, 2018-01-21:
I just released v5.5.11 that includes the code to conclusively identify your number 2 conditional.

zsole2, 2018-01-22:
The new ID numbers confirm that the conditionals were reorder. Anyway, I'll delete all of them and start from scratch.

zsole2, 2018-01-23:
After deleting and adding the conditionals, they appear correct. However, there are errors in the log:

```
22018-01-22 20:40:14,978 - mycodo.input_8 - INFO - Deactivated in 39.7 ms
2018-01-22 20:41:36,916 - mycodo.input_8 - INFO - Activated in 1486.2 ms
2018-01-22 20:41:36,924 - mycodo.input_8 - ERROR - Error: Conflicting edge detection already enabled for this GPIO channel
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_input.py", line 377, in run
    bouncetime=self.switch_bouncetime)
RuntimeError: Conflicting edge detection already enabled for this GPIO channel
2018-01-22 20:43:43,376 - mycodo.input_9 - INFO - Deactivated in 69.0 ms
2018-01-22 20:43:57,632 - mycodo.input_9 - INFO - Activated in 1269.2 ms
2018-01-22 20:44:49,884 - mycodo.input_8 - INFO - Activated in 544.1 ms
2018-01-22 21:04:40,822 - mycodo.conditional_1 - ERROR - Exception reading the GPIO pin
2018-01-22 21:04:40,834 - mycodo.conditional_1 - ERROR - GPIO not configured correctly or GPIO state not verified
```
The last two lines repeat every hour (setting is for 3600s).

After a clean reboot, it returned:
```
2018-01-23 06:13:09,715 - mycodo.daemon - INFO - Mycodo daemon v5.5.12 starting
2018-01-23 06:13:15,420 - mycodo.daemon - INFO - Anonymous statistics enabled
2018-01-23 06:13:16,040 - mycodo.daemon - INFO - Starting rpyc server
2018-01-23 06:13:21,301 - mycodo.output - INFO - Output 1 setup on pin 22 and turned OFF (OFF=HIGH)
2018-01-23 06:13:21,319 - mycodo.output - INFO - Output 3 setup on pin 24 and turned OFF (OFF=HIGH)
2018-01-23 06:13:21,333 - mycodo.output - INFO - Output 4 setup on pin 10 and turned OFF (OFF=HIGH)
2018-01-23 06:13:21,346 - mycodo.output - INFO - Output 5 setup on pin 9 and turned OFF (OFF=HIGH)
2018-01-23 06:13:21,358 - mycodo.output - INFO - Output 6 setup on pin 25 and turned OFF (OFF=HIGH)
2018-01-23 06:13:21,371 - mycodo.output - INFO - Output 7 setup on pin 11 and turned OFF (OFF=HIGH)
2018-01-23 06:13:21,384 - mycodo.output - INFO - Output 8 setup on pin 8 and turned OFF (OFF=HIGH)
2018-01-23 06:13:21,408 - mycodo.output - INFO - Output controller activated in 1423.1 ms
2018-01-23 06:13:24,716 - mycodo.timer_1 - INFO - Activated in 2281.3 ms
2018-01-23 06:13:27,914 - mycodo.timer_2 - INFO - Activated in 2009.7 ms
2018-01-23 06:13:31,147 - mycodo.timer_3 - INFO - Activated in 2073.7 ms
2018-01-23 06:13:31,162 - mycodo.daemon - INFO - All activated Timer controllers started
2018-01-23 06:13:35,426 - mycodo.input_1 - INFO - Activated in 2545.2 ms
2018-01-23 06:13:39,850 - mycodo.input_2 - INFO - Activated in 2467.0 ms
2018-01-23 06:13:43,413 - mycodo.input_4 - INFO - Activated in 2162.7 ms
2018-01-23 06:13:48,198 - mycodo.input_6 - INFO - Activated in 3432.6 ms
2018-01-23 06:13:52,582 - mycodo.input_7 - INFO - Activated in 2841.6 ms
2018-01-23 06:13:56,312 - mycodo.input_8 - INFO - Activated in 2590.4 ms
2018-01-23 06:14:00,672 - mycodo.input_9 - INFO - Activated in 3284.6 ms
2018-01-23 06:14:00,697 - mycodo.daemon - INFO - All activated Input controllers started
2018-01-23 06:14:00,714 - mycodo.daemon - INFO - All activated Math controllers started
2018-01-23 06:14:07,600 - mycodo.pid_1 - INFO - Activated in 5124.4 ms
2018-01-23 06:14:14,647 - mycodo.pid_2 - INFO - Activated in 6186.8 ms
2018-01-23 06:14:14,661 - mycodo.daemon - INFO - All activated PID controllers started
2018-01-23 06:14:33,232 - mycodo.lcd_1 - INFO - Activated in 18086.2 ms
2018-01-23 06:14:33,248 - mycodo.daemon - INFO - All activated LCD controllers started
2018-01-23 06:14:36,274 - mycodo.conditional - INFO - Conditional settings refreshed
2018-01-23 06:14:36,316 - mycodo.conditional - INFO - Conditional controller activated in 2544.7 ms
2018-01-23 06:14:36,313 - mycodo.daemon - INFO - Mycodo daemon v5.5.12 started in 86.588 seconds
2018-01-23 06:14:36,420 - mycodo.daemon - INFO - 41.58 MB RAM in use
2018-01-23 06:15:08,064 - mycodo.pid_1 - ERROR - Failed to read measurement from the influxdb database: Could not connect.
2018-01-23 06:16:08,040 - mycodo.pid_1 - ERROR - Failed to read measurement from the influxdb database: Could not connect.
2018-01-23 06:17:08,142 - mycodo.pid_1 - ERROR - Failed to read measurement from the influxdb database: Could not connect.
2018-01-23 06:18:08,412 - mycodo.pid_1 - ERROR - Failed to read measurement from the influxdb database: Could not connect.
2018-01-23 07:14:37,910 - mycodo.conditional_1 - ERROR - Exception reading the GPIO pin
2018-01-23 07:14:37,912 - mycodo.conditional_1 - ERROR - GPIO not configured correctly or GPIO state not verified
```

kizniche, 2018-01-23:
The issue is that you cannot use GPIO.add_event_detect() more than once for the same pin (at the same time). It appears you have an Edge Input and an Edge Conditional for the same pin. There really isn't a need for both, as the dedicated Conditional section now takes the place of what the Edge Input used to do (which was have conditionals associated with it).

I believe I will convert the Edge Input to a GPIO State Input, with it only storing the GPIO state. This should reduce confusion and reduce the likelihood of setting up two or more Edge detection objects with the same pin.

kizniche, 2018-01-23:
The use of Edge (as opposed to GPIO State) is that it triggers immediately when the edge is detected. With the GPIO State, your conditional can only detect the state change every Conditional Period. The Edge Conditional will trigger as soon as it detects the edge change, which is crucial if you need something to react very fast after a state change.

kizniche, 2018-01-23:
Now that I look closer at all the Edge code, I see some issues. I'm gutting it all and reinstating something more logical. Stay tuned!

kizniche, 2018-01-24:
Whew! I think I broke some stuff with the latest commit to a test branch. If I can get things stable, I'll merge it with master and release it. It's currently pretty confusing.

What I've done (I think this includes everything):

 - Converted all Edge Inputs to GPIO State Inputs
 - Deleted all Edge detection code from the Input Controller
 - Added Bouncetime as an option for Edge Conditionals and added the edge detection code to the Conditional Controller

kizniche, 2018-01-24:
Actually, what I did wasn't that logical. I'm going to scrap everything I've done except add the new GPIO State Input. That should be the best alternative.

kizniche, 2018-01-24:
I also think I figured out a fix for the "Exception reading the GPIO pin" error, and just pushed a fix, and released v5.5.13.

zsole2, 2018-01-24:
Thanks, I'll check it  as soon as I can, I'm on the road for 2 days.

zsole2, 2018-01-27:
I got this error setting up the new conditional:
`Error: Mod Conditional: (sqlite3.InterfaceError) Error binding parameter 1 - probably unsupported type. [SQL: 'UPDATE conditional SET if_sensor_period=?, if_sensor_max_age=? WHERE conditional.id = ?'] [parameters: (3600.0, Decimal('1800'), 1)]`

zsole2, 2018-01-27:
I just checked, and the log is full with this messages:
```
2018-01-27 13:29:23,365 - mycodo.inputs.gpio_state_19 - ERROR - 0
2018-01-27 13:29:27,274 - mycodo.inputs.gpio_state_26 - ERROR - 0
2018-01-27 13:49:23,418 - mycodo.inputs.gpio_state_19 - ERROR - 0
2018-01-27 13:49:27,328 - mycodo.inputs.gpio_state_26 - ERROR - 0
2018-01-27 14:09:23,365 - mycodo.inputs.gpio_state_19 - ERROR - 0
2018-01-27 14:09:27,287 - mycodo.inputs.gpio_state_26 - ERROR - 0
```

kizniche, 2018-01-27:
> the log is full with this message

Sorry about that. I left a debug logging line in there from testing. Remove this line and restart your daemon to stop that:

https://github.com/kizniche/Mycodo/blob/033c109d29e7b4bf6aab461cb14cf003190a6703/mycodo/inputs/gpio_state.py#L53

> I got this error setting up the new conditional

Can you list the steps to reproduce this issue, please?

kizniche, 2018-01-27:
I think I induced the error, or a similar one, by changing the Max Age. Investigating now.

zsole2, 2018-01-28:
Thanks, I was able to set the GPIO state inputs and conditionals up without errors. 
Unfortunately, it does not work as expected. Here is the Live page, showing the same values for a few hours now:
[image]
And this is the conditional settings:
[image]
Everything seems to be normal for me, and the actions (sending e-mails) are not performed.

kizniche, 2018-01-28:
I found a bug that would interpret a value of 0 as no value at all. I'll push a fix in a moment. So, when your state is 1 is should have worked (as my testing did), but when it's 0, it would not.

zsole2, 2018-01-29:
I did the testing, all seems to be OK, but I'll refill/empty the containers and watch them once more to actually see the expected operation. 
I really like the new foolproof upgrade system, but in these type of cases, I think it is really an overkill, especially for testing these simple bugfixes. This morning, it took about 30 mins to finish the upgrade, and this was one of the fastest ones. On the Zero,  all the recompiling and requirement checking taking place can be quite long...

kizniche, 2018-01-29:
You might want to implement a different method for testing on the Zeros, such as doing a direct pull from git.

zsole2, 2018-01-29:
I was thinking about doing so, but a little afraid screwing up the database...

zsole2, 2018-02-01:
I did a second run of the bucket-filling tests, and everything seems to be  in order.
One interesting  thing happened though: 
[image]
The conditional was to run the pump for 5 secs. First I was just curious why no pump action appears on the graph, but hovering over it I accidentally saw the strange 0.03s (and 0.02s at the other instance) value. Since no other process uses this pump, there is no reason to cut its operation short. The notification e-mail indicated 5 secs, but that simply just repeats the action, I guess. If the pump is really turned on for just 0.03s, it would be good to know. 
In the best case scenario, ultimately, the e-mail could show the real event, not just the instructions, but this would involve quite a lot of programming and cross checking of the conditional actions.

kizniche, 2018-02-01:
Can you provide all the conditionals that use ```Pump``` as an output? It would seem to me that there's something else interfering with it's ability to run for the full 5 seconds.

zsole2, 2018-02-02:
That is was surprising, the Pump is used only in this conditional, to fill up the other water container. But I'll check later, when I can access the Pi, I'm having connection issues right now that are -troubleshooted currently.

kizniche, 2018-02-02:
I think I left a flaw in the Output code that could allow for inaccurate on duration timing. I'm investigating now. I'm fairly sure I can fix it and prevent erroneous durations.

kizniche, 2018-02-02:
Nevermind, the code is set up in a way that prevents turning off a duration short. I still suspect you may have a conflict that is turning it off. If not, I'd suggest running the daemon is debug mode to see if you can catch what's happening with the increased logging.

zsole2, 2018-02-02:
I'm quite sure that Pump is only involved in filling the container, unless accidentally entered in another context. Although I'd know from the second conditional when  the refill container is empty.
The shortened period is not a problem, the pump is activated every hour until the level of the humidifier container is low.
 BTW, is there an easy way to see  what component (input, output, whatever) is used and where?

kizniche, 2018-02-03:
You have to either go to each page of the UI or look at the database.

Could you possibly send me your mycodo.db for me to inspect?

zsole2, 2018-02-03:
Here is my [backup](https://github.com/kizniche/Mycodo/files/1691814/Mycodo_5.5.16_Settings_532644870ad6_raspberrypi_2018-02-02_15-50-15.zip)

kizniche, 2018-02-04:
I couldn't find anything out of the ordinary with your database. If you happen to run the daemon in debug mode and can isolate the part of the daemon log when the issue arises, we may be able to figure out what's going on.

closed 2018-02-04 by kizniche

zsole2, 2018-02-04:
I'm not sure that I want to run the daemon in debug mode for this issue. The pump is activated  only once in every few days, and if it does not fill up the container to the sensor's level, it repeats in an hour. So the error corrects itself automatically. Unless this appears with another conditional/relay, I think it can be left as it is.