kizniche/Mycodo #389: Howto?: linux output command

r1rider01, 2018-01-20:
I want to create an input for a ping check against a server.  If that server doesn't respond either the first time or after 5 times, i want to trigger an output.

What do I need to do in the Data:Input section to capture the ping result?

kizniche, 2018-01-22:
I was thinking about how to do this, and decided it would be easier to just integrate this as an Input. I actually made two inputs from this idea, Server Ping, which will ping a server and return 1 if it's up and 0 if it's down, and Server Port Open, which will check if a port is open and return 1 if it's open and 0 if it's closed.

To do what you desire, add a Server Ping Input and set the "Times to Check" option to 5. You can set the "Deadline" to set how long each check will wait before it fails. The default Deadline of 2 seconds should suffice. Next, add a Measurement Conditional on the Function page, and set the Measurement to the Server Ping Input, State to Less Than, and Value to 1 (0 is less than 1). Add an Output Conditional Action and configure it to your liking.

Let me know if there are any issues setting it up or operating it.

Thanks for the good idea!

kizniche, 2018-01-22:
BTW, I just released v5.5.12 that includes these new Inputs.

r1rider01, 2018-01-23:
Thanks again Kyle!

r1rider01, 2018-01-23:
So i have the data configured, the function set for less than 1, but can't figure out how to set what happens when the conitional is less than 1.

kizniche, 2018-01-23:
Expand the Conditional by clicking the plus sign (+) in the top left of the Conditional box, then select Output as the action, and click Add Action. Then configure the output to your liking.

r1rider01, 2018-01-23:
hmm... the ping doesn't seem to work. 

I can ping fine from a windows machine, and from the pi, they both return a valid ping...but mycodo is returning 0 still.

kizniche, 2018-01-23:
What is the response when you ping your host from your Pi with the following command (replace HOST with your host?

```bash
ping -c 1 -w 2 HOST
```

This is the command used in the (Server ping Input](https://github.com/kizniche/Mycodo/blob/ffd01af73fc858803a7c13f88666bb120b973f46/mycodo/inputs/server_ping.py#L57).

r1rider01, 2018-01-23:
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 0.523/0.523/0.523/0.000 ms

kizniche, 2018-01-23:
That should return a success. I'm not sure what's wrong. Make sure your host in Mycodo is spelled correctly and there aren't any stray spaces or other characters.

r1rider01, 2018-01-23:
i didn't type a host name i did IP in mycodo

r1rider01, 2018-01-23:
even tried different IP  same thing

kizniche, 2018-01-23:
Try google.com

r1rider01, 2018-01-23:
still 0

r1rider01, 2018-01-23:
ok, i see whats going on.  i had the times to check set at 5, and was causing 0  changed times to check to 1 and now its working

r1rider01, 2018-01-23:
changing the times to check value above 2 causes false reading....

r1rider01, 2018-01-23:
Here are my settings....  

I have it set times to check to 3 so it fails, condition if less than 1 to turn output off for .5sec, but nothing its not turning off that output at all.

https://imgur.com/a/ZKwjT

Am I missing something?

kizniche, 2018-01-23:
The Output duration is only used for turning On. Turning off will just turn the Output Off.

How is your Output turning On?

kizniche, 2018-01-23:
I'll look into the Times to Check option (I hadn't tested that option beyond "1")

kizniche, 2018-01-23:
The way you have it set up (from the image), when the ping fails, your relay will turn off.

r1rider01, 2018-01-24:
Correct for some reason on this relay it has to be on for it to work.  So I want it to turn off then back on when ping fails. But nothing is happening

kizniche, 2018-01-24:
If a Low state (0 volts) turns the relay on, and you want the relay to turn off for 0.5 seconds, this is how the output and conditional should be set up to accomplish this:

Output:

- On Trigger = High (this will inverse the actual output. When Mycodo says it's on (green), it should actually be off, and when Mycodo says it's off (red) it should actually be on).
 - Start State: Off (this will start with it actually on)

Conditional Action: Output turn On for 0.5 seconds

What this does is it will do is when the conditional is triggered, it will instruct Mycodo to turn the Output On for 0.5 seconds, then Off. What will actually happen is it will turn the Output Off for 0.5 seconds and then back on.

Does that make sense?

kizniche, 2018-01-24:
> nothing is happening

Does turning the relay On and Off from the Output page actually actuate the relay? If not, it sounds like a hardware problem on your end.

r1rider01, 2018-01-24:
did the above settings.... changing the output works (shows red now like my other outputs)... and yes turning if on and off does actuate the relay..

but the server ping and the function still don't work.   I have 2 setup now, and neither work.

kizniche, 2018-01-25:
Thanks for the information. I'll do some tests by the end of the weekend and see what I can fix.

kizniche, 2018-01-28:
I'm now seeing my Ping input show 0 when it should show 1. I'm trying to figure it out.

kizniche, 2018-01-28:
Strange. That must have been a fluke, because it's working now

kizniche, 2018-01-28:
Is the Max Age of your conditional greater than the Period of Server Ping Input? If it's not, the conditional can't operate properly.

kizniche, 2018-03-04:
I figured out this issue. The deadtime is not the amount of time to wait for each response (form each of cound specified), but rather the entire time the command takes to finish. So, if you have the count set to 10, it will take approximately 10 seconds for the ping command to complete. However, if you have the deadtime set to 2 seconds, the ping command will terminate after 2 pings and return that the ping was unsuccessful. So, if you increase your count number, you also have to increase the deadtime proportionally.

closed 2018-03-04 by kizniche