kizniche/Mycodo #390: Long gauge name shifts graphs' position

zsole2, 2018-01-20:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.9
 
#### Problem Description
[image]

Both gauge's height is set at  150

### Additional Notes
The lesson: you should use short names, if you want too many gauges next to each other...

zsole2, 2018-01-20:
Lesson 2: don't do crazy stuff:
[image]

kizniche, 2018-01-20:
It's hard to determine the maximum text length because different devices will render the screen slightly differently. So, I leave it up to the user to self-regulate their text length. I try not to impose restrictions unless it's absolutely necessary and is well-defined (like cutting names on LCD lines so the data is always able to be properly displayed).

kizniche, 2018-01-20:
I could see if I can merely create a div element that's the width of the dashboard element (minus the + icon) and try to restrict the text to that, causing a new line to form at the end.

zsole2, 2018-01-20:
The only reason to start this issue was to leave a trail for later searches to find it (if there are people who search the issues before asking). In no way I wanted you to do more coding for a feature that any sane person could easily avoid.

closed 2018-01-20 by zsole2