kizniche/Mycodo #392: Gauge feature request: tooltip to show timestamp

zsole2, 2018-01-21:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.10
 
Tooltip on gauges only shows Series 1 and the value. Can we get there the detailed data, as with the graphs? I mean specifically to timestamp, as is shown on the Live page. 
When I set up a gauge for an Edge sensor, it did not show its value as is shown on the Live page, until the value got refreshed. Then it sometimes reverts to 0 again, not really sure when and why. 
This is not really convenient as I set my sensor reset period to 1200s, so I have to wait 20 mins to see the sensor status on the gauge.

zsole2, 2018-01-23:
Thanks for this, I forgot to check it. It would be nice if these parameters could be somehow connected, e.g. the  max age could default to the period, or the tooltip could show it. Being on different page makes it hard to remember all the connection between those parameters.

zsole2, 2018-02-04:
Thanks. Not what I had in mind, but certainly as informative. The only advantage of having the timestamp in the  tooltip is that some vertical screen real estate could be saved.

kizniche, 2018-02-04:
It's the first implement I could come up with. I'll revisit it again later to try to get it to appear within the graph itself. I made an attempt but failed, so this was the easiest alternative.

zsole2, 2018-02-04:
I did not want to generate too much extra work, I thought it can be done the same way as it is for the graphs.

closed 2018-02-26 by kizniche