kizniche/Mycodo #394: Edge sensor suggestion

zsole2, 2018-01-23:
I wonder now as I struggle a bit with my edge sensors and conditionals, if it would be good idea to add an option for setting up the edge sensors either for detecting the EDGE  or the GPIO state. This distinction is already there in the conditionals. 
In my case, the sensor detects the edge, then I use conditionals based on the GPIO state. With the level sensors, if the container is full, there should be no edge event if the conditions are perfect: the sensor is either at one of the end states. So this makes the conditional depend on a stochastic event. 
It would add  additional clarity in the GUI:  if  the sensor is set up  to detect edge, then the conditional could only include the EDGE events, and if it is set up to detect the GPIIO state, then the conditional include the LOW/HIGH events.

kizniche, 2018-01-23:
I'm not sure I understand what you're suggesting. Can you provide an example to clarify?

zsole2, 2018-01-23:
My understanding that the edge sensing is based on rpi.gpio, and an edge signal triggers a callback function. This handles the assignment of 1 and -1 values to the sensor-reading. A period is set up where no reading takes place. Am i mistaken? 
My scenario is much simpler, I do not use the edge signal, i.e. do not need to wait for an edge. After the given  period, the state of the GPIO needs to read out, and that's it.

So my suggestion is to set the sensor (Data panel) as either EDGE or STATE (instead of the current rising/falling/both). This is like setting it up as either dynamic or static detection of the given GPIO pin's state. After the user selects one of these, period may be settable, and for the edge the bounce time. On the conditional settings, you already have the two sides for  'Edge Detected' and 'GPIO State'.

kizniche, 2018-01-23:
So you're basically asking if a GPIO State Input can be created?

zsole2, 2018-01-23:
Yes, it can be said so... and the conditional is already there...

zsole2, 2018-01-27:
Thanks for dding the GPIO State Input, I was able to set it up. On the other hand, GPIO State is not available as a conditional to set up, and the already set up Edge conditional says at the measurement type that 'None Available' . Adding a new Edge conditional resulted the same.

kizniche, 2018-01-27:
Add a Measurement conditional then use Greater than 0 and Less than 1 to trigger it.

zsole2, 2018-01-27:
Actually I close this and continue troubleshooting #387

closed 2018-01-27 by zsole2