kizniche/Mycodo #396: Help config PWM LED Light compensation

leoneo, 2018-01-25:
Hi,

I would like some help to configure Mycodo to have a constant brightness during the day even if a cloud passes. I would like to have a sunrise and a sunset progressive.

I am a little lost with all these functions, PID, PWM, method, Timer...

hardware:
Light sensor, Dimmable LED, Relay

Thanks
[image]

kizniche, 2018-01-26:
This is how I think you can accomplish this:

1. Add a light sensor input and run it to find a good baseline for what full light is (eg. measurement of 30000).
2. Create a Daily Method that rises from 0 to 30000 from the sunrise part of the day and falls from 30000 to 0 at sunset part of the day.
3. Create a PWM Output and connect it to your light system that accepts a PWM signal to control the intensity of the light.
4. Create a PWM Method timer that has the PWM Output and the Daily Method from (2) selected.
5. Start the timer.

kizniche, 2018-04-06:
I recently added a Sunrise/Sunset Conditional. This allows actions to be triggered at sunrise and sunset. This is applicable here because you can create a Method that ramps up from 0% to 100% that's triggered at sunrise (or X minutes before sunrise), and a Method that ramps down from 100% to 0% at sunset (or X minutes before sunset). These Methods can be used in two different PID controllers that output PWM to your lights. Then, the Sunrise Conditional can trigger the sunrise Method in the PWM PID, and the Sunset Conditional trigger the sunset Method in the other PWM PID.

closed 2018-04-06 by kizniche