kizniche/Mycodo #397: How to access still photos/timelapse, download them?

vankomicin, 2018-01-25:
5.5.13 

I tried searching but nothing... Where do I access the timelapse photos/still photos? The answer will probably be simple, thanks in advance.

kizniche, 2018-01-26:
They are located at ~/Mycodo/cameras

vankomicin, 2018-01-26:
Well I checked the folder even before posting the issue. Although I can capture stills and see the last timelapse photo and also see the live stream of my picam in the interface, there are no files in the ~/Mycodo/cameras folder. Any suggestions? Thanks.

kizniche, 2018-01-27:
There shouldn't be any images in that folder, but there should be folders with the titles of each camera ID that you must enter to see additional ```still``` ```timelapsr``` folders that contain the images.

vankomicin, 2018-02-03:
Unfortunately I couldn't find any folders in the /cameras folder.

kizniche, 2018-02-03:
This doesn't make sense to me, if you are able to view images. The default directories are hard-coded:

The install directory will be the root directory where Mycodo is installed. This is typically /home/pi/Mycodo or replace 'pi' with whatever your user is. This is where it's hard-coded:

https://github.com/kizniche/Mycodo/blob/aa93c0ff72c09d6029a9c667b8fc77cb47b681ef/mycodo/config.py#L316-L317

This is the ```cameras``` directory being where images are stored. It's a directory called 'cameras' in the install directory from above:

https://github.com/kizniche/Mycodo/blob/aa93c0ff72c09d6029a9c667b8fc77cb47b681ef/mycodo/devices/camera.py#L36

Here is the full code that acquires the image:

https://github.com/kizniche/Mycodo/blob/aa93c0ff72c09d6029a9c667b8fc77cb47b681ef/mycodo/devices/camera.py#L34-L45

Here's the code that serves the acquired image to the camera page, using the same camera directory (called PATH_CAMERAS in the below code):

https://github.com/kizniche/Mycodo/blob/aa93c0ff72c09d6029a9c667b8fc77cb47b681ef/mycodo/mycodo_flask/routes_general.py#L71-L138

This is tested and working on one of my systems, and I have many files (stills and timelapses).

vankomicin, 2018-02-03:
I'm scratching my head.
[image]

kizniche, 2018-02-03:
Try listing the directory from the command line.

vankomicin, 2018-02-03:
Thanks again. Now I could access the directory over terminal as root. So it's there as you stated. I will have to look into permissions and Cyberduck's settings more closely to be able to easily transfer.

kizniche, 2018-02-03:
Aha! Good to hear.

Can you paste a line with ```ls -la``` so I may see what user/group they were? It should be ```mycodo```:

```
-rw-r--r-- 1 mycodo mycodo 251975 Jan 10 21:49 Still-1-PiCam-2018-01-10_21-49-38.jpg
```

vankomicin, 2018-02-03:
root@raspberrypi:/home/pi/Mycodo/cameras/5c4239aa-a19b-493c-815e-b4a2ce30f432# ls -la
total 108
drwxrwxr-- 4 mycodo mycodo  4096 Jan 21 12:32 .
drwxrwxr-- 4 mycodo mycodo  4096 Feb  3 10:18 ..
drwxrwxr-- 2 mycodo mycodo 86016 Feb  3 16:37 still
drwxrwxr-- 2 mycodo mycodo 12288 Feb  3 10:26 timelapse

kizniche, 2018-02-03:
You should be able to view those files as your user ```pi```

kizniche, 2018-02-03:
Looking back at it, can you go into one of those directories so I can see the permissions of the files? Those permissions are what I care about.

kizniche, 2018-02-03:
Now looking back at your image, your software wasn't even seeing the directories. That's strange, if their permissions says they're readable. Yeah, sounds like a software issue on your end.

vankomicin, 2018-02-03:
as root
-rw-r--r-- 1 mycodo mycodo 282023 Feb  3 16:37 Still-1-1-2018-02-03_16-37-16.jpg
now I'll try as pi in terminal

vankomicin, 2018-02-03:
pi@raspberrypi:~/Mycodo $ cd /home/pi/Mycodo/cameras/5c4239aa-a19b-493c-815e-b4a2ce30f432
bash: cd: /home/pi/Mycodo/cameras/5c4239aa-a19b-493c-815e-b4a2ce30f432: Permission denied

vankomicin, 2018-02-03:
pi@raspberrypi:~/Mycodo $ cd /home/pi/Mycodo/cameras
bash: cd: /home/pi/Mycodo/cameras: Permission denied

kizniche, 2018-02-03:
What are the permissions of ```/home/pi/Mycodo/cameras```?

vankomicin, 2018-02-03:
drwxrwxr--  4 mycodo mycodo    4096 Feb  3 10:18 cameras

kizniche, 2018-02-03:
That doesn't make any sense. The permissions state it's readable by all users.

kizniche, 2018-02-03:
What permissions does ```/home/pi/Mycodo``` have?

vankomicin, 2018-02-03:
drwxrwxr-x  9 mycodo mycodo  4096 Feb  2 21:27 Mycodo

kizniche, 2018-02-03:
You have something screwy going on. I think you're on your own ;)

vankomicin, 2018-02-03:
Well, that's kinda cool. Will report back if anything comes up. Thanks for your time!

kizniche, 2018-02-03:
You can certainly try to get around this by adding your user ```pi``` to the group ```mycodo```, with this command:

```bash
sudo adduser pi mycodo
```

You'll need to log out and back in for it to take effect.

vankomicin, 2018-02-03:
Great! That nailed it. I can access the folders now with Cyberduck for easy transfer. Thanks.

kizniche, 2018-02-03:
Good to hear! Glad to help.

closed 2018-02-03 by kizniche