kizniche/Mycodo #398: Add PID Control Hysteresis

kizniche, 2018-02-01:
From [this comment](https://kylegabriel.com/projects/2015/04/mushroom-cultivation-revisited.html#comment-577)

Proposed is the addition of the PID option ```band``` to enable PID hysteresis.

Example: [Temperature Controllers Using Hysteresis](https://na.industrial.panasonic.com/blog/temperature-controllers-using-hysteresis)

To my understanding it should operate like this:

### If PID is set to Raise

Raise to the maximum value (setpoint + band value), then allow the condition to fall to the minimum value (setpoint - band value) before resuming regulation.

### If PID is set to Lower

Lower to the minimum value (setpoint - band value), then allow the condition to raise to the maximum value (setpoint + band value) before resuming regulation.

### If PID is set to Both

When the condition enters the band region (setpoint +/- band), do not apply the PID. When the condition falls outside (above or below), then begin applying the PID again. If it rises above the maximum value, begin lowering until it enters the band again. When it lowers below the minimum value, begin raising until it enters the band again.

Let me know your thoughts, if I'm missing a part of this concept, or if there are alternative methods that would be desirable to add.

kizniche, 2018-02-01:
Currently untested, but I implemented the ```band``` option and the logic to determine the appropriate setpoint if raising/lowering:

https://github.com/kizniche/Mycodo/blob/e9038c9693ee30d52ffc14be04ef42853f6960cf/mycodo/controller_pid.py#L364-L389

I should get around to testing this weekend or sooner and have a release by the end of the weekend.

zsole2, 2018-02-02:
I like this a lot, will be very useful for the health of my refrigerators!

zsole2, 2018-02-03:
Kyle, looking at the link above on hysteresis control got me thinking that PID control and this hysteresis control are alternatives, and should not be combined  together. A little googling seems to support this. 
Do you agree? 
In that case, this needs to be added as a separate function.
At brewing forums, there are plenty of discussions that they use PID for heating and hysteresis for cooling.

kizniche, 2018-02-03:
I added the ```band``` to the PID options with the option of using it. If set to 0, there is no band and no hysteresis. However if you set it to a non-0 value, then it will operate like I describe in the first post, still using PID control. I don't think there needs to be a separation, but being able to turn it off is essential.

kizniche, 2018-02-03:
I can't conceptualize how the same PID can use hysteresis for just lowering and not use it for raising.

I can see two PIDs being used, one raising, and one lowering with hysteresis. This could work well, without the PIDs competing too much if tuned properly.

zsole2, 2018-02-03:
I believe that those were two different operations, but I'm not familiar with brewing. The competition was a main topic of the discussion, though.

kizniche, 2018-02-03:
The current iteration of Hysteresis in Mycodo's PID controllers can theoretically work very well together. Here's one heating/cooling scenario:

### PID 1: Heat, Raise to 30

Setpoint: 30, Hysteresis: 0 (disabled), the PID will raise the temperature to 30 and stop when the temperature is 30 or above.

### PID 2: Cool, Lower to 31

Setpoint: 33, Hysteresis: 2 (Min Band: 31, Max Band: 35), the PID will lower to 31, then stop until the temperature rises above 35, at which point it will begin regulating again to 31.

Leaving the gap of one degree (30 to 31, there shouldn't be any direct competition between the PID controllers. You would even make the gap 0 degrees (Rise to 30, Lower to 30) and the PIDs should still function properly (considering they're well-tuned).

kizniche, 2018-02-03:
I just tested on a PID set to Raise (so should also work for Lower). Looks good. Next to test Both directions.

[image]

kizniche, 2018-02-03:
With testing the PID with direction set to Both, I had to fix some issues related to the wrong outputs being activated, due to the integrator and/or derivator.  To do this, I reset both the integrator and derivator upon the current measurement reaching the opposite band threshold. That is, when the measurement reaches beyond either the Minimum Band or Maximum Band, the integrator and derivator are reset to 0, so the PID acts as though it were just turned on. This creates a smooth output instead of the aberrant behavior in the following graphs;

### Before Fix

[image]

### After Fix

[image]

kizniche, 2018-02-03:
Another test that has less steep slopes so the PID deactivation is a little bit more noticeable when the measurements enter between the bands.

[image]

zsole2, 2018-02-05:
I have a few test examples now.
First, the nice one:
[image]

This is my refrigerator PID, which was on 300s periods. It is clear that the frequency of operation is decreased.
Now the less nice one:
[image]

This is my processor fan, on 60s period. Need some tuning. 
Interestingly, without the band it looked quite OK (not the same system, but the current system looked quite similar):
[image]

About 5-15s run of the fun kept the  the processor at 40C. Now it is much worse. I'll remove the band to check that indeed the fan behaved similarly in the current system.

zsole2, 2018-02-05:
Here is the CPU fan with the previous PID settings, without the band:
[image]

frodus17, 2018-02-12:
Kyle,
When I get time this week, I'll play around with this to test.

kizniche, 2018-03-04:
Let me know if there are any issues. I'm closing this, as I believe it's complete.

closed 2018-03-04 by kizniche

frodus17, 2018-03-04:
Sounds good, looks OK on my end for all of the delays and hysteresis.