kizniche/Mycodo #401: upgrade fail UI

r1rider01, 2018-02-05:
Upgraded from 5.5.13 to 5.5.18 and now the url gives me the following error code:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 507, in do_execute
    cursor.execute(statement, parameters)
sqlite3.OperationalError: no such column: pid.band

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 688, in page_live
    pid = PID.query.all()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2724, in all
    return list(self)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2876, in __iter__
    return self._execute_and_instances(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/orm/query.py", line 2899, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 948, in execute
    return meth(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/sql/elements.py", line 269, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1060, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1200, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1413, in _handle_dbapi_exception
    exc_info
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 203, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/util/compat.py", line 186, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/sqlalchemy/engine/default.py", line 507, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such column: pid.band [SQL: 'SELECT pid.id AS pid_id, pid.unique_id AS pid_unique_id, pid.name AS pid_name, pid.is_activated AS pid_is_activated, pid.is_held AS pid_is_held, pid.is_paused AS pid_is_paused, pid.is_preset AS pid_is_preset, pid.preset_name AS pid_preset_name, pid.period AS pid_period, pid.max_measure_age AS pid_max_measure_age, pid.measurement AS pid_measurement, pid.direction AS pid_direction, pid.setpoint AS pid_setpoint, pid.band AS pid_band, pid.method_id AS pid_method_id, pid.method_start_time AS pid_method_start_time, pid.method_end_time AS pid_method_end_time, pid.p AS pid_p, pid.i AS pid_i, pid.d AS pid_d, pid.integrator_min AS pid_integrator_min, pid.integrator_max AS pid_integrator_max, pid.raise_relay_id AS pid_raise_relay_id, pid.raise_min_duration AS pid_raise_min_duration, pid.raise_max_duration AS pid_raise_max_duration, pid.raise_min_off_duration AS pid_raise_min_off_duration, pid.lower_relay_id AS pid_lower_relay_id, pid.lower_min_duration AS pid_lower_min_duration, pid.lower_max_duration AS pid_lower_max_duration, pid.lower_min_off_duration AS pid_lower_min_off_duration \nFROM pid']

kizniche, 2018-02-05:
Can you attach the upgrade log contents from the most recent upgrade?

r1rider01, 2018-02-05:
well i may have messed those up...

I just a reinstall over the top of the mycodo folder, but still have this same error...and when i run the upgrade via commandline it tells me its up to date already.

r1rider01, 2018-02-05:
[mycodoupgrade.log](https://github.com/kizniche/Mycodo/files/1696616/mycodoupgrade.log)

kizniche, 2018-02-05:
```
[2018-02-05 11:19:37] #### Upgrade to v5.5.18 initiated 2018-02-05_11-19-33 ####
[2018-02-05 11:19:37] 
[2018-02-05 11:19:37] #### Beginning Upgrade: Stage 1 of 2 ####
[2018-02-05 11:20:38] Stopping the Mycodo daemon...Error: Unable to stop the daemon. Continuing anyway...
[2018-02-05 11:20:38] Done.
[2018-02-05 11:20:38] Stopping the Mycodo daemon...Error: Unable to stop the daemon. Continuing anyway...
[2018-02-05 11:20:38] Done.
[2018-02-05 11:20:52] Downloading latest Mycodo version to /home/pi/mycodo-5.5.18.tar.gz...Done.
[2018-02-05 11:20:52] Creating /tmp/Mycodo-5.5.18...Done.
[2018-02-05 11:20:52] Extracting /home/pi/mycodo-5.5.18.tar.gz to /tmp/Mycodo-5.5.18...Done.
[2018-02-05 11:20:52] Removing /home/pi/mycodo-5.5.18.tar.gz...Done.
[2018-02-05 11:20:52] Copying /tmp/Mycodo-5.5.18/.upgrade status file to /tmp/Mycodo-5.5.18...Done.
[2018-02-05 11:20:52] Moving env directory...Done.
[2018-02-05 11:20:52] Copying databases from /home/pi/Mycodo/databases/ to /tmp/Mycodo-5.5.18/databases...Done.
[2018-02-05 11:20:52] Copying flask_secret_key from /home/pi/Mycodo/databases/ to /tmp/Mycodo-5.5.18/databases...Done.
[2018-02-05 11:20:52] Copying statistics ID...Done.
[2018-02-05 11:20:52] Copying SSL certificates...Done.
[2018-02-05 11:20:52] #### Stage 1 of 2 Complete ####
[2018-02-05 11:20:52] 
[2018-02-05 11:20:52] #### Continuing Upgrade: Stage 2 of 2 ####
[2018-02-05 11:20:52] 
[2018-02-05 11:20:52] Moving old Mycodo from /home/pi/Mycodo to /var/Mycodo-backups/Mycodo-backup-2018-02-05_11-19-33-5.5.13...Done.
[2018-02-05 11:20:52] Moving new Mycodo from /tmp/Mycodo-5.5.18 to /home/pi/Mycodo...Done.
[2018-02-05 11:20:54] Downloading latest Mycodo version to /home/pi/mycodo-5.5.18.tar.gz...Done.
[2018-02-05 11:20:54] Creating /tmp/Mycodo-5.5.18...Done.
[2018-02-05 11:20:54] Extracting /home/pi/mycodo-5.5.18.tar.gz to /tmp/Mycodo-5.5.18...Failed: Error while trying to extract files from /home/pi/mycodo-5.5.18.tar.gz to /tmp/Mycodo-5.5.18.
[2018-02-05 11:20:54] 
[2018-02-05 11:20:54] 
[2018-02-05 11:20:54] #### ERROR ####
[2018-02-05 11:20:54] There was an error detected during the upgrade. Please review the log at /var/log/mycodo/mycodoupgrade.log
```

This appears to be the issue:

```
[2018-02-05 11:20:54] Extracting /home/pi/mycodo-5.5.18.tar.gz to /tmp/Mycodo-5.5.18...Failed: Error while trying to extract files from /home/pi/mycodo-5.5.18.tar.gz to /tmp/Mycodo-5.5.18.
```

How much free space do you have? You can use the ```df``` command to quickly find out.

r1rider01, 2018-02-05:
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       15242096 2251892  12338428  16% /
devtmpfs          443792       0    443792   0% /dev
tmpfs             448400       0    448400   0% /dev/shm
tmpfs             448400   11580    436820   3% /run
tmpfs               5120      12      5108   1% /run/lock
tmpfs             448400       0    448400   0% /sys/fs/cgroup
/dev/mmcblk0p1     41853   21329     20524  51% /boot
tmpfs              89680       0     89680   0% /run/user/1000

kizniche, 2018-02-05:
Wait a second. You have some strange stuff going on. Your log indicates Mycodo was downloaded more than once. 😕

kizniche, 2018-02-05:
Here is the upgrade script from Mycodo 5.5.13, which supposedly was running here. You can see that the Stage 2 doesn't have instruction to download Mycodo (this is done in Stage 1):

https://github.com/kizniche/Mycodo/blob/7462b36ddb307afa7501b116b0e10543b59f9198/mycodo/scripts/upgrade_mycodo_release.sh#L201-L233

kizniche, 2018-02-05:
It looks like you have two upgrades occurring simultaneously!

kizniche, 2018-02-05:
It's true, there are double entries in the upgrade log. Two upgrades are occurring and writing to the log. How did you initiate the upgrade?

r1rider01, 2018-02-05:
the web ui was acting funny when i tried to upgrade.

but I eventually just stopped the upgrade and rebooted the PI

then got the error above, so went and did a fresh install (overwrite)  still have error above

kizniche, 2018-02-05:
Move your /home/pi/Mycodo directory elsewhere, then follow the procedure for a new installation. Copy your database from /home/pi/Mycodo-moved/databases/mycodo.db to /home/pi/Mycodo/databases/ (delete the newly-generated one if it exists- overwrite the new one with your old one), then upgrade your database with the following command:

```bash
sudo mycodo-commands update-alembic
```

kizniche, 2018-02-05:
Then restart the daemon with:

```bash
sudo service mycodo restart
```

r1rider01, 2018-02-05:
Thanks!  I'm live again!

kizniche, 2018-02-05:
Great!

closed 2018-02-05 by kizniche