kizniche/Mycodo #402: Heat Pump system with heat and cool solenoid won't run compressor

frodus17, 2018-02-11:
## Mycodo Issue Report:
  - Specific Mycodo Version:  5.5.19
 
#### Problem Description
Please list:
  - what were you trying to do
I'm trying to turn on a compressor when a cooling solenoid is turned on. Then off when it turns off. I also need to turn the same compressor on when a heating solenoid is turned on, and off when it turns off. Unfortunately I set up the compressor to turn on via Function, and it turns on just fine, but Mycodo gets confused when Cooling is off and heating is on, it won't turn on the compressor accordingly.
  - specific setup details that are involved
The function is not possible to have Output 3 follow both output 1 or output 2.
 
### Errors
  - List any errors you encountered.
n/a
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)
n/a

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
Create 3 outputs, 1 PID function, 2 conditional when ON, 2 conditional when OFF
 2. step 2...
Configure conditional on when cooling solenoid is on to turn on the compressor, and turn off the compressor when cooling solenoid goes off. The same for heating solenoid.
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2018-02-11:
Correct me if I get your setup and desired effect incorrect:

### Outputs

 - Heat solenoid
 - Cooling solenoid
 - Compressor
 - Fan

### Desired operation

A PID controller regulates using a temperature in both directions using the heat solenoid as the Raise Output and the cooling solenoid as the Lower Output. When either the heat or cooling solenoids turn on, the compressor and fan should also turn on. When the heat or cooling solenoids turn off, the compressor and fan should turn off.

### Setup

 - Add 4 outputs, one for each device
 - PID set to regulate both directions with the heat solenoid set to the Raise Output and the cooling solenoid set to the Lower Output.
 - Add 4 Output Conditionals.

#### Output conditional settings

 - When Heat solenoid turns on, actions: 1. Turn on fan and 2. turn on compressor
 - When Heat solenoid turns off, actions: 1. Turn off fan and 2. turn off compressor
 - When Cooling solenoid turns on, actions: 1. Turn on fan and 2. turn on compressor
 - When Cooling solenoid turns off, actions: 1. Turn off fan and 2. turn off compressor

frodus17, 2018-02-12:
Kyle, 
Thanks for looking into this.

I confirmed I had all 4 outputs, 1 for each device. They all output as they should.
Under function, I made a PID and made the Heat solenoid the raise, and cooling the lower.
Then I made the output conditional settings, Only difference is I put for On for any duration. 

I can get heat ONLY to work, or cooling ONLY to work. The second I put the other ones active, they seem to override the output.

I think that what happens, is because the compressor is turned off when cooling is off, and heat cannot turn it on.

kizniche, 2018-02-13:
I'll reproduce your setup and find out what's stopping things from working. I suspect it's the positioning of the code to ensure the Raising output turns off when it switches to Lowering, and vise versa.

frodus17, 2018-02-13:
Thanks!

Cooling works! I tested a little tonight.

kizniche, 2018-02-13:
Can you take screenshots of your Outputs (expanded), Conditionals (with it expanded so I can see the Conditional Actions), and the PID Controller (expanded), so I can test your exact configuration?

frodus17, 2018-02-14:
Here you go.

[image]
[image]
[image]
[image]

frodus17, 2018-02-14:
Here you go
[image]
[image]
[image]
[image]

kizniche, 2018-02-14:
Thanks. Everything looks good, configuration-wise. I think I've isolated the offending code:

https://github.com/kizniche/Mycodo/blob/0963e471a568b1f0e4ef4924d7d4ba484d977983/mycodo/controller_pid.py#L521-L525

https://github.com/kizniche/Mycodo/blob/0963e471a568b1f0e4ef4924d7d4ba484d977983/mycodo/controller_pid.py#L600-L604

If you can edit these files and delete these lines, save, then restart the daemon with:

```sudo service mycodo restart```

See if that will fix the issue with the compressor not turning on.

kizniche, 2018-02-14:
This believe this code isn't even necessary anymore. It's actually a vestige of a previous iteration of the PID controller and really doesn't have a useful purpose. If you can confirm it fixed the issue, I think I'll just dete it, instead of adding an option to disable it. I'll have to think about some more use cases to make sure it can be safely removed.

frodus17, 2018-02-14:
I removed that code and it acts the same.

kizniche, 2018-02-14:
Thanks for testing. I've mimicked your setup, reproduced the issue, and am currently investigating now.

frodus17, 2018-02-14:
I "sometimes" see it blip on. Let me change my graph so you can maybe see.

kizniche, 2018-02-14:
I can verify it is turning on for a fraction of a second. I should have a fix in a moment.

frodus17, 2018-02-14:
I hooked up the output to another SSR with an indicator light, and it blips on for a split second.

The graph actually shows a 1 status, so it's working, or trying to!

Thanks for looking into this. Hope you got the donation I sent :)

kizniche, 2018-02-14:
I found the bug. It's a use case I didn't anticipate. I'll push a fix shortly and explain the edit you can perform to test if it works for you. I did, thank you very much. That made my week :)

kizniche, 2018-02-14:
Alright! I got it working as expected. Delete the lines below, save, restart the daemon, and see if it works. You can keep the previous lines I mentioned deleted as well.

https://github.com/kizniche/Mycodo/blob/0963e471a568b1f0e4ef4924d7d4ba484d977983/mycodo/controller_pid.py#L590-L591

https://github.com/kizniche/Mycodo/blob/0963e471a568b1f0e4ef4924d7d4ba484d977983/mycodo/controller_pid.py#L670-L671

frodus17, 2018-02-14:
Kyle, I'll test in a second.

You've made an amazing program. I'm REALLY bummed I didn't just pull the trigger and implement earlier. I wasted a lot of time trying to get PID controllers to work right.

I just realized that I can do set-point profiles over time, which is great for fermentation where it'll sit at one temp, then ramp up or down to another and sit there, then go down lower to lager or cold crash, and email me when I need to check on things. You thought of a LOT of other use cases!

frodus17, 2018-02-14:
Confirmed, it works now!

I tested on heat AND cool. Outputs act as I would expect. I'm going to try also the Fan output now.

kizniche, 2018-02-14:
Thanks. I'm sometimes surprised how large this has become. This started just as a project to learn Python. I kept desiring new uses, and progressively added new features, break things, add new features, tear down and build a new framework, add new features, repeat... But now, after a lot of learning it's now extremely easy to work with (compared to previous versions, v2, v3, v4).

> Confirmed, it works now!

Good to hear. I suspect the other conditionals should also work. I'll incorporate this into the next version release. Thanks for testing.

frodus17, 2018-02-14:
I confirmed. I added another output, and it works.  I should be thanking you, for modifying it to match my need. I hope it doesn't break something else!

Now, I just need to figure out, how best to go about a defrost cycle.... need to think about that some, if I even need it.

kizniche, 2018-02-14:
It was a bug that prevented your setup from working. It's actually illogical to have it done the previous way (the code we just deleted). It *shouldn't* cause any issues to remove it.

You can set up a conditional that will turn off the PID when it gets too cold, then turn the PID back on when it reaches a temp it would be unfrozen.

kizniche, 2018-02-14:
I don't know enough about your system to make a good recommendation. My incubator in the lab has a heat bypass on the AC system so I can continually run the compressor and bypass the heat exchanger when I want to suspend cooling (to defrost or let the chamber heat back up).

frodus17, 2018-02-14:
That's kind of what Mine does. It's a head pump and heat mode bypasses the AC side. The problem is, it runs ALL THE TIME, and turns the heating and cooling solenoids on and off accordingly. It's super great at regulation, but it uses a lot of energy every month. Great thing about MyCodo, is I can run on and off, OR run continuous.

The cooling runs normal. The compressor runs through the air cooled condenser (outside of the chamber), through the cooling solenoid, into a thermal expansion valve and into the evaporator (inside the chamber) to a crankcase pressure regulating valve and back into the compressor.

The cooling bypasses the air cooled condenser and the thermal expansion valve.

Here's the chamber mechanical schematic:
[image]

frodus17, 2018-02-14:
How would I turn on and off a PID? That sounds like what I'd want to do. I could then run some timers, and run until another temp sensor gets high enough.

kizniche, 2018-02-14:
Activate/Deactivate PID is a Conditional Action. It isn't currently available elsewhere, like timers or methods.

frodus17, 2018-02-14:
I'm not sure I see what you're talking about. I don't see activate/deactivate PID anywhere. Am I missing it? I'm under Functions. 

Do you just turn the output pins off? Not the actual PID?

kizniche, 2018-02-14:
The Activate/Deactivate PID will completely turn it off and back on. It may or may not work well with your application. Do you run your cooler cold enough to freeze the inside heat exchanger?

frodus17, 2018-02-14:
NOW I see it. Thanks! Will put to use.

This software keeps getting better the more I delve into it. Setting fermentation profiles with methods is going to be really nice.

closed 2018-02-26 by kizniche