kizniche/Mycodo #403: Camera put slice of pic from right side to left side

RPi-Growbox, 2018-02-11:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.19 (have been also in earlier versions)
 
#### Problem Description
Please list:

  - Take timelapse or still pics
  -  Tried using Logitech Quickcam, MS Lifecam and cheap unbranded cam.

### Errors
  - List any errors you encountered. 

About 1/3 of pics have slice from right side of the pic on the left side, size of slice 1/4 - 1/8 of the pic.
  

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Set cam with fswebcam
 2. Take stills/timelapse
 (3. Flustrate cleaning thousands of pics to make acceptable timelapse)
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

Not sure was did this with opencv-library. Maybe just fswebcam-issue?

RPi-Growbox, 2018-02-11:
Ok, now it does that slicing with only cheap unbranded. But flipping image won't work in any direction for me in any camera. Do I remember right there was more settings to cameras, or is it just for some cameras?

kizniche, 2018-02-11:
Each USB camera may work differently. It's very difficult for software (fswebcam) to be developed to work with all camera hardware, therefore it's a gamble. The Pi Camera will be the only device that will operate consistently because the hardware was designed to work with the Pi. If you have confirmed any command line software to work with your camera (including fswebcam), please copy the commands you used to acquire an image and I'll look into whether it would be a good addition or amendment to the Mycodo code.

closed 2018-02-26 by kizniche