kizniche/Mycodo #405: Errors on startup with math controllers

zsole2, 2018-02-13:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.20
 
### Errors
The following appears in the log after reboots:
```
2018-02-13 04:49:13,135 - mycodo.math_3 - ERROR - Run Error: HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?db=mycodo_db&q=SELECT+value+FROM+temperature+WHERE+device_id%3D%2773103a85-ca36-4f55-9443-8b18f13792fa%27+AND+time+%3E+now%28%29+-+60s (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb4fa2d90>: Failed to establish a new connection: [Errno 111] Connection refused',))
Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.5/site-packages/urllib3/connection.py", line 141, in _new_conn
    (self.host, self.port), self.timeout, **extra_kw)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/urllib3/util/connection.py", line 83, in create_connection
    raise err
  File "/var/mycodo-root/env/lib/python3.5/site-packages/urllib3/util/connection.py", line 73, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 601, in urlopen
    chunked=chunked)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 357, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.5/http/client.py", line 1107, in request
    self._send_request(method, url, body, headers)
  File "/usr/lib/python3.5/http/client.py", line 1152, in _send_request
    self.endheaders(body)
  File "/usr/lib/python3.5/http/client.py", line 1103, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python3.5/http/client.py", line 934, in _send_output
    self.send(msg)
  File "/usr/lib/python3.5/http/client.py", line 877, in send
    self.connect()
  File "/var/mycodo-root/env/lib/python3.5/site-packages/urllib3/connection.py", line 166, in connect
    conn = self._new_conn()
  File "/var/mycodo-root/env/lib/python3.5/site-packages/urllib3/connection.py", line 150, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0xb4fa2d90>: Failed to establish a new connection: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/env/lib/python3.5/site-packages/requests/adapters.py", line 440, in send
    timeout=timeout
  File "/var/mycodo-root/env/lib/python3.5/site-packages/urllib3/connectionpool.py", line 639, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/var/mycodo-root/env/lib/python3.5/site-packages/urllib3/util/retry.py", line 388, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?db=mycodo_db&q=SELECT+value+FROM+temperature+WHERE+device_id%3D%2773103a85-ca36-4f55-9443-8b18f13792fa%27+AND+time+%3E+now%28%29+-+60s (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb4fa2d90>: Failed to establish a new connection: [Errno 111] Connection refused',))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_math.py", line 186, in run
    success, measure = self.get_measurements_from_str(self.equation_input)
  File "/var/mycodo-root/mycodo/controller_math.py", line 325, in get_measurements_from_str
    self.max_measure_age)
  File "/var/mycodo-root/mycodo/utils/influx.py", line 141, in read_last_influxdb
    last_measurement = client.query(query).raw
  File "/var/mycodo-root/env/lib/python3.5/site-packages/influxdb/client.py", line 394, in query
    expected_response_code=expected_response_code
  File "/var/mycodo-root/env/lib/python3.5/site-packages/influxdb/client.py", line 252, in request
    timeout=self._timeout
  File "/var/mycodo-root/env/lib/python3.5/site-packages/requests/sessions.py", line 508, in request
    resp = self.send(prep, **send_kwargs)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/requests/sessions.py", line 618, in send
    r = adapter.send(request, **kwargs)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/requests/adapters.py", line 508, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /query?db=mycodo_db&q=SELECT+value+FROM+temperature+WHERE+device_id%3D%2773103a85-ca36-4f55-9443-8b18f13792fa%27+AND+time+%3E+now%28%29+-+60s (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb4fa2d90>: Failed to establish a new connection: [Errno 111] Connection refused',))
```

It was there in the log since 5.5.16, i.e. since I activated a Math Input.

Is it from `influxd` not started yet?

kizniche, 2018-02-13:
Yes, it's most likely due to influxdb either not running or still starting up.

zsole2, 2018-02-13:
BTW, is it possible to not draw the interpolating line where no measurements were recorded? See this quick and dirty edit of the above screenshot:

[image]

kizniche, 2018-02-13:
I've been meaning to build in a way to automatically deactivate and reactivate controllers to attempt to alleviate issues.

kizniche, 2018-02-13:
I'm not sure about the line drawing. I'll look into it.

kizniche, 2018-02-13:
If the missing points were null values, I could stop the interpolation with https://api.highcharts.com/highstock/plotOptions.series.connectNulls however there are no null values added to the measurement database. Adding the null values before serving to the chart would be challenging to do, so unless an idea comes to mind, I'm going to say this probably can't be done.

zsole2, 2018-02-13:
I see.
As an idea, can a null value be added at the time of activation and of deactivation of an input? 
This would fit nicely into the overall scheme of Mycodo, these events are logged anyway with their respective timestamps. 
Then the connection of null values may be implemented either at the graph level, or  (even better in my opinion) at the input level.

kizniche, 2018-02-13:
Apparently null values cannot be inserted into the influxdb database.

zsole2, 2018-02-16:
Back to the original error, should I close this? 
Can the status of influxd be checked before querying it by the math inputs?

zsole2, 2018-02-16:
BTW, yesterday I've went and looked at  some discussions about inserting null values into influxd, apparently it  is a hotly debated issue. But one place showed a workaround, I believe suggested by one of the developers of influx and claimed that it is fast and efficient. Since I did not understand a bit of it, I just moved on. It looked that something too complicated just to have gaps on our graphs...

kizniche, 2018-02-16:
I think the best I can do is catch the exception and present a nicer, smaller error, like so:

```python
    try:
        last_measurement = client.query(query).raw
    except requests.exceptions.ConnectionError:
        logger.debug("Failed to establish a new influxdb connection. Ensure influxdb is running.")
        last_measurement = None
```

I've been looking in the [API docs](https://influxdb-python.readthedocs.io/en/latest/api-documentation.html), but haven't come across a way to test the connection.

kizniche, 2018-02-16:
I should be able to test this tonight.

kizniche, 2018-03-04:
I wasn't able to reproduce this error, so I'm not sure what caused it. It may be that your influxdb takes a longer time to start, and during that period, connections are refused. I'm going to close this, as I've added exception-handling, and I believe that's all I can do.

closed 2018-03-04 by kizniche