kizniche/Mycodo #408: Memory leak

zsole2, 2018-02-19:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.5.21
 
#### Problem Description
I have Mycodo RAM on a gauge, but was suspicious about it increasing.
Here is the async graph:

[image]

What should I look at for testing?

zsole2, 2018-02-19:
And this is my test system, where only CPU temp, CPU load, RAM usage and Free memory are activated since setup, and nothing else:
[image]

kizniche, 2018-02-19:
Can you deactivate all controllers and let run to see if it increases? Then activate one controller at a time to determine if a particular controller is the offender?

zsole2, 2018-02-20:
I deactivated all inputs except the RAM:
[image]

The big jump is at 16:25, which is the time when the daemon was originally started. 
I have no idea about the slow increase at about 03:25 in the night, I certainly did not do anything then.

kizniche, 2018-02-20:
Did you run the daemon with no controllers started?

zsole2, 2018-02-21:
How can I start the daemon without controllers?
I had  only 4 Inputs (CPU temp, CPU load, RAM usage and Free memory) activated in the first place, and nothing else, no timer, no output, nothing.

kizniche, 2018-02-21:
Just deactivate all the controllers, let the daemon run a few days, then activate the RAM Input to see how much ram is in use.

kizniche, 2018-02-21:
I would suggest restarting after deactivating everything to start from a clean slate.

zsole2, 2018-02-25:
Here is my memory graph, after reboot, and then turned the RAM Input on after a few days, with the last stored data-point highlighted:
[image]
In that period, I did not check the GUI or SSHd in.

zsole2, 2018-02-26:
Here is a quick update, before I turn off this PiZero:
[image]

Apparently the memory use is steadily increasing, and activating the RAM Input did not affect it neither positively nor negatively.

Side question: we were discussing elsewhere missing data points and null values to break the lines on the graphs. It looks that the async graph has this feature.

kizniche, 2018-10-13:
It appears the memory leak has been fixed (see attached image)?? If you have any more info about this, feel free to reopen this issue.

[image]

closed 2018-10-13 by kizniche