kizniche/Mycodo #411: Sensor Support Atlas EC

RynFlutsch, 2018-02-22:
Hi,
 i would like to check my nutrition-level, so i missed the EC sensor.
Maybe add some, i.e.
https://www.atlas-scientific.com/_files/_datasheets/_probe/IEC_probe.pdf
?

kizniche, 2018-02-22:
If you have the sensor and you're willing to test, I'll add a Mycodo module.

RynFlutsch, 2018-02-23:
I will buy one if we get it working :)
Atlas Probe support is for all Probes, they offer ?

kizniche, 2018-02-23:
You have to buy the specific circuit for each specific probe. So, for any of the EC probes, you'll need the [EC circuit](https://www.atlas-scientific.com/product_pages/circuits/ezo_ec.html). I already have the framework in place to communicate with any Atlas Circuit, however it's all theoretical until a physical sensor is available to test with. I predict getting the sensor up and running will be rather easy.

kizniche, 2018-02-23:
The circuit/probe can be calibrated, so I would also add calibration support to Mycodo (much the same way I did for the Atlas pH sensor). Since I've already done it for the pH sensor, it should be easy to copy and modify my code to add support for calibrating the EC sensor.

RynFlutsch, 2018-02-24:
You use these carrierboards ? With or without USB. I think the tentacle T3 suxx cause uart is missing.

kizniche, 2018-02-24:
The circuits support UART and I2C. I generally prefer I2C because I often have multiple sensors connected.

RynFlutsch, 2018-02-24:
[Usb](https://www.atlas-scientific.com/product_pages/components/usb-iso.html) can be used with Mycodo ?
Or you think that just the circuit item + sensor is enough with some soldering?

kizniche, 2018-02-24:
I've never used the USB converter, only the UART and I2C.

RynFlutsch, 2018-03-16:
I've got the Sensors, but atm no time for testing. 
You already have some informations to prepare the testing ?

kizniche, 2018-04-06:
Have you had a chance to test the new Atlas EC module in Mycodo?

RynFlutsch, 2018-04-06:
Hey, no not yet.
Next week, i will setup and test the sensors.

RynFlutsch, 2018-05-01:
Thanks for implementing! to much to do... I hope i can wire and test next week.

kizniche, 2018-05-10:
No hurry. Take your time. I'll be around when/if an issue arises.

kizniche, 2018-11-30:
Have you had a chance to test this?

eyespoint, 2018-12-01:
I was able to test Atlas EC with T3 shield and I2C and it works good. No idea about UART

kizniche, 2018-12-01:
Thanks! I'll close this issue then.

closed 2018-12-01 by kizniche

RynFlutsch, 2019-01-16:
Mycodo 7.0.5
Yesterday i tried atlas ph and EC via i2c:
Sensor reading unseccessful. 
But both return ErrNr 121 Remote i/o error.
Ec probei wired with sda, scl 1
and pH with sda, scl 0.
No overlay or multiplexer.

kizniche, 2019-01-16:
Please start a new issue