kizniche/Mycodo #413: Open relay when PWM greater than value

drgrumpy, 2018-02-27:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.6.0
 
#### Problem Description

 - I have a 12V cooling fan, that want to control via pwm, but when pwm is zero the fan spins at full speed
 - To overcome this the fan power is supplied via a relay
 - A conditional function then added to switch the relay on when the pwm output duty cycle is greater the min duty cycle (e.g. 10%) set in the PID cooling function and off when it is lower.....

But the relay never switches on (yes it works when manually turned on/off)...

The graph output shows the pwm values as negative (by design ? as this is cooling ).... 

Which got me thinking ....
so it seems that the internal value used as input for the conditional function is also using the negative value as when I change the conditional function to switch the relay on when the duty cycle is less than -10% and it now works....

This does not make much sense (to me at least), suggest not to make cooling pwm values negative for graphing or at least only turn them into negative values at the point of drawing the graphs ?

kizniche, 2018-02-27:
It's intentional, for PID regulation in both directions to look nice on graphs. See what I mean: https://github.com/kizniche/Mycodo/issues/398#issuecomment-362862288

kizniche, 2018-02-27:
I agree it should be stored as a positive value and the user should have the option of displaying it as a negative value on a graph. Thanks for the suggestion. I'll work on this soon.

kizniche, 2018-03-01:
One issue will be this will mess up users' currently -configured systems. So, I may slate this for the [next major version release](https://github.com/kizniche/Mycodo/wiki/Changes-for-Mycodo-6.0), so as to not cause potential havoc on currently-running systems.

kizniche, 2018-03-04:
I am slating this for the next major version release. I don't want to introduce something that could potentially break a user's setup or cause it to behave abnormally (would inverse potential actions).

closed 2018-03-04 by kizniche