kizniche/Mycodo #414: TSL2561 cannot be activated

drgrumpy, 2018-02-27:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.6.0
 
#### Problem Description

Connected TSL2561
Shows up with i2cdetect at 0x39

Add the sensor to the data....
Save, Click activate:
#### Result:
Error: Could not activate Input controller with ID 7: No module named 'tentacle_pi'
 
### From daemon log:
2018-02-27 03:34:08,657 - mycodo.daemon - ERROR - Could not activate Input controller with ID 7: No module named 'tentacle_pi'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 476, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 348, in __init__
    self.i2c_bus)
  File "/var/mycodo-root/mycodo/inputs/tsl2561.py", line 19, in __init__
    from tentacle_pi import TSL2561
ImportError: No module named 'tentacle_pi'

#### Tried to install tentacle_pi:
sudo pip install tentacle_pi
fails with:
tentacle_pi/am2315/src/am2315_ext.c:9:20: fatal error: Python.h: No such file or directory

kizniche, 2018-02-27:
Thanks for the bug report. I pushed a fix that you can upgrade to when the next release is made. If you want to edit your tsl2561.py module to be able to use the changes now, replace the contents of that file with the new one, then restart the daemon with ```sudo service mycodo restart```

https://github.com/kizniche/Mycodo/blob/d64b5b9abeabae591257bba685ac501ece71aad5/mycodo/inputs/tsl2561.py#L1-L67

drgrumpy, 2018-02-27:
Wow, thanks for the quick response. Can confirm that the changes are working, and data is now available for the sensor.

kizniche, 2018-02-27:
Thanks for the feedback. Let me know if you find any other issues.

closed 2018-02-27 by kizniche