kizniche/Mycodo #415: Conditional activation of PID

drgrumpy, 2018-02-27:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.6.0
 
#### Problem Description
 - Unable to use a conditional to activate a PID
 - It may be that I misunderstand how this is supposed to work...

#### Steps
 - Function -> conditional measurement or output Add
 - Activate PID add action
 - Expected then to be able to select an existing PID but there is nowhere to do this, like there is a dropdown box missing.
- Clicking on Save button by the 'Activate PID' results in: 
Error: Mod Conditional: PID must be set

Nothing shows up in the logs...

kizniche, 2018-02-27:
Can you take a screenshot (with the settings expanded- click the plus icons to expand) so I can see how you have it configured?

drgrumpy, 2018-02-27:
Screenshot attached

[image]

kizniche, 2018-02-27:
That appears to be a bug. Let me look into it.

kizniche, 2018-02-27:
I just pushed a fix and made a new release that you can upgrade to.

closed 2018-02-27 by kizniche