kizniche/Mycodo #416: Problems with conditionals to manipulate PIDs

zsole2, 2018-02-28:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.6.1
 
#### Problem Description

Started to set up my switches and conditionals to start/pause/stop a PID. There are a few problems.
First, the Edge conditional cannot read the state of the pin, giving the error: 
```
2018-02-28 13:04:51,459 - mycodo.conditional_1 - ERROR - Exception reading the GPIO pin
2018-02-28 13:04:51,461 - mycodo.conditional_1 - ERROR - GPIO not configured correctly or GPIO state not verified
```
This was the setting for the conditionals:
[image]

I was just to curious to see what happens, because there was some strange behavior with the Edge detection. Or the conditionals, I may not know. Sometimes, the switch registered the activation/deactivation of the PID in the log, sometimes not. Even when it was not logged, the action may have been carried out, since re-triggering the action through the GUI give the warning that the PID is already at the desired state. 
With the pause/resume, the same happened, and even registered a paused state in the log when the controller seemed to be deactivated.
Unfortunately, the GUI is not updated according to the conditional action, neither in the Functions page, nor the Dashboard. The color of the controls and the buttons do not change.
So, there are a lot of issues here, how should  I proceed?

kizniche, 2018-03-01:
The commit above should fix PID activate/deactivate/pause/resume conditional actions. I just tested all 4 with the Edge Input. Let me know if it works for you.

zsole2, 2018-03-01:
Thanks, it works perfectly!
I an still a little bit confused the second part of the Edge  conditional: somewhere else you mentioned that if the edge detection is set for a given pin, then its state cannot be read. Then what is the purpose of this second part? And next question, does the Period apply to the Edge Detected selection?
On the other hand, for the activate/deactivate button I set up a GPIO State Input, and checking it every 60 secs seems OK. It now plays nicely with the pause/resume function, although a color indication on the dashboard widget would be nice for the paused state.

kizniche, 2018-03-01:
> I an still a little bit confused the second part of the Edge conditional

The second part is a reoccurring check of the pin state. The rising/falling edge may only occur once, triggering an event once. However, if you require an event to be triggered continually until the state changes, the second part is useful. For instance, you have a float sensor that when the water level drops, it brings the pin to a LOW state. If you have a solenoid open a fluid valve for 5 seconds every 20 seconds that the pin is LOW, then it will trigger over and over until the pin becomes HIGH. This can be useful if you only want to increase the water level small increments at a time because you don't know exactly how far below the float sensor the water level actually is, and increasing the level in small amounts may be a good approach for this particular application.

> you mentioned that if the edge detection is set for a given pin, then its state cannot be read

Where did I say this? There must have been some other context. If the state can't be read from a pin set up for edge detection, I will need to remove this second part of the edge detection conditional.

> does the Period apply to the Edge Detected selection

No, only if the second part is selected does Period have an affect.

> On the other hand, for the activate/deactivate button I set up a GPIO State Input, and checking it every 60 secs seems OK.

This may have been why I added the GPIO State Input, if the state cannot be read form a pin that has been set up for edge detection. I'll investigate this later.

zsole2, 2018-03-01:
This was the discussion I remembered from #387:
> The issue is that you cannot use GPIO.add_event_detect() more than once for the same pin (at the same time). It appears you have an Edge Input and an Edge Conditional for the same pin. There really isn't a need for both, as the dedicated Conditional section now takes the place of what the Edge Input used to do (which was have conditionals associated with it).

So it is not the same. On the other hand, the reason for adding the GPIO State Input was exactly the situation you describe, I had it with my humidifier/refill container-pair. I had lot of trouble with the Edge detection, now they are humming along nicely with State detection.

kizniche, 2018-03-02:
After I had said that I figured out the issue and fixed it with https://github.com/kizniche/Mycodo/commit/aa4642702ef65f5b6c5ab122bc7cb56349dd3312

You can use an Edge Input and create an edge conditional. In fact, Edge conditionals can only use edge inputs. You should be able to read the pin fine by selecting the second edge conditional option.

kizniche, 2018-03-02:
```
2018-02-28 13:04:51,459 - mycodo.conditional_1 - ERROR - Exception reading the GPIO pin
2018-02-28 13:04:51,461 - mycodo.conditional_1 - ERROR - GPIO not configured correctly or GPIO state not verified
```

So this error persists with the latest(ish) version of Mycodo? That's not good. I'll look into it and see what I can do. I may have to remove the second part of the edge conditional. I don't like removing things, but it seems the measurement conditional is a better fit if these things can't play nicely together.

zsole2, 2018-03-02:
This part I did not test, but everything else works great. I did a few switching on-off and pause-resume cycles, so now it is time to cook something... Tomorrow is sous-vide day!
I understand your sentiment not removing parts. On the other hand, it really seems redundant. It  may also confuse new users when trying to decide how to implement things, and they see two seemingly identical output choices for the two different conditionals.

I was thinking about two features with the sous-vide controller:
1) when tuning, the PID widget is really great showing all the P-I-D values and how this affects the outcome. This makes them showing on the graph unnecessary, since there only the single lines are shown, not their interrelation. On the other hand, showing the P-I-D parameters on the graph could have additional value in tuning, when you can look back historically and see that a given  parameter-change resulted in a certain change in the output. 
2) pushing with a double-pole switch to pause-resume can help avoid unnecessary reaction from the PID when the lid is removed (sensor cools down) and put back. I'll have to see the lag in the temperature values, but I could imagine that it takes some time to re-equilibrate the  temperature when the sensor immersed into the water. Ideally the double-pole manual switch could be replaced by a contact-switch to detect the removal of the lid and putting it back. However, in that case the resume action from the switch should have to have a delay that can be set, depending of the delay of the system. This is rather theoretical now, but I'm rather curious how it will turn out.

kizniche, 2018-03-04:
I just tested the GPIO State option in the Edge Conditional, and I concur that it would be best if it is removed, as 1) It doesn't work, and 2) had it worked, it would be redundant (and confusing to new users) because Measurement Conditionals perform that action. With this commit, I believe this issue is completely solved. Let me know if you find any issues with your sous-vide setup.

closed 2018-03-04 by kizniche