kizniche/Mycodo #417: Infinite Page refresh

RynFlutsch, 2018-02-28:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 5.6.1
#### Problem Description
 Clean instal today without errors.
No errors @ mycodo and nginx.log 

Got my browser warning (unsave cert.) Proceeding to the admin creating result in infinite page refresh ( information at the browser adressline appears for some ms between the refresh: Looking for Page)

Older mycodo versions (5.5.22) are updated to 5.6.1 without any problems.

Got any idea how to fix this?

kizniche, 2018-02-28:
It doesn't sound like a Mycodo issue. Try a different browser. You might have cookies, session, or cert saved from a previous install that's causing the issue.

RynFlutsch, 2018-03-02:
Yes, Safari and Firefox fail to load the Page. Opera mini is working... (iOS) But no idea why.

closed 2018-03-02 by RynFlutsch