kizniche/Mycodo #418: Bug: The first sensor to be read won't show the correct value

Kattihatt, 2018-03-01:
Hello again!

So I'm running: 5.5.24 and I had this issue:

I'm trying to read values from the MCP3008, but the first value in the data input list, will not read correct values. 

When trying to read values from the MCP3008, I use a transistor switch to the soil moisture sensor circuit and when I had this turned on at all times, it worked as expected. When I on the other hand turned this off, and turned it on by choosing pre-output, I got no reading on the first sensor in the list. This of course, made me think it was a hardware error, and I looked at the circuit but found nothing wrong, also it was working before, when it was turned on all the time. So the next thing to do was to try to increas the on timing before grabbing the data, which changed nothing, even if I set a pre-output duration of 30 seconds. 

So the next thing I tried, was to simply add another data input from the MCP3008, and place it above the first actual value to be read. Now the first one still shows a nonsensical value, but the second one, which is the value I want, works fine. Now this made me come to the conclusion that it is a software error.

I see nothing of relevance in the logs, no complaints or errors.

I'd say it's not a top priority to try to settle this since the workaround was as simple as creating a new data input, but it might be worth knowing that the problem exists and if someone else has it maybe they have something more to contribute. I hope it's good for something :)

Kattihatt, 2018-03-01:
There was a connection between analog output and digital output, for some reason this made it work in above mentioned way. I can't understand why keeping the power on, or why changing in the above mentioned way would affect it. But I blamed your fine software for no reason and I apologize :)

Kattihatt, 2018-03-01:
But on the other hand, the problem persists, even though the hardware flaw is fixed.

kizniche, 2018-03-02:
Does the test script that displays all the measured values at once produce this issue?

> 10 second pre-output

What are you using the pre-output for, in relation to this issue?

kizniche, 2018-03-02:
I have a hunch what might be happening. I just pushed a potential fix. I haven't made a release, so you can't do a release upgrade. However, if you want to update to the latest github commit, you can do this:

1. Edit Mycodo/mycodo/config.py and change line 18 ```FORCE_UPGRADE_MASTER = False``` from ```False``` to ```True```
2. Restart the frontend by selecting [Gear Icon] -> Restart Frontend
3. Go to the upgrade page, [Gear Icon] -> Upgrade

Then you will have the option to upgrade to the repo master.

Kattihatt, 2018-03-10:
Hi! I forgot to update the thread, sorry! I haven't been sure it's not a hardware error, so I redesigned the circuit, and redesigned it again. I'm pretty certain there's not a problem with the hardware and I've control measured. The problem doesn't occur if the the probes are constantly powered, but that is not a viable solution since that means they'll corrode in no-time, so I've been trying to fiddle with different pre-output settings but to no avail.

Here are some printscreens, the sensors are working fine and the script returns correct values. https://imgur.com/a/kx3OY

The first two parts of the soil moisture sensor readings where they go up and down was me measuring things, so that's nothing to worry about.

Kattihatt, 2018-03-10:
I upgraded, restarted backend and frontend, tested, but the first reading is still not showing the correct value. The script grabs the correct value each time, so I wouldn't know what's wrong. In the datasheet: https://cdn-shop.adafruit.com/datasheets/MCP3008.pdf I got told in #raspberrypi to check 6.1 and 6.2 and that it might explain the error. I am trying to understand how but then I know nothing of how the library is working and the script is using the same library as Mycodo, so that is still confusing.

Kattihatt, 2018-03-10:
It seems to have no issues with turning on and off in itself: https://i.imgur.com/SL5kM56.jpg Did it repeatedly here and I could not see any deviations here. I looks just as it should.

kizniche, 2018-03-10:
The pre-output turns off directly before taking a measurement. Do you think this could be causing the issue?

Kattihatt, 2018-03-11:
I guess it could, I've wondered how it all works. What I really would need would be one pre-output, turn on the transisor switch before taking ALL the readings, not turning it on and off as I suspect is happening now. Turn on 5 seconds before, read channel 0 to 7, turn it off. If it turns off right before, then I guess that might explain it, if it turns off the switch before taking a reading, since that is the probe floating and it's pretty much just taking a reading across one resistor, but why wouldn't it affect the other readings, but only the first?

kizniche, 2018-03-11:
I could add the option to run the pre-output during the measurement. That will at least rule out that as the cause.

Kattihatt, 2018-03-11:
Yes, I think that might be a very useful option to have anyway. This is probably not the only scenario where it might be used. I wish I could add these things myself! I'm glad your so helpful and seem to like working on this project so much at least!

kizniche, 2018-03-11:
I just released v5.6.4 that includes the new Pre Output During Measurement option for Inputs. Give it a test when you can and see if it fixes the measurement issue.

Kattihatt, 2018-03-11:
Now I have no idea what's going on. https://i.imgur.com/7iXD2AG.jpg

Two of the sensors are at 90% which I guess is somewhat realistic, three of them are close to 0%. The Adafruit MCP3008 library example script still returns values close to eachother and that seems representative. I can't even begin to speculate why this happened.

kizniche, 2018-03-11:
Is the scale set up properly for each input?

kizniche, 2018-03-11:
To me it looks like the voltages are being measured correctly, it's just the scales that aren't working nicely with the measured voltage.

Kattihatt, 2018-03-11:
The voltages vary? Sensor #1 and #3 are around 1,2 V, whilst the other three are around 3,3 V. They're all in the same pot, so I didn't think that they were correct. I'm also just looking at the voltage right now. 

They are set up identically: https://i.imgur.com/Eey9W6l.png and should therefore show the same result.

kizniche, 2018-03-11:
Can you switch a low-reading probe with a high-reading probe and verify if the two channels' measurements adjust appropriately?

Kattihatt, 2018-03-11:
I switched the probes, not the boards themselves, since they're soldered. I waited until it took a new reading, looks completely different. Can't see anything depending on the switching of probes, it seams to be either completely off or a reasonable value that I've found for a bit below 100%.

https://imgur.com/a/050vQ

kizniche, 2018-03-11:
Which probes did you switch?

Kattihatt, 2018-03-11:
#2 with #3.

kizniche, 2018-03-11:
I'm not sure what to make of your readings. Is the lower measurement associated with the hardware?

Kattihatt, 2018-03-12:
Before, I would've said "Maybe, I'll have to check". Now I've checked and redesigned three times with no different result so I consider it ruled out. It's not a hardware problem. I would also say it's confirmed by the Python script returning correct values. I did another test like the one above and it shows no incorrect readings in connection to turning the transistor switch on and off. 

https://imgur.com/a/fNN51

The first reading is lower, but I guess this is because it takes a reading every half second and it takes about 2 seconds before the voltage stabilizes based on what I've seen when measuring with the multimeter. But it's not close to what is being shown in Mycodo on some of the readings. This time it's only the first reading that is correct, and the other ones are showing near to 0%. So like I said, I have no clue what's going on but I'm also starting to suspect the pre-output settings. Is there a way to make a pre output setting for a group of data inputs rather than doing it individually? Maybe if it wouldn't turn on and off, but be on for the whole duration of taking readings from 0 - 7 and then turn off when it has the values already? I'm suspecting that it turns the transistor switch on and off, whilst it is taking a reading from one of the other channels, so that it becomes very unrealiable. I guess one way to test this could be to do very different timers on the data inputs. I will try this right now.

kizniche, 2018-03-12:
> I would also say it's confirmed by my multimeter and the Python script returning correct values.

So when switching a probe from a low-reading channel with a probe from a high-reading channel, the low reading channel continues to read low values (and vise-versa with the other channel)?

I see why your percentages are not working correctly. You have "Inverse Scale" enabled. I'm not sure why you enabled it, but it is effectively making your 3.3v measurement register 0% and 0v register 100%.

>  I run the ADC on 5 V and my 0% readings are 5 V

I thought it only ran with a max of 5v. I will fix the code to allow a voltage of whatever you set in the Input settings.

kizniche, 2018-03-12:
I just pushed a fix for correcting the maximum voltage and I found an issue with locking that I also fixed, which will prevent two measurements from attempting to be acquired at the same time, which may or may not be causing your measurement issues. I'll make a release soon so you can update your system and test out the new code. I'm currently in the middle of upgrading from Bootstrap 3 to 4, which is a big undertaking, so I may not be able to make a release as soon as I would like because I must ensure everything is working with Bootstrap 4 before release.

Kattihatt, 2018-03-12:
> So when switching a probe from a low-reading channel with a probe from a high-reading channel, the low reading channel continues to read low values (and vise-versa with the other channel)?

No, it doesn't, it showed up completely random. If you look at the image above (https://imgur.com/a/050vQ) it doesn't seem at all related to the probes themselves. Also, if you look at https://i.imgur.com/j5vC8aK.png there's no real connection between the readings in Mycodo and the readings through the Adafruit_Python_MCP3008 example file. 

This is how it's set up:
```
Mycodo:         ADC:
Channel #1 - Channel 1
Channel #2 - Channel 2
Channel #3 - Channel 3
Channel #4 - Channel 6
Channel #5 - Channel 7
```

I enabled it because before it showed 0% when the probes were immersed in water and 100% when they were completely dry, then I got the correct readings. But the issue right now is the fluctuating values during reading from the ADC rather than any of those practicalities I'd say. Before you pushed the update with being able to keep the output on during reading, I only had the issue with the first reading being off and the other ones being correct, now I got unpredictable readings on all channels. Used to be good readings in 90% of the cases for other channels than the first reading. 

> I thought it only ran with a max of 3.3v. I will fix the code to allow a voltage of whatever you set in the Input settings.

Datasheet says: "Single supply operation: 2.7V - 5.5V" 5 V has quite a lot higher possible sampling rate than 3,3 V.

I made a script today to grab the values and what I did was to take 100 readings and get an average, since the first value could be a bit off. But this wasn't a large issue, it was just slightly off and that if I didn't have enough of a delay. What I did notice though, is that if I had a time.sleep that was under 2 seconds, the readings could be really off. I don't know why this is, because when I turn the transistor switch on and measure with the multimeter, the voltage goes up quick, but then slowly rise and stabilize during two seconds. So I can't see why it would return a value close to 0% or close to 100%, the difference shouldn't be that large.

Don't worry, it's far to summer here in Sweden and there's no greenhouse being used in quite a long time x) Still covered in snow, and of course you work with whatever you like in the pace you like. I don't know how to do many of these things myself and it's my responsibility to learn it rather than putting that on you and expecting anything from you. I'm just very thankful for the help I get.

Kattihatt, 2018-03-15:
So, I updated and tried again. Now I got the complete opposite problem as I had initially. Only the first reading is correct, the following readings are not taken and it's as if the transistor switch is turned off during this time so that the probes are floating. https://i.imgur.com/5oz5eeO.png.

Is it possible to make it so that there is a delay between every read on the MCP3008, so that the delay is equal to the pre-output duration between every reading? Do you think that would help? In case it takes one reading per data input and there is a conflict between readings still. Even better, would probably be  if it was possible to set one pre-output duration for several data inputs, and then have a chosen delay between every read. So that I have ONE pre-output setting for Soil moisture sensor #1-#5, which turns them all on at once, then choose a 2 second delay between every read. Not that such a long duration would be needed but just a customizable one. Would this make sense?

kizniche, 2018-03-15:
> Only the first reading is correct

It looks like your first 2 readings are correct (light and soil).

> Is it possible to make it so that there is a delay between every read on the MCP3008, so that the delay is equal to the pre-output duration between every reading?

This is the expected behavior

> Even better, would probably be if it was possible to set one pre-output duration for several data inputs, and then have a chosen delay between every read. So that I have ONE pre-output setting for [...]

It makes sense but I'm not prepared to for coding such an endeavor.

kizniche, 2018-03-15:
I'm reviewing the code for issues. Stay tuned.

kizniche, 2018-03-15:
I think I found the issue

kizniche, 2018-03-15:
I just pushed a potential fix. Can you follow the instructions on the [wiki page I just created](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) to update your system to test this new code?

I'm not going to release this code yet because I don't know how stable it is, so if you're willing to test, we can iron out the bugs before a release.

kizniche, 2018-03-15:
Have you tested at all without using the pre-output?

Kattihatt, 2018-03-15:
> It looks like your first 2 readings are correct (light and soil).

Yes, I didn't count light, it's always correct and is not connected to any pre-output. It's always on.

I'll try it out later and get back to you with the results.

Kattihatt, 2018-03-15:
What I did now after some fiddling, was to set the last pre-output duration to 15 seconds, then I deactivated all of the other pre-output settings. Completely turned them off. Then it seems to be turned on 15 seconds before taking the last reading, after that it takes all the readings and turn it off when the last reading is taken. So the issue seems to be somewhere in the pre-ouput handling. But this fixed it for now at least!

kizniche, 2018-03-15:
Did you update to the latest code with my instructions, or not?

kizniche, 2018-03-15:
What you just described is unexpected behavior. We can't have the system running unexpectedly. If you can test my latest code (the fix), we can see if the expected behavior is restored.

kizniche, 2018-03-15:
Input measurements can become out of sync, and if that happens, your workaround will no longer work. My fix incorporates a lock for each input during the pre-duration, so only one can utilize a specific pre-output at a time, ensuring th inputs are read sequentially. I just need someone with the hardware that can test the software (the fix).

Kattihatt, 2018-03-15:
I did not. I will do that tomorrow. Already in bed. Just checked the graph, and for most of the time, the values are correct now, but suddenly drops whilst taking a reading, just to go back to normal right after. Don't know how to deal with this. Got to get back.

Kattihatt, 2018-03-16:
So I did:
>  "FORCE_UPGRADE_MASTER = True
> Then save your edit.
> 
> Restart the Mycodo frontend
> sudo service mycodoflask restart"

It did an update. I removed the temporary "fix" from yesterday and turned the pre-output back to 5 seconds. Now I got the same thing back, first reading is correct, the rest are at 0%.

kizniche, 2018-03-16:
It is very difficult to test fixes without the hardware. Diagnostics can occur exponentially faster if I have access to the Pi and Mycodo. Could you provide me SSH and Mycodo credentials to test code to come up with a fix?

Kattihatt, 2018-03-16:
I was going to suggest that but forgot about it. I can't find any PM function here, shall I mail the information to you?

kizniche, 2018-03-16:
Contact me at kylegabriel.com/contact

Kattihatt, 2018-03-16:
I did send. Don't know if you got it, it just stays at "Processing"

kizniche, 2018-03-16:
My recaptcha key had expired. Try it again. I just tested and it now works.

Kattihatt, 2018-03-16:
Yes, now it worked.

kizniche, 2018-03-17:
Success!

[image]

kizniche, 2018-03-17:
I'm testing a few more things, so don't cut my access just yet ;)

I found the issue was the locking library I was using wasn't thread safe.

kizniche, 2018-03-17:
You can now see the measurement times are now 5 seconds apart.

kizniche, 2018-03-17:
Okay. I think I fixed all issues with locking throughout Mycodo. I also slated potential support for multiple ADC measurements per ADC Input in the [6.0 Changes](https://github.com/kizniche/Mycodo/wiki/Changes-for-Mycodo-6.0) wiki page that I've been tracking major code refactors.

Thanks for letting me test on your system. Everything looks like it's running nicely and I've logged out. If you notice any more issues, let me know.

kizniche, 2018-03-17:
I have the sample rate set to 60 seconds, not the 300 you originally had. I forgot to change them back before logging out.

Kattihatt, 2018-03-17:
Yeah I just noticed! Good work! The readings now look a lot more wobbly and like they're taking accurate readings, but the first sensor is completely straight and looks different than the others. Don't know why but I'll see if that is something to worry about. 

I'll keep trying it out and see if it works. I'll let it sit for a day or two and if it works fine then I'll start working on conditionals :)

kizniche, 2018-03-17:
I suspect it's due to the power being turned on/off (pre-output). Try disabling the pre-outputs and having it stay on constantly and acquire measurments, to test.

Kattihatt, 2018-03-18:
It's all been running fine up until now when I checked, and saw that that two of the channels show 0%. I check the graph, and at 19:00, they appear to have stopped getting any results. The line just ends abruptly.  It looked to be working well up until now.

kizniche, 2018-03-18:
Can you check the daemon log for errors?

Kattihatt, 2018-03-18:
http://paste.debian.net/1015392/

Kattihatt, 2018-03-18:
Those are the last 100 lines and the only errors that occurred today at least. 

Here are the last 200 lines: http://paste.debian.net/1015393/ and apparently something similar happened the other day as well, don't know why I didn't notice that.

kizniche, 2018-03-18:
I'm not exactly sure why measurements stopped, but I can fix the locking issue. I've also pushed a potential fix for the MCP3008 not working. Try a [manual update](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and give that code a run.

Kattihatt, 2018-03-18:
The temperature on the RPi spiked at the same time as it locked. Could this be connected? I had this last week too when the RPi crashed. It was really hot to the touch, to the degree I added a heatsink.

kizniche, 2018-03-18:
Not sure. The CPU is throttled if it gets too hot, so temperature shouldn't be an issue, unless you have your Pi in an environment with a high ambient temperature.

kizniche, 2018-11-30:
In version 7.0 (yet-to-be-released), the ADC system has been completely redesigned to work better. I'm closing this issue for now, and if you would like to test the 7.0 branch either when it's released or prior to release ([this comment](https://github.com/kizniche/Mycodo/issues/562#issuecomment-441150035) details how), please let me know if your issues are resolved. Part of why I redesigned the ADC system was from several user issues, including yours.

closed 2018-11-30 by kizniche