kizniche/Mycodo #419: USB cam bad quality image

ralphknoops, 2018-03-07:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.6.1
 
#### Problem Description
Please list:

  - Try to make a picture in Mycodo with USB Cam (Lifecam-3000 microsoft), i get bad quality pictures
  - I'm using the standard Mycodo configuration
 
### Errors
I think it's more off a camera problem. I found a topic where they stated that some camera's have problems with the first frames. So did some testing on an other pi and changed the framecapture to 5.
Now it's making good quality pictures :).

Sample code:
fswebcam -S 5 --set brightness=20% -r 1280x720 --no-banner /home/pi/webcam/$DATE.jpg

### Steps to Reproduce the issue:
How can this issue be reproduced?
You need the cam to reproduce it.
 
### Additional Notes
Is it possible to add the funtionality to change the frame you capture (-S) in the webinterface?
I'm no github hero yet so I don't know how to point to the right code.. Mycodo/mycodo/devices/camera.py

Thanks in advance

ralphknoops, 2018-03-09:
Nice!

Trying to do a clean Mycodo install on an other Pi and going to test it. Just need to find out how to get the new fix in.

kizniche, 2018-03-09:
I meant to make a release last night but fell asleep. I'll make the release today, then you can use the upgrade option under the Config menu.

kizniche, 2018-03-09:
I just released v5.6.3, so your upgrade menu should now allow you to upgrade to test the new code.

ralphknoops, 2018-03-11:
Just tested it. Works like a charm, thanks!

closed 2018-03-11 by ralphknoops