kizniche/Mycodo #420: Add value to output command?

fry42, 2018-03-08:
Hi,

It would be great to have the option of switching the cmd output between a simple on/off variant and a pwm style with a value (0-100).
I'm thinking of a wireless LED light with varying brightness or a wireless fan...

Tanks!
 Stephan

kizniche, 2018-03-08:
Can you elaborate on what you're talking about?

Mycodo already supports PWM output.

fry42, 2018-03-09:
Sure:
When you add a "PWM Output" it can be set to a duty cycle (0-100%)
The "Execute Command" output can only handle on or off.
i.e. "/bin/lighton.sh" and "/bin/lightoff.sh"
I'm talking about passing a duty cycle value to a command like:
"/bin/lightlevel.sh 75", "/bin/lightlevel.sh 0", "/bin/lightlevel.sh 100" 

Or did I completely miss something?

kizniche, 2018-03-09:
Ah! Yes. There is an executable in the Mycodo/mycodo/mycodo_client.py that can be used to directly control the Mycodo daemon, to turn outputs on and off. I'll add the option to set the duty cycle for the case of PWM outputs.

I meant to make a release last night but fell asleep. I'll add this new feature, then make a release today.

kizniche, 2018-03-09:
I'm not quite understanding what you want to accomplish.

> But I can't define a command as an output with a % value. So that mycodo calculates the value and passes it on to my script.

This is what's confusing me. Can you elaborate on how you would like the system to operate and what happens when you do something (or if it's automated)?

kizniche, 2018-03-09:
> I want to build a LED Light that can be controlled via WiFi=>TCP/IP and is not directly connected to the RPi.

This also confuses me. How does the light receive the PWN duty cycle?

kizniche, 2018-03-09:
Some questions I have:

1. How is your PWM duty cycle determined?
2. How is the interface with Mycodo so that the PWM duty cycle is calculated?
3. Where in Mycodo are you referencing the use of a command that you would like PWM duty cycle incorporated?

fry42, 2018-03-09:
If, for instance, I want to simulate a sunrise/sunset with a lamp that can be controlled in brightness but is not connected to the RPi. I would generate the values with one of the curve functions and then pass them on to the light.
Mycodo would execute my script with the value, like "/bin/lightcontrol.sh 67".
My script would send the value to the lamp by generating a data packet or some http request.

kizniche, 2018-03-09:
I get it now. Let me see what I can do.

kizniche, 2018-03-09:
This can now be used with PWM Method timers that will take the duty cycle derived from a Method and then execute whatever command you want with the included duty cycle in the command. This should be what you requested, which is a nice feature addition. Even the other features I added were good additions, even though I actually misinterpreted what you wanted. Let me know if this allows you to accomplish what you wanted to do.

kizniche, 2018-03-09:
I just made a release, so your system can now use the upgrade feature to test the new code.

fry42, 2018-03-10:
Wow! Thanks! It's exactly what I wanted - and faster than I would have ever hoped for :)
With those new features it should be easy to integrate in any external/wireless controller system.  
I'll test it as soon as possible!

zsole2, 2018-03-10:
In the way you have added originally the duty cycle of the PWM output as a parameter to  mycodo_client.py, the PID setpoint could be also useful to control as a parameter of mycodo_client.py. Thinking of my sous-vide controller...

kizniche, 2018-03-10:
> the PID setpoint could be also useful to control as a parameter of mycodo_client.py. 

Agreed! I just pushed a commit that adds support for both getting and setting the following PID values: Setpoint, Error (get only), Integrator, Derivator, Kp, Ki, and Kd.

closed 2018-03-31 by kizniche